coolsnowwolf/lede #8319: Redmi AC2100 WEB界面修改LAN保存后会出现WiFi不能自动启动

Raytheon-S, 2021-11-26:
在WEB界面修改LAN页面任何信息保存后，会出现路由器开机WiFi不能自动启动。
Redmi AC2100
Powered by LuCI Master (git-21.329.83689-00cb491) / OpenWrt R21.11.11

临时的解决办法是不要在LAN界面修改任何信息。登录SSH，用vim /etc/config/network修改lan里需要的信息。

JiaRZ, 2021-11-27:
k2p也出现这个问题，昨天编译的最新版

Raytheon-S, 2021-11-27:
>我试了在编译之前修改config_generate文件，或者编译后用SSH登录改写Lan IP都可以临时解决。只等L大进一步完善了。

choyri, 2022-01-06:
问题 +1；padavan 内核旧且苹果设备打开缓慢，而 lede 无线有问题，难受。

kzkzzzz, 2022-01-30:
> 
同样的问题，发现是开机没把wifi网卡桥接进去，暂时在自启加上命令，手动开启就好了
```bash
# 启动2.4g 和 5g 信号
ip link set ra0 up
ip link set rai0 up

# 桥接网卡
brctl addif br-lan ra0
brctl addif br-lan rai0
```

choyri, 2022-01-31:
@kzkzzzz 我除了没启动的问题之外，还有一个更无语的问题，启动后随机一段时间就自动关闭，且无法再次启动，只能重启设备解决 🥲

LAhuangao, 2023-07-08:
> > 
> 
> 同样的问题，发现是开机没把wifi网卡桥接进去，暂时在自启加上命令，手动开启就好了 @choyri
> 
> ```shell
> # 启动2.4g 和 5g 信号
> ip link set ra0 up
> ip link set rai0 up
> 
> # 桥接网卡
> brctl addif br-lan ra0
> brctl addif br-lan rai0
> ```

The reason seems to be that the LAN ip address was changed in the config.

M4TRIX04, 2023-10-10:
lanCheck=`uci get network.lan.ifname` && uci set network.lan.ifname="$lanCheck rai0 ra0" &&  uci commit

kelvin-ye, 2023-11-17:
这样设置也是有效的，每次修改完lan接口的设置，原本的eth0.1 ra0 rai0桥接不知为何就会被删除，
只能手动修改以下位置，也能修复这个问题。貌似这才是最根本的原因：
lan接口的“物理设置”，勾选“桥接接口”，勾选并在“自定义接口”填入：“ra0 rai0”，勾选“交换机 VLAN: "eth0.1" (lan)”

coolsnowwolf, 2026-09-05:
该报告基于 LuCI/OpenWrt 21.11（2021 年）的旧版 netifd/wifi-scripts，当前 LEDE 已更新 netifd、wifi-scripts 及 Redmi AC2100 的 MT7621 设备支持；相关无线重载、配置生成和 NVMEM/MAC 修复已在后续提交中完成。issue 没有提供保存后的 `/etc/config/network`、`wifi reload` 输出或启动日志，无法在现行代码复现。建议升级到当前固件并在仍复现时提供完整日志，关闭该历史 issue。

closed 2026-09-05 by coolsnowwolf