coolsnowwolf/lede #8323: 断网重拨后DHCP 服务器失效

feng5064, 2021-11-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [* ] 没有类似的issue

## 2. 详细叙述
op是PVE（7.0）的虚拟机，电信单线双拨，WAN口直通网卡，LAN口VirtIO (半虚拟化)。使用上没什么问题，但有一致命问题是一旦断网重拨后DHCP 服务器失效（每次都是这样），这样导致家里局域网内所有主机因为获取不了IP而无法联网，台式机要手动指定IP地址后才能上网。OP重启后DHCP 服务器正常。
同一台主机硬件，以前用实体机安装OP从未发生过DHCP 服务器失效的问题，但用PVE的虚拟机后就有这问题。我不清楚这到底是OP的问题还是PVE有问题。
### (1) 具体问题
 断网重拨后DHCP 服务器失效
A：

### (2) 路由器型号和固件版本

A：主机名	OpenWrt
型号	Common KVM processor : 2 Core 8 Thread (CpuMark : 118220.777302 Scores)
CPU 信息	2400.000 MHz
固件版本	OpenWrt R21.11.11 / LuCI Master (git-21.280.14233-9b796b1)
内核版本	5.4.158

### (3) 详细日志

A：Sat Nov 27 13:13:08 2021 daemon.info pppd[6417]: LCP terminated by peer
Sat Nov 27 13:13:08 2021 daemon.info pppd[6417]: Connect time 2885.1 minutes.
Sat Nov 27 13:13:08 2021 daemon.info pppd[6417]: Sent 2064828016 bytes, received 2398139219 bytes.
Sat Nov 27 13:13:08 2021 daemon.info pppd[6416]: LCP terminated by peer
Sat Nov 27 13:13:13 2021 user.info mwan3track[7392]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:13:14 2021 user.info mwan3track[8182]: Check (ping) failed for target "www.baidu.com" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:13:16 2021 user.info mwan3track[7392]: Check (ping) failed for target "114.114.114.114" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:13:17 2021 user.info mwan3track[8182]: Check (ping) failed for target "114.114.114.114" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:13:19 2021 user.info mwan3track[7392]: Check (ping) failed for target "119.29.29.29" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:13:20 2021 user.info mwan3track[8182]: Check (ping) failed for target "119.29.29.29" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:13:29 2021 user.info mwan3track[7392]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:13:30 2021 user.info mwan3track[8182]: Check (ping) failed for target "www.baidu.com" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:13:32 2021 user.info mwan3track[7392]: Check (ping) failed for target "114.114.114.114" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:13:33 2021 user.info mwan3track[8182]: Check (ping) failed for target "114.114.114.114" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:13:35 2021 user.info mwan3track[7392]: Check (ping) failed for target "119.29.29.29" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:13:36 2021 user.info mwan3track[8182]: Check (ping) failed for target "119.29.29.29" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:13:45 2021 user.info mwan3track[7392]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:13:46 2021 user.info mwan3track[8182]: Check (ping) failed for target "www.baidu.com" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:13:48 2021 user.info mwan3track[7392]: Check (ping) failed for target "114.114.114.114" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:13:49 2021 user.info mwan3track[8182]: Check (ping) failed for target "114.114.114.114" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:13:51 2021 user.info mwan3track[7392]: Check (ping) failed for target "119.29.29.29" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:13:52 2021 user.info mwan3track[8182]: Check (ping) failed for target "119.29.29.29" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:14:01 2021 user.info mwan3track[7392]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:14:02 2021 user.info mwan3track[8182]: Check (ping) failed for target "www.baidu.com" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:14:04 2021 user.info mwan3track[7392]: Check (ping) failed for target "114.114.114.114" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:14:05 2021 user.info mwan3track[8182]: Check (ping) failed for target "114.114.114.114" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:14:07 2021 user.info mwan3track[7392]: Check (ping) failed for target "119.29.29.29" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:14:08 2021 kern.err kernel: [173216.511991] rcu: INFO: rcu_sched self-detected stall on CPU
Sat Nov 27 13:14:08 2021 kern.err kernel: [173216.515751] rcu: 	1-....: (15000 ticks this GP) idle=fa6/1/0x4000000000000002 softirq=16471255/16471255 fqs=7500
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.520213] 	(t=15002 jiffies g=32272401 q=7970)
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.523328] NMI backtrace for cpu 1
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.526327] CPU: 1 PID: 6417 Comm: pppd Tainted: G         C        5.4.158 #0
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.530027] Hardware name: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.14.0-0-g155821a1990b-prebuilt.qemu.org 04/01/2014
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.537073] Call Trace:
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.539941]  <IRQ>
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.542788]  dump_stack+0x6d/0x8b
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.545873]  nmi_cpu_backtrace.cold.7+0x14/0x53
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.549224]  ? lapic_can_unplug_cpu+0x80/0x80
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.552560]  nmi_trigger_cpumask_backtrace+0xff/0x110
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.556109]  arch_trigger_cpumask_backtrace+0x14/0x20
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.559616]  rcu_dump_cpu_stacks+0x99/0xc9
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.563004]  rcu_sched_clock_irq.cold.91+0x157/0x391
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.566578]  ? account_system_time+0x4b/0x60
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.569938]  ? tick_init_highres+0x20/0x20
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.573254]  update_process_times+0x23/0x50
Sat Nov 27 13:14:08 2021 kern.warn kernel: [173216.576585]  tick_sched_timer+0x54/0x170
[... 476 lines cut ...]
Sat Nov 27 13:33:02 2021 user.err ddns-scripts[18551]: 3322: BusyBox nslookup error: '1'
Sat Nov 27 13:33:03 2021 user.warn ddns-scripts[18551]: 3322: Get registered/public IP for 'jmpazu5064.3322.org' failed - retry 8/0 in 60 seconds
Sat Nov 27 13:33:06 2021 user.info mwan3track[20258]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:33:11 2021 user.info mwan3track[18059]: Check (ping) failed for target "www.baidu.com" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:33:17 2021 user.info mwan3track[20258]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:33:22 2021 user.info mwan3track[18059]: Check (ping) failed for target "www.baidu.com" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:33:28 2021 user.info mwan3track[20258]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:33:33 2021 user.info mwan3track[18059]: Check (ping) failed for target "www.baidu.com" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:33:39 2021 user.info mwan3track[20258]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:33:44 2021 user.info mwan3track[18059]: Check (ping) failed for target "www.baidu.com" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:33:50 2021 user.info mwan3track[20258]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:33:55 2021 user.info mwan3track[18059]: Check (ping) failed for target "www.baidu.com" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:34:01 2021 user.info mwan3track[20258]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:34:06 2021 user.info mwan3track[18059]: Check (ping) failed for target "www.baidu.com" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:34:12 2021 user.info mwan3track[20258]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:34:13 2021 user.err ddns-scripts[18551]: 3322: BusyBox nslookup error: '1'
Sat Nov 27 13:34:13 2021 user.warn ddns-scripts[18551]: 3322: Get registered/public IP for 'jmpazu5064.3322.org' failed - retry 9/0 in 60 seconds
Sat Nov 27 13:34:17 2021 user.info mwan3track[18059]: Check (ping) failed for target "www.baidu.com" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:34:24 2021 user.info mwan3track[20258]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
[... cut ...]
Sat Nov 27 13:34:40 2021 user.info mwan3track[18059]: Check (ping) failed for target "www.baidu.com" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:34:46 2021 user.info mwan3track[20258]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
Sat Nov 27 13:34:51 2021 user.info mwan3track[18059]: Check (ping) failed for target "www.baidu.com" on interface vwan1 (pppoe-vwan1)
Sat Nov 27 13:34:57 2021 user.info mwan3track[20258]: Check (ping) failed for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)

coolsnowwolf, 2021-11-27:
尝试关闭  SFE 加速

feng5064, 2021-11-30:
> 尝试关闭 SFE 加速

感谢L大指导，果然跟SFE加速有关，关闭 SFE 加速后现在正常了！

closed 2021-11-30 by feng5064