coolsnowwolf/lede #8333: uugamebooster-x86_64-v2.18.0.tar 找不到

zhiwen1987, 2021-11-27:
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.cdn.openwrt.org/uugamebooster-x86_64-v2.18.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/uugamebooster-x86_64-v2.18.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
Warning: Transient problem: timeout Will retry in 1 seconds. 5 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:03 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/uugamebooster-x86_64-v2.18.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://downloads.openwrt.org.cn/sources/uugamebooster-x86_64-v2.18.0.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
Warning: Transient problem: timeout Will retry in 1 seconds. 5 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
Warning: Transient problem: timeout Will retry in 2 seconds. 4 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
Warning: Transient problem: timeout Will retry in 4 seconds. 3 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
Warning: Transient problem: timeout Will retry in 8 seconds. 2 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
Warning: Transient problem: timeout Will retry in 16 seconds. 1 retries left.
  0     0    0     0    0     0      0      0 --:--:--  0:00:20 --:--:--     0
curl: (28) Connection timed out after 20000 milliseconds

zhiwen1987, 2021-11-27:
怎么解决

zhiwen1987, 2021-11-27:
time: package/boot/grub2/efi/compile#302.54#63.20#349.77
make[2]: Leaving directory '/home/bl/lede'
package/Makefile:110: recipe for target '/home/bl/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/bl/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bl/lede'
Build failed - please re-run with -j1 to see the real error message
/home/bl/lede/include/toplevel.mk:228: recipe for target 'world' failed

VergilGao, 2021-11-29:
openwrt的uu是从网易那下载的啊
```
PKG_SOURCE_URL:=http://uu.gdl.netease.com/uuplugin/openwrt-$(UU_ARCH)/$(PKG_VERSION)/uu.tar.gz?
```

closed 2021-11-29 by AmadeusGhost