coolsnowwolf/lede #8334: 弄了两天了 还没编译出来 各种报错

zhdot, 2021-11-27:
shadowsocksr-libev-2.5.6.tar.xz: Download from https://github.com/shadowsocksrr/shadowsocksr-libev failed
HTTP Error 500: Internal Server Error
Checking out files from the git repository...
Cloning into 'shadowsocksr-libev-2.5.6'...
fatal: unable to access 'https://github.com/shadowsocksrr/shadowsocksr-libev/': The requested URL returned error: 504
make[3]: *** [Makefile:63: /home/zhdot/Desktop/lede/dl/shadowsocksr-libev-2.5.6.tar.xz] Error 128
make[3]: Leaving directory '/home/zhdot/Desktop/lede/feeds/helloworld/shadowsocksr-libev'
time: package/feeds/helloworld/shadowsocksr-libev/compile#0.48#0.45#107.61
    ERROR: package/feeds/helloworld/shadowsocksr-libev failed to build.
make[2]: *** [package/Makefile:116: package/feeds/helloworld/shadowsocksr-libev/compile] Error 1
make[2]: Leaving directory '/home/zhdot/Desktop/lede'
make[1]: *** [package/Makefile:110: /home/zhdot/Desktop/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zhdot/Desktop/lede'
make: *** [/home/zhdot/Desktop/lede/include/toplevel.mk:230: world] Error 2

MBRjun, 2021-11-28:
看起来是网络问题？

zhdot, 2021-11-28:
> 看起来是网络问题？

用的全局代理 应该不是

WYC-2020, 2021-11-28:
九十九的人编译都没问题，大佬会直接关闭的

zhdot, 2021-11-28:
> 九十九的人编译都没问题，大佬会直接关闭的

关闭就关闭吧 实在不行我就用别人现成的

WYC-2020, 2021-11-28:
编译这玩意，只要是按大佬步骤，并且自己的楼梯要稳基本没问题，如果是二次编译遇到有错误，万能大法基本都能过，如果真是源码哪里有问题，很多人早就issue了哈哈

AmadeusGhost, 2021-11-28:
确实是你网络问题

closed 2021-11-28 by AmadeusGhost