coolsnowwolf/lede #8338: ipv6 ddns更新问题

csfeng888, 2021-11-28:
1，型号竞斗云2
2，最新编译的系统
3，ipv6的ddns只有第一次启动可以正常更新，之后就不行了，以下的错误
 004922  WARN : NO valid IP found
 004922  WARN : Get registered/public IP for 'markw.top' failed - retry 200/0 in 60 seconds
 005022       : #> /usr/bin/nslookup markw.top  >/var/run/ddns/myddns_ipv6.dat 2>/var/run/ddns/myddns_ipv6.err
 005022  WARN : NO valid IP found
 005022  WARN : Get registered/public IP for 'markw.top' failed - retry 201/0 in 60 seconds
 005123       : #> /usr/bin/nslookup markw.top  >/var/run/ddns/myddns_ipv6.dat 2>/var/run/ddns/myddns_ipv6.err
 005123  WARN : NO valid IP found
 005123  WARN : Get registered/public IP for 'markw.top' failed - retry 202/0 in 60 seconds
 005223       : #> /usr/bin/nslookup markw.top  >/var/run/ddns/myddns_ipv6.dat 2>/var/run/ddns/myddns_ipv6.err
 005224  WARN : NO valid IP found
 005224  WARN : Get registered/public IP for 'markw.top' failed - retry 203/0 in 60 seconds
 005324       : #> /usr/bin/nslookup markw.top  >/var/run/ddns/myddns_ipv6.dat 2>/var/run/ddns/myddns_ipv6.err
 005324  WARN : NO valid IP found
 005324  WARN : Get registered/public IP for 'markw.top' failed - retry 204/0 in 60 seconds

csfeng888, 2021-11-28:
还有k2p和 小米的ax6 都是一样的情况

EnnawYang, 2021-11-29:
是不是把ipv6解析关掉了？

wi11iamzhao, 2021-12-03:
最近一个月内的版本X86软路由均正常，DDNS应该和硬件无关
cloudflare ddns ip v4+ip v6
建议检查一下获取ip的方式（特别注意是不是获取到了一个本地IP）

missorforget, 2022-02-11:
DNS-服务器 填入一个ipv6 dns服务器

coolsnowwolf, 2026-09-05:
IPv6 DDNS 报告只有 `nslookup` 无有效地址，缺少接口地址、地址来源设置和完整 ddns 配置；无法判断是脚本缺陷还是网络/地址选择问题，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf