coolsnowwolf/lede #8342: 大佬帮忙解决一下，编译过程中总是报错

SterlingCarter, 2021-11-29:
搞了好几天了，一个固件都没有编译出来，总是各种报错，解决一个又来另一个，最新一个报错，请看
[image]
图片

hzjnet, 2021-11-29:
取消luci-app-music-remote-center、luci-app-shairplay以及关联包的编译选项即可

> 搞了好几天了，一个固件都没有编译出来，总是各种报错，解决一个又来另一个，最新一个报错，请看 [image] 图片

SterlingCarter, 2021-11-29:
> 取消luci-app-music-remote-center、luci-app-shairplay以及关联包的编译选项即可
> 
> > 搞了好几天了，一个固件都没有编译出来，总是各种报错，解决一个又来另一个，最新一个报错，请看 [image] 图片

你好，我的menuconfig配置里面没有选你说的这两个插件，但是还是报错啊
[image]
[image]

hzjnet, 2021-12-02:
> > 取消luci-app-music-remote-center、luci-app-shairplay以及关联包的编译选项即可
> > > 搞了好几天了，一个固件都没有编译出来，总是各种报错，解决一个又来另一个，最新一个报错，请看 [image] 图片
> 
> 你好，我的menuconfig配置里面没有选你说的这两个插件，但是还是报错啊 !
取消luci-app-music-remote-center、luci-app-shairplay以及**关联包**的编译选项即可:
**Sound----shairplay
Libraries---libffmpeg-full**

coolsnowwolf, 2026-09-06:
只有编译失败截图，没有文本日志、失败包或配置，无法定位和验证具体源码问题；当前无法复现，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf