coolsnowwolf/lede #8344: k2p无线模块部分显示不全，手机连接速度2.4G和5G分别为192和433

JiaRZ, 2021-11-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：1.使用默认WiFi驱动编译（mt7615d和mt7615d_dbdc）时，手机连接速度只有使用插件application里的 mtwifi 插件编译时的一半， 
         实际测速 时又感觉没啥问题.(新版插件库不是取消了mtwifi吗？)
      2.默认wifi驱动下，状态>概览下的5G wifi传输速率一直显示的是问号.
      3.默认wifi驱动下，无线>已连接站点下的接受速率和发送速率那里频宽一直为0Mhz.

### (2) 路由器型号和固件版本

A：K2P    OpenWrt R21.11.11 / LuCI Master (git-21.329.83689-00cb491)最新编译

### (3) 详细日志

A：

linkeqi1002, 2022-05-14:
一样的问题，不知道怎么回事

Journalist-HK, 2022-09-01:
22.03源码编译mt7615d驱动，WiFi工作正常，可以上网，但是无法在LuCi中设置密码等参数，求助如何手动配置。#9384
通过SDK编译kmod，通过Image Builder生成固件，使用如下参数
```
make image PROFILE="phicomm_k2p" PACKAGES="-kmod-mt7615e -kmod-mt7615-firmware -wpad-openssl kmod-mt7615d_dbdc wireless-tools -dnsmasq dnsmasq-full luci-ssl luci-compat luci-app-firewall luci-app-opkg luci-app-uhttpd luci-app-upnp luci-app-wol"
```
固件应该是没有问题，就是和官方无线界面兼容不是很好。
[image]

Journalist-HK, 2022-09-01:
经测试，直接编辑 config 可以修改密码，加密方式设置为 psk2。其他参数未测试。

Journalist-HK, 2022-09-04:
发现无线网络无法获得IPv6地址，有线网可以。

closed 2022-10-31 by aiamadeus