coolsnowwolf/lede #8347: 编译报错，卡在ffmpeg这里，无法进行编译

SterlingCarter, 2021-11-29:
最近编译小米R3P路由器的OpenWrt的固件，老是卡在ffmpeg这里，请问大神们该怎么解决呢？有大佬支下招么？

mipsel-openwrt-linux-musl-gcc -Llibavcodec -Llibavdevice -Llibavfilter -Llibavformat -Llibavresaml/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/angel/lede/staging_dir/toWl,--as-needed -Wl,-z,noexecstack -Wl,--warn-common -Wl,-rpath-link=:libpostproc:libswresample:liresample   -o ffmpeg_g fftools/ffmpeg_opt.o fftools/ffmpeg_filter.o fftools/ffmpeg_hw.o fftools/c-lswresample -lswscale -lavutil  -lm -L/home/angel/lede/staging_dir/target-mipsel_24kc_musl/usr/lt-mipsel_24kc_musl/usr/lib -lz -L/home/angel/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lgsl/usr/lib -lz -L/home/angel/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lfdk-aac -L/home/agel/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lm -lshine -lm -lm -pthread -lm
/home/angel/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-muvformat.so: undefined reference to `__atomic_fetch_sub_8'
/home/angel/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-muvformat.so: undefined reference to `__atomic_store_8'
/home/angel/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-muvformat.so: undefined reference to `__atomic_load_8'
/home/angel/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-muvformat.so: undefined reference to `__atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:124: ffmpeg_g] Error 1
make[4]: Leaving directory '/home/angel/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg
make[3]: *** [Makefile:751: /home/angel/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg
make[3]: Leaving directory '/home/angel/lede/feeds/packages/multimedia/ffmpeg'
time: package/feeds/packages/ffmpeg/full/compile#652.47#53.77#713.96
    ERROR: package/feeds/packages/ffmpeg failed to build (build variant: full).
make[2]: *** [package/Makefile:116: package/feeds/packages/ffmpeg/compile] Error 1
make[2]: Leaving directory '/home/angel/lede'
make[1]: *** [package/Makefile:110: /home/angel/lede/staging_dir/target-mipsel_24kc_musl/stamp/.p
make[1]: Leaving directory '/home/angel/lede'
make: *** [/home/angel/lede/include/toplevel.mk:230：world] 错误 2
[image]

yonggedebaqi, 2021-11-29:
用4.0.6版本吧

SterlingCarter, 2021-11-29:
> 用4.0.6版本吧

什么意思？

hzjnet, 2021-11-30:
> 用4.0.6版本吧

谢谢，能给个链接吗？

SterlingCarter, 2021-11-30:
> > 用4.0.6版本吧
> 
> 谢谢，能给个链接吗？

请问大神我这个问题怎么解决啊？

SterlingCarter, 2021-11-30:
> > 用4.0.6版本吧
> 
> 谢谢，能给个链接吗？

https://ffmpeg.org/releases

hzjnet, 2021-12-06:
> > > 用4.0.6版本吧
> > 
> > 
> > 谢谢，能给个链接吗？
> 
> https://ffmpeg.org/releases

谢谢 ,不过这个链接估计不能用吧? patch这些咋办? 我去试试

SterlingCarter, 2021-12-07:
> > > > 用4.0.6版本吧
> > > 
> > > 
> > > 谢谢，能给个链接吗？
> > 
> > 
> > https://ffmpeg.org/releases
> 
> 谢谢 ,不过这个链接估计不能用吧? patch这些咋办? 我去试试
> 
> [image] 不行啊
> 
> 我测试7621 79xx不行，但是原版源码中4.4版本ipq4/6/8都正常

这我就不知道了，最近编译没有报这个错误了

closed 2022-02-12 by Beginner-Go