coolsnowwolf/lede #8354: 代码升级后编译通不过

gswangmd-gssi, 2021-11-29:
pplying ./patches/200-ecc-rng.patch using plaintext: 
patching file wolfcrypt/src/ecc.c
patching file wolfssl/wolfcrypt/ecc.h
touch /home/gswangmd/lede/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/.prepared_1d71c21b3882e195552b366a5f671dda_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/gswangmd/lede/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/.configured_*
rm -f /home/gswangmd/lede/staging_dir/target-x86_64_musl/stamp/.wolfssl_installed
(cd /home/gswangmd/lede/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /home/gswangmd/lede/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/home/gswangmd/lede/staging_dir/host/bin/autom4te AUTOCONF=/home/gswangmd/lede/staging_dir/host/bin/autoconf AUTOMAKE=/home/gswangmd/lede/staging_dir/host/bin/automake ACLOCAL=/home/gswangmd/lede/staging_dir/host/bin/aclocal AUTOHEADER=/home/gswangmd/lede/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/gswangmd/lede/staging_dir/host/bin/libtoolize LIBTOOL=/home/gswangmd/lede/staging_dir/host/bin/libtool M4=/home/gswangmd/lede/staging_dir/host/bin/m4 AUTOPOINT=true LIBTOOLIZE='/home/gswangmd/lede/staging_dir/host/bin/libtoolize --install' /home/gswangmd/lede/staging_dir/host/bin/autoreconf -v -f -i -s -B /home/gswangmd/lede/staging_dir/host/share/aclocal -I /home/gswangmd/lede/staging_dir/target-x86_64_musl/host/share/aclocal -I /home/gswangmd/lede/staging_dir/hostpkg/share/aclocal -I /home/gswangmd/lede/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . . || true; fi; );
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /home/gswangmd/lede/staging_dir/host/bin/aclocal -I /home/gswangmd/lede/staging_dir/target-x86_64_musl/host/share/aclocal -I /home/gswangmd/lede/staging_dir/hostpkg/share/aclocal -I /home/gswangmd/lede/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . --force -I m4
configure.ac:35: error: Libtool version 2.4.2 or higher is required
/home/gswangmd/lede/staging_dir/target-x86_64_musl/../host/share/aclocal/libtool.m4:48: LT_PREREQ is expanded from...
configure.ac:35: the top level
autom4te: /home/gswangmd/lede/staging_dir/host/bin/m4 failed with exit status: 63
aclocal.real: error: echo failed with exit status: 63
autoreconf: /home/gswangmd/lede/staging_dir/host/bin/aclocal failed with exit status: 63
(cd /home/gswangmd/lede/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/./; if [ -x ./configure ]; then find /home/gswangmd/lede/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/ -name config.guess | xargs -r chmod u+w; find /home/gswangmd/lede/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/ -name config.guess | xargs -r -n1 cp --remove-destination /home/gswangmd/lede/scripts/config.guess; find /home/gswangmd/lede/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/ -name config.sub | xargs -r chmod u+w; find /home/gswangmd/lede/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/ -name config.sub | xargs -r -n1 cp --remove-destination /home/gswangmd/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/gswangmd/lede/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable=wolfssl-4.8.1-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -fomit-frame-pointer -flto -DFP_MAX_BITS=8192 -DWOLFSSL_ALT_CERT_CHAINS" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/gswangmd/lede/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable=wolfssl-4.8.1-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -fomit-frame-pointer -flto -DFP_MAX_BITS=8192 -DWOLFSSL_ALT_CERT_CHAINS " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/gswangmd/lede/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable=wolfssl-4.8.1-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -fomit-frame-pointer -flto -DFP_MAX_BITS=8192 -DWOLFSSL_ALT_CERT_CHAINS " CPPFLAGS="-I/home/gswangmd/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/gswangmd/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/gswangmd/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/gswangmd/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/gswangmd/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro -flto "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  --enable-reproducible-build --enable-lighty --enable-opensslall --enable-opensslextra --enable-sni --enable-stunnel --disable-crypttests --disable-examples --disable-jobserver --enable-ipv6 --disable-aesccm --disable-certgen --disable-chacha --disable-poly1305 --disable-dh --disable-arc4 --disable-tlsv10 --disable-tls13 --disable-session-ticket --disable-dtls --disable-curve25519 --disable-afalg --disable-openvpn --enable-devcrypto=no ; fi; )
sed -i -e 's,^soname_spec=.*,soname_spec="\\${libname}\\${shared_ext}.4.8.1.9c91d6fa",' -e 's,^library_names_spec=.*,library_names_spec="\\${libname}\\${shared_ext}.4.8.1.9c91d6fa \\${libname}\\${shared_ext}",' /home/gswangmd/lede/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/libtool
sed: can't read /home/gswangmd/lede/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/libtool: No such file or directory
make[3]: *** [Makefile:132: /home/gswangmd/lede/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/.configured_9c91d6fab8bdec71fcc929050cfe99da] Error 2
make[3]: Leaving directory '/home/gswangmd/lede/package/libs/wolfssl'
time: package/libs/wolfssl/compile#2.56#0.51#7.05
    ERROR: package/libs/wolfssl failed to build.
make[2]: *** [package/Makefile:116: package/libs/wolfssl/compile] Error 1
make[2]: Leaving directory '/home/gswangmd/lede'
make[1]: *** [package/Makefile:110: /home/gswangmd/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gswangmd/lede'
make: *** [/home/gswangmd/lede/include/toplevel.mk:230: world] Error 2
gswangmd@DESKTOP-HP:~/lede$

dengchan0416, 2021-11-30:
我也是遇到同样问题

WYC-2020, 2021-11-30:
万能大法试过了？

gswangmd-gssi, 2021-11-30:
不知万能大法，请赐教！

SterlingCarter, 2021-11-30:
> 万能大法试过了？

大神帮我回答一下这个主题帖的问题吧，我也是卡在这里，都搞了一个多周了，依旧不能解决，谢谢
[https://github.com/coolsnowwolf/lede/issues/8347](url)

closed 2021-11-30 by gswangmd-gssi

zhoucheng45, 2021-12-20:
> 万能大法试过了？

大佬，万能大法指的是啥方法？