coolsnowwolf/lede #8355: 急急急！！！编译总是报错，卡在ffmpeg这里，无法进行编译，在线求解决。

SterlingCarter, 2021-11-30:
最近编译小米R3P路由器的OpenWrt的固件，总是卡在ffmpeg这里，实在是不知道怎么解决，请问大神们该怎么解决呢？有大佬支下招么？
编译一个固件真的是各种问题都会遇到，搞得身心疲惫，请大神帮忙解决一下，谢谢。

mipsel-openwrt-linux-musl-gcc -Llibavcodec -Llibavdevice -Llibavfilter -Llibavformat -Llibavresaml/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/home/angel/lede/staging_dir/toWl,--as-needed -Wl,-z,noexecstack -Wl,--warn-common -Wl,-rpath-link=:libpostproc:libswresample:liresample -o ffmpeg_g fftools/ffmpeg_opt.o fftools/ffmpeg_filter.o fftools/ffmpeg_hw.o fftools/c-lswresample -lswscale -lavutil -lm -L/home/angel/lede/staging_dir/target-mipsel_24kc_musl/usr/lt-mipsel_24kc_musl/usr/lib -lz -L/home/angel/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lgsl/usr/lib -lz -L/home/angel/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lfdk-aac -L/home/agel/lede/staging_dir/target-mipsel_24kc_musl/usr/lib -lm -lshine -lm -lm -pthread -lm
/home/angel/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-muvformat.so: undefined reference to __atomic_fetch_sub_8' /home/angel/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-muvformat.so: undefined reference to __atomic_store_8'
/home/angel/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-muvformat.so: undefined reference to __atomic_load_8' /home/angel/lede/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-muvformat.so: undefined reference to __atomic_fetch_add_8'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:124: ffmpeg_g] Error 1
make[4]: Leaving directory '/home/angel/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg
make[3]: *** [Makefile:751: /home/angel/lede/build_dir/target-mipsel_24kc_musl/ffmpeg-full/ffmpeg
make[3]: Leaving directory '/home/angel/lede/feeds/packages/multimedia/ffmpeg'
time: package/feeds/packages/ffmpeg/full/compile#652.47#53.77#713.96
ERROR: package/feeds/packages/ffmpeg failed to build (build variant: full).
make[2]: *** [package/Makefile:116: package/feeds/packages/ffmpeg/compile] Error 1
make[2]: Leaving directory '/home/angel/lede'
make[1]: *** [package/Makefile:110: /home/angel/lede/staging_dir/target-mipsel_24kc_musl/stamp/.p
make[1]: Leaving directory '/home/angel/lede'
make: *** [/home/angel/lede/include/toplevel.mk:230：world] 错误 2
[image]

Dimitri2020007, 2021-11-30:
x86-64在线编译同样卡在ffmpeg这块了

WYC-2020, 2021-11-30:
卡什么卡明显就是出错了，原子操作函数没找到定义

WYC-2020, 2021-11-30:
@SterlingCarter 这个编译什么库需要用到这个嘛，明天我也编译看下

WYC-2020, 2021-12-01:
@SterlingCarter  我这边编译没问题
[image]

SterlingCarter, 2021-12-01:
> @SterlingCarter 我这边编译没问题 [image]

请问大神你的ffmpeg版本是多少？4.4么？我之前就是用源码4.4的版本会报错，但是我把\lede\feeds\packages\multimedia\ffmpeg路径下的ffmpeg替换成去年编译时生成的3.4.1版本的文件就不在报错

WYC-2020, 2021-12-01:
@SterlingCarter 4.4

SterlingCarter, 2021-12-01:
> @SterlingCarter 这个编译什么库需要用到这个嘛，明天我也编译看下

我配置的时候并没有选择ffmpeg相关的配置，我看menuconfig 图形界面里面也没有选择，但是他就是会一直会在这里报错

WYC-2020, 2021-12-01:
那是其他库依赖ffmpeg @SterlingCarter

SterlingCarter, 2021-12-01:
> 那是其他库依赖ffmpeg @SterlingCarter

昨天我就把\lede\feeds\packages\multimedia\ffmpeg路径下的ffmpeg替换成去年编译时生成的3.4.1版本，然后就不在报错，最后辨已完成，但是编译完成的固件刷入小米路由器R3P又提示“不支持所上传的映像文件格式，请选择适合当前平台的通用镜像文件”，，配置里面的芯片型号都是选对了的，就是不能刷固件，请问大佬是怎么回事呢？
[image]
[image]
[image]

Neverends4, 2021-12-01:
menuconfig 选中某项后点HELP
可以看selected by 就是被哪个选中的
ffmpeg在multimedia下自己看

SterlingCarter, 2021-12-01:
> menuconfig 选中某项后点HELP 可以看selected by 就是被哪个选中的 ffmpeg在multimedia下自己看

multimedia下并没有选ffmpeg这个插件

closed 2022-02-12 by Beginner-Go