coolsnowwolf/lede #8356: 升级 wolfssl 后编译失败

deplives, 2021-11-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
wolfssl 编译失败

### (2) 路由器型号和固件版本

A：
x86

### (3) 详细日志

A：
```
2021-11-29T20:50:43.1653466Z configure.ac:35: the top level
2021-11-29T20:50:43.1658908Z autom4te: /workdir/openwrt/staging_dir/host/bin/m4 failed with exit status: 63
2021-11-29T20:50:43.1686392Z aclocal.real: error: echo failed with exit status: 63
2021-11-29T20:50:43.1855889Z autoreconf: /workdir/openwrt/staging_dir/host/bin/aclocal failed with exit status: 63
2021-11-29T20:50:43.1897806Z (cd /workdir/openwrt/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/./; if [ -x ./configure ]; then find /workdir/openwrt/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/ -name config.guess | xargs -r chmod u+w; find /workdir/openwrt/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/ -name config.guess | xargs -r -n1 cp --remove-destination /workdir/openwrt/scripts/config.guess; find /workdir/openwrt/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/ -name config.sub | xargs -r chmod u+w; find /workdir/openwrt/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/ -name config.sub | xargs -r -n1 cp --remove-destination /workdir/openwrt/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="ccache_cc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable=wolfssl-4.8.1-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -fomit-frame-pointer -flto -DFP_MAX_BITS=8192 -DWOLFSSL_ALT_CERT_CHAINS" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable=wolfssl-4.8.1-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -fomit-frame-pointer -flto -DFP_MAX_BITS=8192 -DWOLFSSL_ALT_CERT_CHAINS " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable=wolfssl-4.8.1-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -fomit-frame-pointer -flto -DFP_MAX_BITS=8192 -DWOLFSSL_ALT_CERT_CHAINS " CPPFLAGS="-I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro -flto "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  --enable-reproducible-build --enable-lighty --enable-opensslall --enable-opensslextra --enable-sni --enable-stunnel --disable-crypttests --disable-examples --disable-jobserver --enable-ipv6 --disable-aesccm --disable-certgen --disable-chacha --disable-poly1305 --disable-dh --disable-arc4 --disable-tlsv10 --disable-tls13 --disable-session-ticket --disable-dtls --disable-curve25519 --disable-afalg --disable-openvpn --enable-devcrypto=no ; fi; )
2021-11-29T20:50:43.1931678Z sed -i -e 's,^soname_spec=.*,soname_spec="\\${libname}\\${shared_ext}.4.8.1.9c91d6fa",' -e 's,^library_names_spec=.*,library_names_spec="\\${libname}\\${shared_ext}.4.8.1.9c91d6fa \\${libname}\\${shared_ext}",' /workdir/openwrt/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/libtool
2021-11-29T20:50:43.1959921Z sed: can't read /workdir/openwrt/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/libtool: No such file or directory
2021-11-29T20:50:43.1963911Z make[3]: *** [Makefile:132: /workdir/openwrt/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/.configured_9c91d6fab8bdec71fcc929050cfe99da] Error 2
2021-11-29T20:50:43.1969877Z make[3]: Leaving directory '/workdir/openwrt/package/libs/wolfssl'
2021-11-29T20:50:43.1973954Z time: package/libs/wolfssl/compile#2.95#0.19#4.11
2021-11-29T20:50:43.1975188Z     ERROR: package/libs/wolfssl failed to build.
2021-11-29T20:50:43.1992827Z make[2]: *** [package/Makefile:116: package/libs/wolfssl/compile] Error 1
2021-11-29T20:50:43.2001824Z make[2]: Leaving directory '/workdir/openwrt'
2021-11-29T20:50:43.2022288Z make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2021-11-29T20:50:43.2041368Z make[1]: Leaving directory '/workdir/openwrt'
2021-11-29T20:50:43.2052347Z make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
2021-11-29T20:50:43.2090957Z ##[error]Process completed with exit code 2.

```

martin0607, 2021-11-30:
一样一样

harleywong, 2021-11-30:
好像项目Actions在线编译也有wolfssl的错误，今天直接没成功。

martin0607, 2021-11-30:
先换回旧版的吧

github-xihe, 2021-11-30:
同样问题

zhb7670, 2021-11-30:
5h 9m 56s
(cd /workdir/openwrt/build_dir/target-mipsel_24kc_musl/wolfssl-4.8.1-stable/./; if [ -x ./configure ]; then find /workdir/openwrt/build_dir/target-mipsel_24kc_musl/wolfssl-4.8.1-stable/ -name config.guess | xargs -r chmod u+w; find /workdir/openwrt/build_dir/target-mipsel_24kc_musl/wolfssl-4.8.1-stable/ -name config.guess | xargs -r -n1 cp --remove-destination /workdir/openwrt/scripts/config.guess; find /workdir/openwrt/build_dir/target-mipsel_24kc_musl/wolfssl-4.8.1-stable/ -name config.sub | xargs -r chmod u+w; find /workdir/openwrt/build_dir/target-mipsel_24kc_musl/wolfssl-4.8.1-stable/ -name config.sub | xargs -r -n1 cp --remove-destination /workdir/openwrt/scripts/config.sub; AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/wolfssl-4.8.1-stable=wolfssl-4.8.1-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -fomit-frame-pointer -flto -DFP_MAX_BITS=8192 -DWOLFSSL_ALT_CERT_CHAINS" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/wolfssl-4.8.1-stable=wolfssl-4.8.1-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -fomit-frame-pointer -flto -DFP_MAX_BITS=8192 -DWOLFSSL_ALT_CERT_CHAINS " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/wolfssl-4.8.1-stable=wolfssl-4.8.1-stable -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DPIC -fpic -fomit-frame-pointer -flto -DFP_MAX_BITS=8192 -DWOLFSSL_ALT_CERT_CHAINS " CPPFLAGS="-I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro -flto "   ./configure --target=mipsel-openwrt-linux --host=mipsel-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  --enable-reproducible-build --enable-lighty --enable-opensslall --enable-opensslextra --enable-sni --enable-stunnel --disable-crypttests --disable-examples --disable-jobserver --enable-ipv6 --disable-aesccm --disable-certgen --disable-chacha --disable-poly1305 --disable-dh --disable-arc4 --disable-tlsv10 --disable-tls13 --disable-session-ticket --disable-dtls --disable-curve25519 --disable-afalg --disable-openvpn --enable-devcrypto=no ; fi; )
sed -i -e 's,^soname_spec=.*,soname_spec="\\${libname}\\${shared_ext}.4.8.1.9c91d6fa",' -e 's,^library_names_spec=.*,library_names_spec="\\${libname}\\${shared_ext}.4.8.1.9c91d6fa \\${libname}\\${shared_ext}",' /workdir/openwrt/build_dir/target-mipsel_24kc_musl/wolfssl-4.8.1-stable/libtool
sed: can't read /workdir/openwrt/build_dir/target-mipsel_24kc_musl/wolfssl-4.8.1-stable/libtool: No such file or directory
make[3]: *** [/workdir/openwrt/build_dir/target-mipsel_24kc_musl/wolfssl-4.8.1-stable/.configured_9c91d6fab8bdec71fcc929050cfe99da] Error 2
Makefile:127: recipe for target '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/wolfssl-4.8.1-stable/.configured_9c91d6fab8bdec71fcc929050cfe99da' failed
make[3]: Leaving directory '/workdir/openwrt/package/libs/wolfssl'
time: package/libs/wolfssl/compile#2.96#0.18#4.10
    ERROR: package/libs/wolfssl failed to build.
make[2]: *** [package/libs/wolfssl/compile] Error 1
package/Makefile:114: recipe for target 'package/libs/wolfssl/compile' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:110: recipe for target '/workdir/openwrt/staging_dir/tar

zhb7670, 2021-11-30:
干掉wol(网络唤醒)就好了！原始路径不能访问

VergilGao, 2021-11-30:
看了下代码，这里的wolfssl和openwrt同步了，4.8.1的wolfssl我已经编译了好几个版本都没什么问题，最好还是排查下有没有别的插件和他有冲突。

WYC-2020, 2021-11-30:
什么库依赖他了？我按我需要的库选择了编译没问题勒，大部分库我都选了

felixxiao99, 2021-11-30:
同样问题

WYC-2020, 2021-12-01:
一切正常，昨晚十点才编译的

WYC-2020, 2021-12-01:
你们怕是gcc版本太低了，没有相关的函数吧，自己更改gcc版本到11，我是用的gcc11 编译

VergilGao, 2021-12-01:
> 

他们的报错都是没找到 ../wolfssl-4.8.1-stable/libtool 文件夹，这就不知道是哪里的原因了，也许是他们用的SDK缺少某些工具？

WYC-2020, 2021-12-01:
谁知道反正我这里没问题，如果大部分没问题，大雕都会无视关闭的，肯定是那里姿势不对

WYC-2020, 2021-12-01:
@VergilGao 4.8.1 libtool 本来就不是文件夹，是文件，4.7才是文件夹

Gzxhwq, 2021-12-01:
libtool的问题在这个Commit之后就正常了 ，实测编译能通过了
https://github.com/coolsnowwolf/lede/commit/f571a6bf66ee87c88bfa2b95672d0ca695c1eabb

WYC-2020, 2021-12-01:
看来要么是没有及时更新要么是组合怪组合有问题导致没法编译 @Gzxhwq

waynesg, 2021-12-01:
actions把缓存关了

closed 2021-12-07 by AmadeusGhost