coolsnowwolf/lede #8357: 最近版本编译的 .img 正常吗？

renfei, 2021-11-30:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

有类似，但未解决我的问题：#6480

## 2. 详细叙述

各位技术大佬，最近版本的编译 .img 正常吗？我自己编译的和下载 GitHub Actions 的包都会写入失败。

### (1) 具体问题

A：.img 写盘失败

#### physdiskwrite

报错：writtenWrite error after XXX bytes

#### Roadkil's DiskImage

报错：Error #0 occured while writting to disk at sector XXX

#### dd if= of=

启动报错：XZ-Compressed data is corrupt

### (2) 路由器型号和固件版本

A：x86

WYC-2020, 2021-11-30:
哪个玩意不用在意 因为img包末尾有那个签名，再磁盘写入的时候 会出现不是512字节对齐，去掉签名就不会出现哪个错误，或者忽略

renfei, 2021-12-01:
> 哪个玩意不用在意 因为img包末尾有那个签名，再磁盘写入的时候 会出现不是512字节对齐，去掉签名就不会出现哪个错误，或者忽略

这位老师说的是对的，即使报错了，也是可以正常使用的，如果也遇到了这个问题可以参考下。我是将 `Root filesystem partition size` 调大到了 512MB 了，然后虽然显示“写入失败”，但是可以正常使用：
[image]
[image]

closed 2021-12-01 by renfei

Jefffish09, 2022-03-24:
> 哪个玩意不用在意 因为img包末尾有那个签名，再磁盘写入的时候 会出现不是512字节对齐，去掉签名就不会出现哪个错误，或者忽略

想请教下，我编译出来.img.gz，在win下用7zip解压失败，会提示“显示有效数据外包含额外数据”。
在ubuntu用gizp命令解压，能解压出img文件，但会提示“decompression OK, trailing garbage ignored”，这样正常吗？能不能用这个img来刷机？怕出问题

Jennal, 2022-07-14:
> > 哪个玩意不用在意 因为img包末尾有那个签名，再磁盘写入的时候 会出现不是512字节对齐，去掉签名就不会出现哪个错误，或者忽略
> 
> @WYC-2020 想请教下，我编译出来.img.gz，在win下用7zip解压失败，会提示“显示有效数据外包含额外数据”。 在ubuntu用gizp命令解压，能解压出img文件，但会提示“decompression OK, trailing garbage ignored”，这样正常吗？能不能用这个img来刷机？怕出问题

没问题的，我就是用windows下用7z解压出错的img，刷写正常，并且正常使用。