coolsnowwolf/lede #8364: Turbo ACC里的DNS 缓存会导致无线主机名变成bogon

fengqi, 2021-11-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

最近一次升级是10月19号，后面惯例升级了两次，发现wifi的主机名变成了bogon，但是在DHCP/DNS显示正常，一直没当回事，昨天又升级了一次导致部分设备的无线速率全部变成了6.0 Mbit/s，回滚到10月27号的版本速率正常，继续回滚到10月19号的版本发现无线主机名也正常了。

目前在用10月27号的版本，对比了19号和27号的manifest发现，多了dnsproxy和dnsforwarder两个包，卸载后发现无线的部分主机名显示正常，部分为空。但是同时Turbo ACC也被卸载了，重新装回来后，关闭DNS缓存，暂时保持部分正常部分不正确的状态。

### (2) 路由器型号和固件版本

ASUS RT-ACRH17
git-21.329.83689-00cb491

### (3) 详细日志

没有发现报错。

ZGNCZMY, 2021-11-30:
distclean

WYC-2020, 2021-12-01:
这东西我看10月11号更新后在没更新了，应该不是这个东西惹的祸吧

fengqi, 2021-12-01:
> 这东西我看10月11号更新后在没更新了，应该不是这个东西惹的祸吧

我也不确定，不过当时没想到是dns缓存的包就直接卸载了，然后发现加速相关也被卸载了，再看Wi-Fi主机名就能显示一部分了。

fengqi, 2021-12-01:
我可以晚上回去趁没人使用Wi-Fi的时候再次回滚到19号的，并且开启dns缓存看看是否能全部显示主机名

fengqi, 2021-12-01:
27号的版本，实际上最新代码是到26号：https://github.com/coolsnowwolf/lede/commits/master?after=974cee715d8433660d007041238c38e3cf642fa0+1&branch=master
19号的版本，实际上最带代码是到9月22号的：https://github.com/coolsnowwolf/lede/commits/master?after=69b0f1cf689817cb2e77cca8f4ae8ea4c5d4a256+1&branch=master

WYC-2020, 2021-12-04:
这个要么自己改下，或者去看下lienol的op，他的好像吧这个功能直接裁剪掉了，dns这块。自己改的话也简单

closed 2021-12-25 by fengqi