coolsnowwolf/lede #8371: ipv6导致每30分钟重启防火墙

kkfuzi, 2021-12-01:
请教个问题。用的是第三方编译的openwrt，配合openclash，一直正常使用。

9月份开始新的固件发现个问题，openclash会比较精准的每隔30分钟自动重置插件防火墙（见下图），间接导致断流。

开始怀疑是openclash的问题，排查了一下，9月份之前的固件用不同的oc版本都是好的，9月份之后的固件用不同的oc版本都有问题。向openclash方也提过，回应openclash的重置可能是因为系统防火墙重置而间接导致。

自己排查了插件和节点链接，没找到原因。

我用的是软路由PPPoE拨号，今天尝试把WAN口的获取 IPv6 地址禁用，关闭公网v6 IP，问题就解决了。但相同网络相同配置9月份的固件启用v6一切正常。

想问一下，有谁碰到过类似的问题，或者说是9月份之后的某些设置升级导致的这个问题？想问一下有没有保留v6开启解决这个问题的方法，谢谢。
[image]

kagurareina, 2022-01-25:
同问，我这里的状况是ipv6每拨号30分钟必定闪断一下。放行wan所有流量也断。令人费解啊为什么

zepingwu, 2022-09-19:
我也是，不过我不是三十分钟，我是每过几秒就会，而且在重启系统6天之后，也就是运营商断我两次连接就会出现这个问题。

xuan-wei, 2022-09-28:
同样的问题，有大佬解决下嘛

trotack, 2022-10-21:
我近期也遇到了，之前都一直很正常。ipv6用的也都正常，就这几天开始出现了，好神奇

zepingwu, 2022-10-21:
我已经换原版openwrt了，很稳。

hcym, 2022-10-21:
这个应该是需要nft的，大雕还是默认老的

trotack, 2022-10-21:
https://zhiliao.h3c.com/Theme/details/55576
不知道是不是这个的原因

kkfuzi, 2024-06-10:
问题解决了。。。https://github.com/vernesong/OpenClash/issues/3471
开源社区就是赞

coolsnowwolf, 2026-09-05:
该 issue 创建于 2021 年，问题发生在第三方固件 + OpenClash + IPv6 组合；评论中报告者已明确表示“问题解决了”，并链接 OpenClash issue #3471。没有提供 LEDE 当前代码可复现的日志或补丁需求，属于已由上游插件侧解决的历史问题，关闭为已完成。

closed 2026-09-05 by coolsnowwolf