coolsnowwolf/lede #8375: 有海外的源吗，我在国外这个mirror.cloud.tencent访问很慢

KrisCris, 2021-12-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：有海外的源吗，我在国外这个mirror.cloud.tencent访问很慢, 老下载不了

### (2) 路由器型号和固件版本

A：xiaomi rm2100

renfei, 2021-12-02:
这个不是项目代码的事儿吧？这个貌似应该是你使用的云主机配置的事儿，好像不归这个项目管。
在境外用官方的就行了吧，Ubuntu应该是修改 `/etc/apt/sources.list`

KrisCris, 2021-12-02:
哦，我的意思是，他opkg的那个源，是腾讯的镜像。我想直接用openwrt的，但是发现直接复制过来好像不太行

renfei, 2021-12-02:
哦哦，是我理解错了，你看我找的这个对应关系对不对：
https://mirrors.cloud.tencent.com/lede/snapshots/targets/x86/64/
https://downloads.openwrt.org/snapshots/targets/x86/64/

KrisCris, 2021-12-02:
谢谢我试试看
我把他们改成：
```
src/gz openwrt_core https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/packages
src/gz openwrt_base https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base
src/gz openwrt_luci https://downloads.openwrt.org/releases/18.06.9/packages/mipsel_24kc/luci
src/gz openwrt_packages https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages
src/gz openwrt_routing https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing
src/gz openwrt_telephony https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony
```
了

KrisCris, 2021-12-02:
但这样改了之后我发现我装东西会遇到一个问题。。他跟我说
`cannot find dependency kernel (= 5.4.162-1-8d349977f35216b2fbea1308d622cbe5)`

renfei, 2021-12-02:
好像是不同编译的版本用的 Linux 内核版本号不同，snapshots 是开发快照每天自动构建，我猜得指定到某个版本，得跟你的固件的 Linux 内核对应上，再详细的就得技术大佬来解答了

KrisCris, 2021-12-02:
嗯嗯

WYC-2020, 2021-12-03:
这个我记得esir有讲解到这个kernel不配备的问题

closed 2021-12-07 by AmadeusGhost