coolsnowwolf/lede #8378: 请移除docker！编译时加入docker会使端口转发失效！

realJustinLee, 2021-12-02:
请移除docker！编译时加入docker会使端口转发失效！

realJustinLee, 2021-12-02:
> 不喜欢自己编译的时候diy就是了，为啥要求你不适合就删除

端口转发是个普遍使用的功能，如果你对docker有特殊需求，应该在你自己的firmware里加入docker支持，而不是ruin所有人的使用体验。并浪费大家的时间去debug，最后发现新版固件不工作是docker导致的。

WYC-2020, 2021-12-02:
那就慢慢等吧，我无所谓，我记得前不久才有一个说不要集成docker来的，哈哈

realJustinLee, 2021-12-02:
> 那就慢慢等吧，我无所谓，我记得前不久才有一个说不要集成docker来的，哈哈

有什么好等的呢？大不了自己提PR呗。弄得似乎这个repo是您管一样。

tonyk520, 2021-12-02:
可以自己取消，luci-application取消luci-app-dockerman，Utilities中取消containerd、docker、docker-compose、dockerd、libnetwork（也许）

WYC-2020, 2021-12-02:
@realJustinLee 说话这么冲，你牛逼，爱咋咋的

deplives, 2021-12-02:
请慎用感叹号，如果你发现了问题，可以选择自己修复，也可以选择提交PR，而不是在issues里面用命令的口吻去要求别人，你说端口转发是个普遍使用的功能，那是对你。你需要，那就自行取消docker，还有很多人需要docker而不需要端口转发。

realJustinLee, 2021-12-02:
@WYC-2020 @deplives fine, 各位真的是高高在上，一时间我居然忘了我等开民我说话还要跪下。惹不起，大人，磕几个头啊？

realJustinLee, 2021-12-02:
@tonyk520 其实也不不必这么麻烦，参考commit 2672bfdc9d34a1cf308222c05fd91c1bedb7d3ed 做一下revert就好。改动 target/linux/x86/Makefile，删掉dockerman，在make defconfig的时候会自动构建依赖。

EnnawYang, 2021-12-02:
> @WYC-2020 @deplives fine, 各位真的是高高在上，一时间我居然忘了我等开民我说话还要跪下。惹不起，大人，磕几个头啊？

难不成是不让你取消了？docker难道是写死进去的？

EnnawYang, 2021-12-02:
> @EnnawYang 惹不起，惹不起，睾贵的大人。怎么？我等开民连表达不满的自由都没有了吗？

请问是没给您取消选择的自由吗？ 还是本库源码不自由了？ 不开源是吗？ 本项目收您使用费了吗？ 自己不会取消赖别人是吧？

closed 2021-12-02 by AmadeusGhost