coolsnowwolf/lede #8381: X86软路由中端口转发，部分有效，部分失效

Nothend, 2021-12-03:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：大雕 你好 冒昧打扰一下 ，想请教个问题，x86 最新固件，端口转发问题，路由器自身的转发可以使用，但无法端口转发到内网其他机器上，比如作为ap的TPLINK路由器
[image]
[image]
[image]

### (2) 路由器型号和固件版本

A：X86软路由，OpenWrt R21.12.1 / LuCI Master (git-21.329.83689-00cb491)

### (3) 详细日志

A：Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:18 2021 daemon.err uhttpd[3567]: Dump terminated
Fri Dec  3 08:39:40 2021 daemon.err uhttpd[3567]: lsblk: write error
Fri Dec  3 08:39:40 2021 daemon.err uhttpd[3567]: /usr/lib/lua/luci/dispatcher.lua:474: Failed to execute call dispatcher target for entry '/admin/system/diskman/get_disk_info/sda'.
Fri Dec  3 08:39:40 2021 daemon.err uhttpd[3567]: The called action terminated with an exception:
Fri Dec  3 08:39:40 2021 daemon.err uhttpd[3567]: /usr/lib/lua/luci/model/diskman.lua:190: attempt to index a nil value
Fri Dec  3 08:39:40 2021 daemon.err uhttpd[3567]: stack traceback:
Fri Dec  3 08:39:40 2021 daemon.err uhttpd[3567]: 	[C]: in function 'assert'
Fri Dec  3 08:39:40 2021 daemon.err uhttpd[3567]: 	/usr/lib/lua/luci/dispatcher.lua:474: in function 'dispatch'
Fri Dec  3 08:39:40 2021 daemon.err uhttpd[3567]: 	/usr/lib/lua/luci/dispatcher.lua:126: in function </usr/lib/lua/luci/dispatcher.lua:125>
Fri Dec  3 08:43:04 2021 kern.err kernel: [65257.102380] vmxnet3 0000:04:00.0 eth0: tx hang
Fri Dec  3 08:43:04 2021 kern.notice kernel: [65257.102750] vmxnet3 0000:04:00.0 eth0: resetting
Fri Dec  3 08:43:04 2021 kern.info kernel: [65257.107967] vmxnet3 0000:04:00.0 eth0: intr type 3, mode 0, 5 vectors allocated
Fri Dec  3 08:43:04 2021 kern.info kernel: [65257.109138] vmxnet3 0000:04:00.0 eth0: NIC Link is Up 10000 Mbps
Fri Dec  3 08:47:23 2021 kern.err kernel: [65515.896451] vmxnet3 0000:04:00.0 eth0: tx hang
Fri Dec  3 08:47:23 2021 kern.notice kernel: [65515.896772] vmxnet3 0000:04:00.0 eth0: resetting
Fri Dec  3 08:47:23 2021 kern.info kernel: [65515.902000] vmxnet3 0000:04:00.0 eth0: intr type 3, mode 0, 5 vectors allocated
Fri Dec  3 08:47:23 2021 kern.info kernel: [65515.903174] vmxnet3 0000:04:00.0 eth0: NIC Link is Up 10000 Mbps
Fri Dec  3 08:52:06 2021 kern.err kernel: [65799.012454] vmxnet3 0000:04:00.0 eth0: tx hang
Fri Dec  3 08:52:06 2021 kern.notice kernel: [65799.012774] vmxnet3 0000:04:00.0 eth0: resetting
Fri Dec  3 08:52:06 2021 kern.info kernel: [65799.017947] vmxnet3 0000:04:00.0 eth0: intr type 3, mode 0, 5 vectors allocated
Fri Dec  3 08:52:06 2021 kern.info kernel: [65799.019129] vmxnet3 0000:04:00.0 eth0: NIC Link is Up 10000 Mbps
Fri Dec  3 09:04:30 2021 kern.err kernel: [66542.881640] vmxnet3 0000:04:00.0 eth0: tx hang
Fri Dec  3 09:04:30 2021 kern.notice kernel: [66542.881967] vmxnet3 0000:04:00.0 eth0: resetting
Fri Dec  3 09:04:30 2021 kern.info kernel: [66542.887701] vmxnet3 0000:04:00.0 eth0: intr type 3, mode 0, 5 vectors allocated
Fri Dec  3 09:04:30 2021 kern.info kernel: [66542.888923] vmxnet3 0000:04:00.0 eth0: NIC Link is Up 10000 Mbps
Fri Dec  3 09:08:51 2021 kern.err kernel: [66803.983680] vmxnet3 0000:04:00.0 eth0: tx hang
Fri Dec  3 09:08:51 2021 kern.notice kernel: [66803.983998] vmxnet3 0000:04:00.0 eth0: resetting
Fri Dec  3 09:08:51 2021 kern.info kernel: [66803.989785] vmxnet3 0000:04:00.0 eth0: intr type 3, mode 0, 5 vectors allocated
Fri Dec  3 09:08:51 2021 kern.info kernel: [66803.991034] vmxnet3 0000:04:00.0 eth0: NIC Link is Up 10000 Mbps
Fri Dec  3 09:12:22 2021 kern.err kernel: [67014.905961] vmxnet3 0000:04:00.0 eth0: tx hang
Fri Dec  3 09:12:22 2021 kern.notice kernel: [67014.906282] vmxnet3 0000:04:00.0 eth0: resetting
Fri Dec  3 09:12:22 2021 kern.info kernel: [67014.911961] vmxnet3 0000:04:00.0 eth0: intr type 3, mode 0, 5 vectors allocated
Fri Dec  3 09:12:22 2021 kern.info kernel: [67014.913201] vmxnet3 0000:04:00.0 eth0: NIC Link is Up 10000 Mbps
Fri Dec  3 09:17:15 2021 user.err ddns-scripts[9277]: ali: wget 错误代码: '8'
Fri Dec  3 09:17:16 2021 user.warn ddns-scripts[9277]: ali: 传输失败 - 1/0 在 60 秒后重试
Fri Dec  3 09:28:00 2021 kern.err kernel: [67952.811240] vmxnet3 0000:04:00.0 eth0: tx hang
Fri Dec  3 09:28:00 2021 kern.notice kernel: [67952.811592] vmxnet3 0000:04:00.0 eth0: resetting
Fri Dec  3 09:28:00 2021 kern.info kernel: [67952.817528] vmxnet3 0000:04:00.0 eth0: intr type 3, mode 0, 5 vectors allocated
Fri Dec  3 09:28:00 2021 kern.info kernel: [67952.818740] vmxnet3 0000:04:00.0 eth0: NIC Link is Up 10000 Mbps
Fri Dec  3 09:45:53 2021 kern.err kernel: [69025.616223] vmxnet3 0000:04:00.0 eth0: tx hang
Fri Dec  3 09:45:53 2021 kern.notice kernel: [69025.616557] vmxnet3 0000:04:00.0 eth0: resetting
Fri Dec  3 09:45:53 2021 kern.info kernel: [69025.622679] vmxnet3 0000:04:00.0 eth0: intr type 3, mode 0, 5 vectors allocated
Fri Dec  3 09:45:53 2021 kern.info kernel: [69025.623899] vmxnet3 0000:04:00.0 eth0: NIC Link is Up 10000 Mbps
Fri Dec  3 09:55:36 2021 kern.err kernel: [69608.735318] vmxnet3 0000:04:00.0 eth0: tx hang
Fri Dec  3 09:55:36 2021 kern.notice kernel: [69608.735656] vmxnet3 0000:04:00.0 eth0: resetting
Fri Dec  3 09:55:36 2021 kern.info kernel: [69608.741495] vmxnet3 0000:04:00.0 eth0: intr type 3, mode 0, 5 vectors allocated
Fri Dec  3 09:55:36 2021 kern.info kernel: [69608.742725] vmxnet3 0000:04:00.0 eth0: NIC Link is Up 10000 Mbps

gudumuyu, 2021-12-04:
出现同样的问题，部分端口转发失败，然后我把docker去掉重新编译后，就没问题了

Nothend, 2021-12-05:
> 

谢谢提醒，你说的docker 是luci application 里面的吗？在moudles里面有个dockerd 这个取消不掉，这个有影响吗？

Nothend, 2021-12-05:
> 出现同样的问题，部分端口转发失败，然后我把docker去掉重新编译后，就没问题了

谢谢提醒，你说的docker 是luci application 里面的吗？在moudles里面有个dockerd 这个取消不掉，这个有影响吗？

gudumuyu, 2021-12-05:
> > 出现同样的问题，部分端口转发失败，然后我把docker去掉重新编译后，就没问题了
> 
> 谢谢提醒，你说的docker 是luci application 里面的吗？在moudles里面有个dockerd 这个取消不掉，这个有影响吗？

我是把Utilities里面的docker取消勾选的

Nothend, 2021-12-05:
> 

> > > 出现同样的问题，部分端口转发失败，然后我把docker去掉重新编译后，就没问题了
> > 
> > 
> > 谢谢提醒，你说的docker 是luci application 里面的吗？在moudles里面有个dockerd 这个取消不掉，这个有影响吗？
> 
> 我是把Utilities里面的docker取消勾选的

好的非常感谢耐心回复，我去试试。顺便问下这算是bug 吗？大雕怎么不修复一下，我之前搜的时候看他回复过类似的解决办法

gudumuyu, 2021-12-05:
> > 
> 
> > > > 出现同样的问题，部分端口转发失败，然后我把docker去掉重新编译后，就没问题了
> > > 
> > > 
> > > 谢谢提醒，你说的docker 是luci application 里面的吗？在moudles里面有个dockerd 这个取消不掉，这个有影响吗？
> > 
> > 
> > 我是把Utilities里面的docker取消勾选的
> 
> 好的非常感谢耐心回复，我去试试。顺便问下这算是bug 吗？大雕怎么不修复一下，我之前搜的时候看他回复过类似的解决办法

这个我就不太了解了

Nothend, 2021-12-05:
> > > > > 出现同样的问题，部分端口转发失败，然后我把docker去掉重新编译后，就没问题了
> > > > 
> > > > 
> > > > 谢谢提醒，你说的docker 是luci application 里面的吗？在moudles里面有个dockerd 这个取消不掉，这个有影响吗？
> > > 
> > > 
> > > 我是把Utilities里面的docker取消勾选的
> > 
> > 
> > 好的非常感谢耐心回复，我去试试。顺便问下这算是bug 吗？大雕怎么不修复一下，我之前搜的时候看他回复过类似的解决办法
> 
> 这个我就不太了解了

好的，谢谢

ZqinKing, 2021-12-09:
设置了源MAC的原因吧。把WAN口上源MAC的条件去了。

82kg, 2021-12-13:
> 设置了源MAC的原因吧。把WAN口上源MAC的条件去了。

源MAC 到底有啥用呢，路由能取到访问者的mac地址？

Nothend, 2021-12-14:
> 设置了源MAC的原因吧。把WAN口上源MAC的条件去了。

[image]

大佬，我本来就没有设置

imkenf, 2022-01-22:
同样的问题，用截至今天的源码编译，只要选dockerman nat无论怎么设置都无效，搜了几篇文章，猜测是不是和docker不能调用iptables等原因

Nothend, 2022-01-22:
> 同样的问题，用截至今天的源码编译，只要选dockerman nat无论怎么设置都无效，搜了几篇文章，猜测是不是和docker不能调用iptables等原因

还是搞不清楚，我目前刚编译了最新版本，连openwrt 自己的端口都无法映射成功

Nothend, 2022-01-23:
> 同样的问题，用截至今天的源码编译，只要选dockerman nat无论怎么设置都无效，搜了几篇文章，猜测是不是和docker不能调用iptables等原因

我在，防火墙，自定义规则，里面加了一条就可以了，你参考下吧。
iptables -t nat -I PREROUTING -i pppoe-WAN -p tcp --dport XXXX -j DNAT --to-destination 10.0.0.1:XXXX

imkenf, 2022-01-24:
> > 同样的问题，用截至今天的源码编译，只要选dockerman nat无论怎么设置都无效，搜了几篇文章，猜测是不是和docker不能调用iptables等原因
> 
> 我在，防火墙，自定义规则，里面加了一条就可以了，你参考下吧。 iptables -t nat -I PREROUTING -i pppoe-WAN -p tcp --dport XXXX -j DNAT --to-destination 10.0.0.1:XXXX

谢谢，晚上找个我重新编译试试。这么看来还是iptables的原因

WYC-2020, 2022-01-24:
其实docker 失效我怀疑是sysctl-br-netfilter-ip.conf这个文件的锅，dockerd加入了这个文件并且net.bridge.bridge-nf-call-ip6tables=1
net.bridge.bridge-nf-call-iptables=1
你们吧他们都改成0试下看看，哈哈

Nothend, 2022-01-24:
> 其实docker 失效我怀疑是sysctl-br-netfilter-ip.conf这个文件的锅，dockerd加入了这个文件并且net.bridge.bridge-nf-call-ip6tables=1 net.bridge.bridge-nf-call-iptables=1 你们吧他们都改成0试下看看，哈哈

非常感谢呢，我来试试

sonic3366, 2022-03-11:
请问这个问题解决了吗，我还是遇到了和楼主一样的问题

Nothend, 2022-03-12:
> 请问这个问题解决了吗，我还是遇到了和楼主一样的问题

结论了，看下面这个链接，要在防火墙里设置一下：https://blog.csdn.net/aplsc/article/details/122684437?utm_source=app&app_version=5.0.0

sonic3366, 2022-03-14:
这个办法之前测试过，不行

Nothend, 2022-03-16:
> > > 请问这个问题解决了吗，我还是遇到了和楼主一样的问题
> > 
> > 
> > 结论了，看下面这个链接，要在防火墙里设置一下：https://blog.csdn.net/aplsc/article/details/122684437?utm_source=app&app_version=5.0.0
> > 这个办法之前测试过，不行

呃，那不好意思了，我这里试了可以

signigelchan, 2022-05-16:
遇到同样的问题，看了 Issues 类似的问题，回复大多说的是 docker 的锅，折腾了好久，刚不久貌似找到了问题，我这边尝试关闭 openclash ，然后就可以了 (openclash 使用 Redit-host (TUN) 模式)。

我这边的环境：

1. 主路由：Newifi D2
2. 旁路由：N1 盒子 (编入了 dockerman、openclash)

Travel2Here, 2023-06-01:
> > 同样的问题，用截至今天的源码编译，只要选dockerman nat无论怎么设置都无效，搜了几篇文章，猜测是不是和docker不能调用iptables等原因
> 
> 我在，防火墙，自定义规则，里面加了一条就可以了，你参考下吧。 iptables -t nat -I PREROUTING -i pppoe-WAN -p tcp --dport XXXX -j DNAT --to-destination 10.0.0.1:XXXX

近期遇到相同问题，也含dockerman，环境基本一致；
这个额外的dnat配置能成功生效！
但有个疑问，我看了系统状态里的防火墙规则记录，那些端口设置明明已经在【链 zone_wan_input】里，其中有的端口能正常，有的就不行（用上面的额外设置才行），很费解。。。不知道有没有哪位哥还能深入研究下。

另，我日常用的passwall2，但启动与否都不影响这个dnat问题。

closed 2026-09-05 by coolsnowwolf