coolsnowwolf/lede #8382: samba 插件 匿名用户 无法新建、修改文件及文件夹

LXY631, 2021-12-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：samba 插件 匿名用户、新建用户 无法新建、修改文件及文件夹，文件权限已给，samba设置如图。

使用官方配置 https://openwrt.org/docs/guide-user/services/nas/samba_configuration#adding_samba_user_s，新建用户无法操作，匿名分享文件仍然无法修改，chown samba_new_user:samba_new_user /mnt/sda1/share/
提示 chown: unknown user/group new_user:samba_new_user。

按照该文设置root用户可使用，https://www.chenyajun.net/index.php/post/65.html
---
drwxrwxrw-    2 root     root       4096 Dec  3 22:51 share

[image]
---

### (2) 路由器型号和固件版本

A：路由器：NanoPi R1；固件：OpenWrt R21.12.1

### (3) 详细日志

A：

WYC-2020, 2021-12-04:
建议你去看esir教程专门又讲这个

LXY631, 2021-12-04:
> 建议你去看 esir 教程专门又讲这个

谢谢回复，再次编译后使用samba4插件，看到有强制root选项，勾选后可以执行读写、删除等操作，应给是权限没有设置好。不过感觉 菜单选项中  创建权限掩码 功能 好像并没有起到什么作用啊，不知道权限这些东西都怎么管理的，还望大佬指点下，引下路，因为喜欢上路由器个人定制所以自己网上瞎琢磨来着，费了不少时间。

LXY631, 2021-12-05:
还是使用samba4版本吧，按照官方的使用说明基本没有问题，不过权限问题一直搞不懂，哎。。。

closed 2021-12-05 by LXY631