coolsnowwolf/lede #8384: Apple Silicon M1 编译出错

maomaoinchina, 2021-12-04:
## 1. Apple M1 based macbookpro 用VMware Fusion M1 Tech review版本出错
Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：基于Ubuntu 20.04.3 LTS for ARM编译

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：[image](https://user-images.githubusercontent.com/16549108/144695310-a5f97324-5f8a-4760-a8a2-bf82332978c8.png)

AmadeusGhost, 2021-12-04:
没支持过 m1 上编译

maomaoinchina, 2021-12-04:
https://github.com/openwrt/packages/issues/12793#issuecomment-657207434 
貌似可以解决 golang for linux arm64的问题

martin0607, 2021-12-04:
silicon编译性能怎么样，打算搞个m1pro玩玩

maomaoinchina, 2021-12-04:
只能说比我之前intel 2.7Ghz的 2016老mbp强太多了。
之前intel单线程风扇呼呼吹；这个M1火力全开，风扇前半程基本不转，后半程也是几乎听不到声音的转速。

closed 2021-12-04 by maomaoinchina

zcc0030, 2022-06-10:
我研究了大半年，终于找到问题所在，已经成功出包了，除了ssr+ 和passwall不能一起编译，naiveproxy不能编译，选插件的时候记得不要选，其他没问题。可以参考我写的教程 https://www.zicaidan.xyz/?m=202206

minfengyk, 2022-10-31:
> 我研究了大半年，终于找到问题所在，已经成功出包了，除了ssr+ 和passwall不能一起编译，naiveproxy不能编译，选插件的时候记得不要选，其他没问题。可以参考我写的教程 https://www.zicaidan.xyz/?m=202206

我按照您的教程，安装arm系统依赖还是报错，请指点，是源的原因吗？
minfeng@ubuntu22:~$ sudo apt install gcc-multilib-arm-linux-gnueabi gcc-multilib-arm-linux-gnueabihf gcc-multilib-i686-linux-gnu gcc-multilib-s390x-linux-gnu gcc-multilib-x86-64-linux-gnu gcc-multilib-x86-64-linux-gnux32
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package gcc-multilib-arm-linux-gnueabi
E: Unable to locate package gcc-multilib-arm-linux-gnueabihf
minfeng@ubuntu22:~$ sudo apt install g++-multilib-arm-linux-gnueabi g++-multilib-arm-linux-gnueabihf g++-multilib-i686-linux-gnu g++-multilib-s390x-linux-gnu g++-multilib-x86-64-linux-gnu g++-multilib-x86-64-linux-gnux32
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package g++-multilib-arm-linux-gnueabi
E: Unable to locate package g++-multilib-arm-linux-gnueabihf

ysuolmai, 2023-04-30:
> 我按照您的教程，安装arm系统依赖还是报错，请指点，是源的原因吗？ minfeng@ubuntu22:~$ sudo apt install gcc-multilib-arm-linux-gnueabi gcc-multilib-arm-linux-gnueabihf gcc-multilib-i686-linux-gnu gcc-multilib-s390x-linux-gnu [...]

我也遇到和你一样的问题，你解决了吗？

我查了下这两个包，不支持22.04的arm，20.04支持，现在paralell desktop默认装的就是22.04的ubuntu。我等下试试找20.04的iso，用iso安装试试。

minfengyk, 2023-05-01:
我后来在paralell desktop上安装20.04和22.04版arm都成功了，也是按教程操作的，可能是代理的原因吧！现附上教程：

M系列芯片电脑上ARM版Ubuntu 22安装依赖教程

一、安装编译所需依赖
       ssh登陆Ubuntu，输入：
sudo apt install -y ack antlr3 aria2 asciidoc autoconf automake autopoint binutils bison build-essential \
bzip2 ccache cmake cpio curl device-tree-compiler fastjar flex gawk gettext \
git gperf haveged help2man intltool libelf-dev libglib2.0-dev libgmp3-dev libltdl-dev \
libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libreadline-dev libssl-dev libtool lrzsz \
mkisofs msmtp nano ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 python3-pip libpython3-dev qemu-utils \
rsync scons squashfs-tools subversion swig texinfo uglifyjs upx-ucl unzip vim wget xmlto xxd zlib1g-dev

二、安装arm系统依赖
       gcc-multilib 、g++-multilib、 libc6-dev-i386在arm系统里对应的依赖

sudo apt install gcc-multilib-i686-linux-gnu gcc-multilib-s390x-linux-gnu gcc-multilib-x86-64-linux-gnu gcc-multilib-x86-64-linux-gnux32

sudo apt install g++-multilib-i686-linux-gnu g++-multilib-s390x-linux-gnu g++-multilib-x86-64-linux-gnu g++-multilib-x86-64-linux-gnux32

sudo apt install libc6-dev-i386-amd64-cross libc6-dev-i386-cross libc6-dev-i386-x32-cross

三、安装go
       输入：sudo apt install golang-go

四、配置go环境
       编辑.config，这是一个隐藏文件，就在lede目录下：
1、输入：gedit .config
2、找到CONFIG_GOLANG_EXTERNAL_BOOTSTRAP_ROOT=””这行，在双引号中填入/usr/bin/go

依赖成功安装后，就可以正常编译了，祝顺利！

Yonsm, 2024-07-04:
mark 感谢 [minfengyk](https://github.com/minfengyk)

PunkFleet, 2024-12-06:
> 我后来在paralell desktop上安装20.04和22.04arm版都成功了，也是按教程操作的，可能是代理的原因吧！现附上教程：
> 
> M系列芯片电脑上ARM版Ubuntu 22安装依赖教程
> 
> 一、安装编译所需依赖 ssh登陆Ubuntu，输入： sudo apt install -y ack antlr3 aria2 asciidoc autoconf automake autopoint binutils bison build-essential bzip2 ccache cmake cpio curl device-tree-compiler fastjar flex gawk gettext git gperf haveged help2man intltool libelf-dev libglib2.0-dev libgmp3-dev libltdl-dev libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libreadline-dev libssl-dev libtool lrzsz mkisofs msmtp nano ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 python3-pip libpython3-dev qemu-utils rsync scons squashfs-tools subversion swig texinfo uglifyjs upx-ucl unzip vim wget xmlto xxd zlib1g-dev
> 
> 二、安装arm系统依赖 gcc-multilib 、g++-multilib、 libc6-dev-i386在arm系统里对应的依赖：
> 
> sudo apt install gcc-multilib-i686-linux-gnu gcc-multilib-s390x-linux-gnu gcc-multilib-x86-64-linux-gnu gcc-multilib-x86-64-linux-gnux32
> 
> sudo apt install g++-multilib-i686-linux-gnu g++-multilib-s390x-linux-gnu g++-multilib-x86-64-linux-gnu g++-multilib-x86-64-linux-gnux32
> 
> sudo apt install libc6-dev-i386-amd64-cross libc6-dev-i386-cross libc6-dev-i386-x32-cross
> 
> 三、安装go 输入：sudo apt install golang-go
> 
> 四、配置go环境 编辑.config，这是一个隐藏文件，就在lede目录下： 1、输入：gedit .config 2、找到CONFIG_GOLANG_EXTERNAL_BOOTSTRAP_ROOT=””这行，在双引号中填入/usr/bin/go
> 
> 依赖成功安装后，就可以正常编译了，祝顺利！

M2芯片上使用arm ubuntu22.04虚拟机编译出错：
```
touch 735ed8d476ae1e4beabe55fccf81cbc74e713b45.intermediate
  LD_LIBRARY_PATH=/home/lima.linux/openwrt/build_dir/target-x86_64_musl/node-v18.20.5/out/Release/lib.host:/home/lima.linux/openwrt/build_dir/target-x86_64_musl/node-v18.20.5/out/Release/lib.target:$LD_LIBRARY_PATH; export LD_LIBRARY_PATH; cd ../tools/v8_gypfiles; mkdir -p /home/lima.linux/openwrt/build_dir/target-x86_64_musl/node-v18.20.5/out/Release/obj/gen/inspector-generated-output-root/include/inspector /home/lima.linux/openwrt/build_dir/target-x86_64_musl/node-v18.20.5/out/Release/obj/gen/inspector-generated-output-root/src/inspector/protocol; python ../../deps/v8/third_party/inspector_protocol/code_generator.py --jinja_dir ../../deps/v8/third_party --output_base "/home/lima.linux/openwrt/build_dir/target-x86_64_musl/node-v18.20.5/out/Release/obj/gen/inspector-generated-output-root/src/inspector" --config ../../deps/v8/src/inspector/inspector_protocol_config.json --config_value "protocol.path=../../deps/v8/include/js_protocol.pdl" --inspector_protocol_dir ../../deps/v8/third_party/inspector_protocol
  touch 2a9dfe09c01151119d01629dd683b49a979a3f62.intermediate
  LD_LIBRARY_PATH=/home/lima.linux/openwrt/build_dir/target-x86_64_musl/node-v18.20.5/out/Release/lib.host:/home/lima.linux/openwrt/build_dir/target-x86_64_musl/node-v18.20.5/out/Release/lib.target:$LD_LIBRARY_PATH; export LD_LIBRARY_PATH; cd ../tools/v8_gypfiles; mkdir -p /home/lima.linux/openwrt/build_dir/target-x86_64_musl/node-v18.20.5/out/Release/obj/gen/inspector-generated-output-root/include/inspector /home/lima.linux/openwrt/build_dir/target-x86_64_musl/node-v18.20.5/out/Release/obj/gen/inspector-generated-output-root/src/inspector/protocol; python ../../deps/v8/third_party/inspector_protocol/code_generator.py --jinja_dir ../../deps/v8/third_party --output_base "/home/lima.linux/openwrt/build_dir/target-x86_64_musl/node-v18.20.5/out/Release/obj/gen/inspector-generated-output-root/src/inspector" --config ../../deps/v8/src/inspector/inspector_protocol_config.json --config_value "protocol.path=../../deps/v8/include/js_protocol.pdl" --inspector_protocol_dir ../../deps/v8/third_party/inspector_protocol
  g++  -o /home/lima.linux/openwrt/build_dir/target-x86_64_musl/node-v18.20.5/out/Release/obj.host/v8_base_without_compiler/deps/v8/src/trap-handler/handler-inside-posix.o ../deps/v8/src/trap-handler/handler-inside-posix.cc '-D_GLIBCXX_USE_CXX11_ABI=1' '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DICU_NO_USER_DATA_OVERRIDE' '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-D__STDC_FORMAT_MACROS' '-DV8_TARGET_ARCH_X64' '-DV8_HAVE_TARGET_OS' '-DV8_TARGET_OS_LINUX' '-DV8_EMBEDDER_STRING="-node.37"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DV8_ENABLE_PRIVATE_MAPPING_FORK_OPTIMIZATION' '-DV8_SHORT_BUILTIN_CALLS' '-DOBJECT_PRINT' '-DV8_INTL_SUPPORT' '-DV8_ATOMIC_OBJECT_FIELD_WRITES' '-DV8_ENABLE_LAZY_SOURCE_POSITIONS' '-DV8_USE_SIPHASH' '-DV8_SHARED_RO_HEAP' '-DV8_WIN64_UNWINDING_INFO' '-DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH' '-DV8_ENABLE_WEBASSEMBLY' '-DV8_ENABLE_JAVASCRIPT_PROMISE_HOOKS' '-DV8_ALLOCATION_FOLDING' '-DV8_ALLOCATION_SITE_TRACKING' '-DV8_SCRIPTORMODULE_LEGACY_LIFETIME' '-DV8_ADVANCED_BIGINT_ALGORITHMS' '-DICU_UTIL_DATA_IMPL=ICU_UTIL_DATA_STATIC' '-DUCONFIG_NO_SERVICE=1' '-DU_ENABLE_DYLOAD=0' '-DU_STATIC_IMPLEMENTATION=1' '-DU_HAVE_STD_STRING=1' '-DUCONFIG_NO_BREAK_ITERATION=0' -I../deps/v8 -I../deps/v8/include -I/home/lima.linux/openwrt/build_dir/target-x86_64_musl/node-v18.20.5/out/Release/obj/gen/inspector-generated-output-root -I../deps/v8/third_party/inspector_protocol -I/home/lima.linux/openwrt/build_dir/target-x86_64_musl/node-v18.20.5/out/Release/obj/gen -I/home/lima.linux/openwrt/build_dir/target-x86_64_musl/node-v18.20.5/out/Release/obj/gen/generate-bytecode-output-root -I../deps/icu-small/source/common -I../deps/icu-small/source/i18n -I../deps/icu-small/source/tools/toolutil -I../deps/v8/third_party/zlib -I../deps/v8/third_party/zlib/google -I/home/lima.linux/openwrt/staging_dir/target-x86_64_musl/usr/include  -pthread -Wno-unused-parameter [... cut ...]
time: package/feeds/packages/node/compile#6.27#0.20#6.47
    ERROR: package/feeds/packages/node failed to build.
make[2]: *** [package/Makefile:129: package/feeds/packages/node/compile] Error 1
make[2]: Leaving directory '/home/lima.linux/openwrt'
make[1]: *** [package/Makefile:123: /home/lima.linux/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lima.linux/openwrt'
make: *** [/home/lima.linux/openwrt/include/toplevel.mk:233: world] Error 2
```

minfengyk, 2024-12-06:
> M2芯片上使用arm ubuntu22.04虚拟机编译出错：
>
> ```
> touch 735ed8d476ae1e4beabe55fccf81cbc74e713b45.intermediate [...]

大雕Readme编译所需的依赖已经更新好几个版本了，请把把操作步骤一安装的依赖替换为最新。我最近重新安装依赖，编译成功出包，如果还报错，肯定是其它原因。