coolsnowwolf/lede #8387: 系统日志一大堆错误

binge8, 2021-12-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：系统日志一大堆错误

### (2) 路由器型号和固件版本

A：x64，OpenWrt R21.12.1 / LuCI Master (git-21.329.83689-00cb491)

### (3) 详细日志

A：见截图
[image]

minlang112, 2021-12-05:
exfat的报错……

OleStore, 2021-12-05:
Sun Dec  5 07:50:00 2021 kern.err kernel: [390801.266439] SQUASHFS error: Unable to read fragment cache entry [1ede45a]
Sun Dec  5 07:50:00 2021 kern.err kernel: [390801.274778] SQUASHFS error: Unable to read page, block 1ede45a, size 33f94
Sun Dec  5 07:50:01 2021 kern.err kernel: [390802.260548] SQUASHFS error: Unable to read fragment cache entry [1ede45a]
Sun Dec  5 07:50:01 2021 kern.err kernel: [390802.268902] SQUASHFS error: Unable to read page, block 1ede45a, size 33f94
Sun Dec  5 07:50:02 2021 kern.err kernel: [390803.265078] SQUASHFS error: Unable to read fragment cache entry [1ede45a]
Sun Dec  5 07:50:02 2021 kern.err kernel: [390803.273428] SQUASHFS error: Unable to read page, block 1ede45a, size 33f94
Sun Dec  5 07:50:03 2021 kern.err kernel: [390804.265540] SQUASHFS error: Unable to read fragment cache entry [1ede45a]
Sun Dec  5 07:50:03 2021 kern.err kernel: [390804.273878] SQUASHFS error: Unable to read page, block 1ede45a, size 33f94
Sun Dec  5 07:50:04 2021 kern.err kernel: [390805.260122] SQUASHFS error: Unable to read fragment cache entry [1ede45a]
Sun Dec  5 07:50:04 2021 kern.err kernel: [390805.268443] SQUASHFS error: Unable to read page, block 1ede45a, size 33f94
Sun Dec  5 07:50:05 2021 kern.err kernel: [390806.262551] SQUASHFS error: Unable to read fragment cache entry [1ede45a]
Sun Dec  5 07:50:05 2021 kern.err kernel: [390806.270889] SQUASHFS error: Unable to read page, block 1ede45a, size 33f94
Sun Dec  5 07:50:05 2021 kern.err kernel: [390806.453002] SQUASHFS error: Unable to read fragment cache entry [1ede45a]
Sun Dec  5 07:50:05 2021 kern.err kernel: [390806.461327] SQUASHFS error: Unable to read page, block 1ede45a, size 33f94
Sun Dec  5 07:50:06 2021 kern.err kernel: [390807.437127] SQUASHFS error: Unable to read fragment cache entry [1ede45a]
Sun Dec  5 07:50:06 2021 kern.err kernel: [390807.445854] SQUASHFS error: Unable to read page, block 1ede45a, size 33f94
Sun Dec  5 07:50:07 2021 kern.err kernel: [390808.441034] SQUASHFS error: Unable to read fragment cache entry [1ede45a]
Sun Dec  5 07:50:07 2021 kern.err kernel: [390808.449657] SQUASHFS error: Unable to read page, block 1ede45a, size 33f94
Sun Dec  5 07:50:08 2021 kern.err kernel: [390809.437046] SQUASHFS error: Unable to read fragment cache entry [1ede45a]
Sun Dec  5 07:50:08 2021 kern.err kernel: [390809.445324] SQUASHFS error: Unable to read page, block 1ede45a, size 33f94
Sun Dec  5 16:13:05 2021 kern.err kernel: [420986.632003] SQUASHFS error: Unable to read fragment cache entry [1ede45a]
Sun Dec  5 16:13:05 2021 kern.err kernel: [420986.640261] SQUASHFS error: Unable to read page, block 1ede45a, size 33f94
Sun Dec  5 16:13:06 2021 kern.err kernel: [420987.615582] SQUASHFS error: Unable to read fragment cache entry [1ede45a]
Sun Dec  5 16:13:06 2021 kern.err kernel: [420987.623862] SQUASHFS error: Unable to read page, block 1ede45a, size 33f94
Sun Dec  5 16:13:07 2021 kern.err kernel: [420988.617502] SQUASHFS error: Unable to read fragment cache entry [1ede45a]
Sun Dec  5 16:13:07 2021 kern.err kernel: [420988.625780] SQUASHFS error: Unable to read page, block 1ede45a, size 33f94

binge8, 2021-12-05:
> exfat的报错……

卸载exfat，automount也跟着卸载了。。。。。

closed 2021-12-18 by binge8