coolsnowwolf/lede #8392: 近期源码编译X86不拨号，CPU高占用，无法WEB重刷固件。

mcucore, 2021-12-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：近期源码编译X86不拨号，CPU高占用，无法WEB重刷固件。
系统
主机名	OpenWrt
主机型号	Intel(R) Atom(TM) CPU N270 @ 1.60GHz : 1 Core 2 Thread
CPU 信息	1596.157 MHz +33.0°C (crit = +90.0°C)
固件版本	OpenWrt R21.12.1 / LuCI Master (git-21.335.48743-5f363d9)
内核版本	5.4.162
本地时间	Sun Dec 5 13:14:44 2021
运行时间	1h 18m 49s
平均负载	3.07, 3.02, 2.88
CPU 使用率（%）	82 %

WEB重刷失败：
Bad Gateway
The process did not produce any response

### (2) 路由器型号和固件版本

A：X86 N270

### (3) 详细日志

A：

系统日志：

Sun Dec  5 13:11:25 2021 daemon.notice netifd: wan (12559): uci: Entry not found
Sun Dec  5 13:11:25 2021 daemon.notice netifd: wan (12559): sh: out of range
Sun Dec  5 13:11:25 2021 daemon.info pppd[12569]: Plugin pppoe.so loaded.
Sun Dec  5 13:11:25 2021 daemon.info pppd[12569]: PPPoE plugin from pppd 2.4.9
Sun Dec  5 13:11:26 2021 daemon.notice pppd[12569]: pppd 2.4.9 started by root, uid 0
Sun Dec  5 13:11:26 2021 daemon.warn pppd[12569]: Timeout waiting for PADO packets
Sun Dec  5 13:11:26 2021 daemon.err pppd[12569]: Unable to complete PPPoE Discovery
Sun Dec  5 13:11:26 2021 daemon.info pppd[12569]: Exit.
Sun Dec  5 13:11:26 2021 daemon.notice netifd: Interface 'wan' is now down
Sun Dec  5 13:11:26 2021 daemon.notice netifd: Interface 'wan' is setting up now
Sun Dec  5 13:11:26 2021 daemon.err insmod: module is already loaded - slhc
Sun Dec  5 13:11:26 2021 daemon.err insmod: module is already loaded - ppp_generic
Sun Dec  5 13:11:26 2021 daemon.err insmod: module is already loaded - pppox
Sun Dec  5 13:11:26 2021 daemon.err insmod: module is already loaded - pppoe
Sun Dec  5 13:11:26 2021 daemon.notice netifd: wan (12580): uci: Entry not found
Sun Dec  5 13:11:26 2021 daemon.notice netifd: wan (12580): sh: out of range
Sun Dec  5 13:11:26 2021 daemon.info pppd[12619]: Plugin pppoe.so loaded.
Sun Dec  5 13:11:26 2021 daemon.info pppd[12619]: PPPoE plugin from pppd 2.4.9
Sun Dec  5 13:11:26 2021 daemon.notice pppd[12619]: pppd 2.4.9 started by root, uid 0
Sun Dec  5 13:11:26 2021 daemon.warn pppd[12619]: Timeout waiting for PADO packets
Sun Dec  5 13:11:26 2021 daemon.err pppd[12619]: Unable to complete PPPoE Discovery
Sun Dec  5 13:11:26 2021 daemon.info pppd[12619]: Exit.
Sun Dec  5 13:11:26 2021 daemon.notice netifd: Interface 'wan' is now down
Sun Dec  5 13:11:26 2021 daemon.notice netifd: Interface 'wan' is setting up now
Sun Dec  5 13:11:26 2021 daemon.err insmod: module is already loaded - slhc
Sun Dec  5 13:11:26 2021 daemon.err insmod: module is already loaded - ppp_generic
Sun Dec  5 13:11:26 2021 daemon.err insmod: module is already loaded - pppox
Sun Dec  5 13:11:26 2021 daemon.err insmod: module is already loaded - pppoe
Sun Dec  5 13:11:26 2021 daemon.notice netifd: wan (12653): uci: Entry not found
Sun Dec  5 13:11:26 2021 daemon.notice netifd: wan (12653): sh: out of range
Sun Dec  5 13:11:26 2021 daemon.info pppd[12686]: Plugin pppoe.so loaded.
Sun Dec  5 13:11:26 2021 daemon.info pppd[12686]: PPPoE plugin from pppd 2.4.9
Sun Dec  5 13:11:26 2021 daemon.notice pppd[12686]: pppd 2.4.9 started by root, uid 0
[... 1665 lines cut ...]
[   28.529227] bnx2x: QLogic 5771x/578xx 10/20-Gigabit Ethernet Driver bnx2x 1.713.36-0 (2014/02/10)
[   28.656245] PPP generic driver version 2.4.2
[   28.678543] PPP MPPE Compression module registered
[   28.705128] NET: Registered protocol family 24
[   28.747544] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[   28.776344] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[   28.862482] usbcore: registered new interface driver rt2800usb
[   28.896940] usbcore: registered new interface driver ath9k_htc
[   28.932110] kmodloader: done loading kernel modules from /etc/modules.d/*
[   56.077390] 8139too 0000:03:02.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC5E1
[   56.111096] br-lan: port 1(eth0) entered blocking state
[   56.133715] br-lan: port 1(eth0) entered disabled state
[   56.156356] device eth0 entered promiscuous mode
[   56.179436] br-lan: port 1(eth0) entered blocking state
[   56.201489] br-lan: port 1(eth0) entered forwarding state
[   56.234901] RTL8211B Gigabit Ethernet r8169-0-318:00: attached PHY driver [RTL8211B Gigabit Ethernet] (mii_bus:phy_addr=r8169-0-318:00, irq=IGNORE)
[   56.399619] r8169 0000:03:03.0 eth1: Link is Down
[   56.422173] squashfs: Unknown parameter 'discard'
[   56.445313] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0xd44d89e4)
[   56.474881] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
[   56.504326] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x3619be5f)
[   56.533680] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
[   56.565102] [EXFAT] trying to mount...
[   57.086949] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   75.850880] r8169 0000:03:03.0 eth1: Link is Up - 100Mbps/Full - flow control off

renfei, 2021-12-06:
我提个小建议哈，你这个日志太长了，TL;DR

既然CPU占用高，不妨SSH进去，htop 看一下是哪个进程在占用CPU，或者在菜单 状态->系统进程，里面看看谁在占用 CPU，然后再找原因呢。

我也是 x86的，用的 11月30号的版本，刚开始也是 CPU 占用高，后来我把编译时的插件尽量少选，之选我用的上的，然后就正常了，估计是插件之间有啥冲突

tonyk520, 2021-12-06:
12月1号以后我编译的固件也无法WEB升级，只能重刷，无论是否保留配置都一样......应该是内核、驱动或者分区之类的改动太大无法兼容之前的版本了，不过我也习惯重装了，改几处配置累不到人......

nivalxer, 2021-12-06:
pppoe拨号失败的问题，试试
make distclean 或 make clean
然后重新编译就可以解决。

mcucore, 2021-12-07:
> pppoe拨号失败的问题，试试 make distclean 或 make clean 然后重新编译就可以解决。

好的，谢谢！我再试试。我编译的是32位版本，看前几天源码有更新过PPP，高CPU的问题有空再刷下之前固件看看问题在哪。

badcrazy, 2021-12-08:
> 12月1号以后我编译的固件也无法WEB升级，只能重刷，无论是否保留配置都一样......应该是内核、驱动或者分区之类的改动太大无法兼容之前的版本了，不过我也习惯重装了，改几处配置累不到人......

2个月没编译固件，上月中编译了个11.11版本，web刷入还特地没保留配置，结果重启还是起不来，接上显示器发现看似一切正常，但就是启动出现op字样前一刻，光标一直闪，就是不继续了，网络没出来，后来pe下重刷镜像就好了，问题很诡异，之前从没遇到过

closed 2022-01-08 by AmadeusGhost