coolsnowwolf/lede #8393: 今天编译出错busybox下载失败

dengchan0416, 2021-12-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
编译busybox下载失败
* [ ] 没有类似的issue

### (1) 具体问题

在编译时候提示busybox下载失败

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志
[image]
[image]

WYC-2020, 2021-12-06:
自己网络的问题

dengchan0416, 2021-12-06:
我下载了很多次

WYC-2020, 2021-12-06:
刚下载试的
[image]
实在不行 自己手动复制地址出来下载

dengchan0416, 2021-12-06:
> 刚下载试的 [image] 实在不行 自己手动复制地址出来下载

请问下载好了放到那个目录里

hzjnet, 2021-12-06:
> > 刚下载试的 [image] 实在不行 自己手动复制地址出来下载
> 
> 请问下载好了放到那个目录里

dl/

closed 2021-12-08 by AmadeusGhost