coolsnowwolf/lede #8394: 云编译出错 这个怎么解

Jason6111, 2021-12-06:
sed -i -e 's,^soname_spec=.*,soname_spec="\\${libname}\\${shared_ext}.4.8.1.9c91d6fa",' -e 's,^library_names_spec=.*,library_names_spec="\\${libname}\\${shared_ext}.4.8.1.9c91d6fa \\${libname}\\${shared_ext}",' /workdir/openwrt/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/libtool
2489
sed: can't read /workdir/openwrt/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/libtool: No such file or directory
2490
make[3]: *** [Makefile:132: /workdir/openwrt/build_dir/target-x86_64_musl/wolfssl-4.8.1-stable/.configured_9c91d6fab8bdec71fcc929050cfe99da] Error 2
2491
make[3]: Leaving directory '/workdir/openwrt/package/libs/wolfssl'
2492
time: package/libs/wolfssl/compile#2.82#0.16#3.94
2493
    ERROR: package/libs/wolfssl failed to build.
2494
make[2]: *** [package/Makefile:116: package/libs/wolfssl/compile] Error 1
2495
make[2]: Leaving directory '/workdir/openwrt'
2496
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2497
make[1]: Leaving directory '/workdir/openwrt'
2498
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
2499
Error: Process completed with exit code 2.

WYC-2020, 2021-12-07:
稍微看下以前issue 你就不会这么问，去掉云编译哪个什么cache 那玩意不靠谱，去掉在编译看下

Jason6111, 2021-12-07:
> 稍微看下以前issue 你就不会这么问，去掉云编译哪个什么cache 那玩意不靠谱，去掉在编译看下

一语点醒蒙圈的我 感谢

closed 2021-12-07 by Jason6111