coolsnowwolf/lede #8396: 这里大佬多，问一下Action编译总被中断怎么破

eightsheep67, 2021-12-07:
几个月前就发现了，症状就是action通过ssh手动make menuconfig，编译个1小时左右没有啥报错就直接被cancelled
如果ssh选择false用上传好的.config文件直接编译的话就可以成功。
但是最近大佬仓库的默认配置改了，编译出来会多出docker之类的我不需要的，想要添加一些配置也不知道怎么编辑config。

有大佬遇到这个问题有解决方法的吗？？

PyXiaoxin, 2021-12-07:
SSH进去把配置导出来，然后再直接编译。看下这个大佬的[脚本](https://github.com/281677160/build-actions)你就明白了

binge8, 2021-12-07:
只要用SSH，就会一小时中断

eightsheep67, 2021-12-07:
> SSH进去把配置导出来，然后再直接编译。看下这个大佬的[脚本](https://github.com/281677160/build-actions)你就明白了

我研究一下，谢谢大佬！！

ZqinKing, 2021-12-07:
> 只要用SSH，就会一小时中断

修改本地SSH的心跳时间。

eightsheep67, 2021-12-07:
> > 只要用SSH，就会一小时中断
> 
> 修改本地SSH的心跳时间。

这。。。
我具体说一下，不是SSH连接中断，是如果你action云编译用了SSH连接上更改config配置了以后。这个编译大概坚持1个小时就会被中断。

Neverends4, 2021-12-07:
github action？被薅羊毛薅秃了，买会员吧

Purple-wind-bell, 2021-12-07:
所以为什么要连ssh，本地编一次确定没问题配置文件传上去不好吗

eightsheep67, 2021-12-07:
> 所以为什么要连ssh，本地编一次确定没问题配置文件传上去不好吗

我不想浪费这时间本地编译，能云编译肯定就云编译了。配置文件又不能用一辈子，按你这样说想稍微更改一下还要编译两次？

binge8, 2021-12-08:
> > 所以为什么要连ssh，本地编一次确定没问题配置文件传上去不好吗
> 
> 我不想浪费这时间本地编译，能云编译肯定就云编译了。配置文件又不能用一辈子，按你这样说想稍微更改一下还要编译两次？

这也不会那也不会，毛病倒不少。要求还这么高。自己去做一个不中断的SSH啊

eightsheep67, 2021-12-08:
> 

请问您是有什么毛病吗？？怎么好好的突然就开始咬人了呢？

eightsheep67, 2021-12-08:
> > > 所以为什么要连ssh，本地编一次确定没问题配置文件传上去不好吗
> > 
> > 
> > 我不想浪费这时间本地编译，能云编译肯定就云编译了。配置文件又不能用一辈子，按你这样说想稍微更改一下还要编译两次？
> 
> 这也不会那也不会，毛病倒不少。要求还这么高。自己去做一个不中断的SSH啊

请问您是有什么毛病吗？？怎么好好的突然就开始咬人了呢？

closed 2021-12-08 by AmadeusGhost