coolsnowwolf/lede #8405: 小米路由器3默认配置生成的pb-boot.bin文件从pb-boot输入后，路由器无限重启。

MortalM, 2021-12-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
小米路由器3默认配置生成的pb-boot.bin文件从pb-boot输入后，路由器无限重启。
### (1) 具体问题
为了产生pb-boot，在target/linux/ramips/image/mt7620.mk里新增了1条命令
  define Device/xiaomi_miwifi-r3
  SOC := mt7620a
  BLOCKSIZE := 128k
  PAGESIZE := 2048
  KERNEL_SIZE := 4096k
  IMAGE_SIZE := 32768k
  UBINIZE_OPTS := -E 5
  IMAGES += kernel1.bin rootfs0.bin pb-boot.bin
  IMAGE/kernel1.bin := append-kernel | check-size $$$$(KERNEL_SIZE)
  IMAGE/rootfs0.bin := append-ubi | check-size $$$$(IMAGE_SIZE)
  IMAGE/sysupgrade.bin := sysupgrade-tar | append-metadata
  IMAGE/pb-boot.bin := append-kernel | pad-to $$(KERNEL_SIZE) | \
                             append-kernel | pad-to $$(KERNEL_SIZE) | \
                             append-ubi | check-size $$$$(IMAGE_SIZE)
  DEVICE_VENDOR := Xiaomi
  DEVICE_MODEL := Mi Router R3
  DEVICE_PACKAGES := kmod-mt76x2 kmod-usb2 kmod-usb-ohci uboot-envtools
endef
TARGET_DEVICES += xiaomi_miwifi-r3

### (2) 路由器型号和固件版本

A：小米路由器3     21.12.1

编译正常，生成16m左右的openwrt-ramips-mt7620-xiaomi_miwifi-r3-squashfs-pb-boot.bin  文件，刷入后机器无限重启，
但是刷最新X-WRT 21.10_b202111280050 Impish / LuCI Master git-21.331.59232-43dea66的x-wrt 正常，刷其他人编译的较老版本的openwrt也正常。  请问怎么回事呢。

deplives, 2021-12-08:
一个问题有必要开两个issues么

renfei, 2021-12-09:
# 1.关于你要提交的问题
Q：是否搜索了issue (使用 "x" 选择)

- [ ] 没有类似的issue

就是为了防止重复的，#8402 为啥又发一遍

MortalM, 2021-12-09:
不好意思，网络不好重复提交了。
感谢2位大佬更改代码，我今早又重新清空了lede文件夹，重新默认编译了一次，就可以正常使用了。
之前编译的内核只有1.94m大小，大约持续半个月时间也多次清空过全部代码，编译出来的一直都是这么大，
今天早上重新编译的内核文件有2.08m大小，在用pb-boot.bin文件在pb-boot刷入，就能直接启动了，系统也很流畅，感谢这个项目！

closed 2021-12-09 by MortalM