coolsnowwolf/lede #8406: 使用usb3.0 2.5G网卡 绿联的8156，访问局域网内nas大量报错

alexliu269, 2021-12-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
局域网下访问nas时出现大量报错以及速度很慢
### (1) 具体问题

A：使用usb3.0 2.5G网卡 绿联的8156，访问局域网内nas大量报错。	

### (2) 路由器型号和固件版本

A：X86 软路由 OpenWrt R21.12.1 / LuCI Master (git-21.335.48743-5f363d9)

### (3) 详细日志

A：[  351.104827] ------------[ cut here ]------------
[  351.104918] NETDEV WATCHDOG: eth6 (r8152): transmit queue 0 timed out
[  351.105032] WARNING: CPU: 1 PID: 0 at dev_watchdog+0x21c/0x220
[  351.105126] Modules linked in: xt_FULLCONENAT ath9k ath9k_htc ath9k_common xt_connlimit rt2800usb rt2800lib pppoe ppp_async nf_conncount iptable_nat ath9k_hw ath5k ath10k_pci ath10k_core ath xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_FLOWOFFLOAD xt_CT rt2x00usb rt2x00lib pppox ppp_mppe ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mac80211 ipt_REJECT cfg80211 bnx2x ax88179_178a asix xt_time xt_tcpudp xt_tcpmss xt_statistic xt_socket xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_ecn xt_dscp xt_comment xt_TPROXY xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY via_velocity usbnet ums_usbat ums_sddr55
[  351.105145]  ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_kmp ts_fsm ts_bm tcp_bbr slhc rtl8150 r8168 r8152 r8125 pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 mmc_spi mlx5_core mlx4_en mlx4_core macvlan libcrc32c iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables igc iavf i40e forcedeth e1000e crc7 crc_itu_t crc_ccitt compat bnx2 asn1_decoder alx 8139too 8139cp evdev i2c_dev cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 bonding ixgbevf ixgbe igbvf igb e1000 amd_xgbe mdio vfat
[  351.106594]  fat nls_utf8 nls_iso8859_1 nls_cp437 ena crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha256_ssse3 sha256_generic libsha256 sha1_ssse3 sha1_generic seqiv jitterentropy_rng drbg hmac ghash_generic gcm ctr cmac ccm authenc arc4 uas sdhci_pltfm xhci_plat_hcd exfat(C) antfs button_hotplug ptp pps_core mii libphy
[  351.108357] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G         C        5.4.162 #0
[  351.108476] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 12/12/2018
[  351.108651] RIP: 0010:dev_watchdog+0x21c/0x220
[  351.108727] Code: 00 49 63 4d e8 eb 92 4c 89 e7 c6 05 1f 3f b1 00 01 e8 08 43 fc ff 89 d9 4c 89 e6 48 c7 c7 18 62 08 82 48 89 c2 e8 57 12 10 00 <0f> 0b eb c0 55 48 89 e5 41 57 49 89 d7 41 56 4d 89 c6 41 55 49 89
[  351.109025] RSP: 0018:ffffc900000c4e90 EFLAGS: 00010286
[  351.109111] RAX: 0000000000000039 RBX: 0000000000000000 RCX: 0000000000000006
[  351.109224] RDX: 0000000000000007 RSI: 0000000000000002 RDI: ffff888131e9c4d0
[  351.109337] RBP: ffffc900000c4ec0 R08: 0000000000000000 R09: 0000000000000570
[  351.109451] R10: 7420302065756575 R11: 74756f2064656d69 R12: ffff88812ba34000
[  351.109593] R13: ffff88812ba34478 R14: 0000000000000001 R15: 0000000000000001
[  351.109707] FS:  0000000000000000(0000) GS:ffff888131e80000(0000) knlGS:0000000000000000
[  351.109836] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  351.109930] CR2: 00007f9ca3eae000 CR3: 0000000002208005 CR4: 00000000003606e0
[  351.110072] Call Trace:
[  351.110120]  <IRQ>
[  351.110162]  ? dev_reset_queue.constprop.48+0x80/0x80
[  351.110247]  call_timer_fn.isra.40+0x16/0x80
[  351.110320]  run_timer_softirq+0x164/0x180
[  351.110391]  ? lapic_next_deadline+0x27/0x30
[  351.110464]  ? clockevents_program_event+0x8b/0xf0
[  351.110546]  __do_softirq+0xc9/0x20e
[  351.110609]  irq_exit+0x6f/0x80
[  351.110666]  smp_apic_timer_interrupt+0x5e/0x90
[  351.110742]  apic_timer_interrupt+0xf/0x20
[  351.110812]  </IRQ>
[  351.110854] RIP: 0010:native_safe_halt+0x17/0x20
[  351.110932] Code: 00 a8 08 75 c3 e9 77 ff ff ff 90 90 90 90 90 90 90 90 90 55 8b 05 a9 6b b1 00 48 89 e5 85 c0 7e 07 0f 00 2d cb 60 54 00 fb f4 <5d> c3 0f 1f 80 00 00 00 00 55 8b 05 89 6b b1 00 48 89 e5 85 c0 7e
[  351.111222] RSP: 0018:ffffc9000007fea0 EFLAGS: 00000246 ORIG_RAX: ffffffffffffff13
[  351.111346] RAX: 0000000000000000 RBX: 0000000000000001 RCX: 0000000000000001
[  351.111460] RDX: 00000000003d1ba6 RSI: 0000000000000000 RDI: 0000000000000000
[  351.111575] RBP: ffffc9000007fea0 R08: ffff888131518080 R09: 0000000000000000
[  351.111689] R10: 0000000000000000 R11: 0000000000000010 R12: ffffffff82294b40
[  351.111803] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[  351.111918]  default_idle+0x9/0x10
[  351.111979]  arch_cpu_idle+0x10/0x20
[  351.112042]  default_idle_call+0x27/0x30
[  351.112110]  do_idle+0x19f/0x1c0
[  351.112168]  cpu_startup_entry+0x18/0x20
[  351.112236]  start_secondary+0x145/0x170
[  351.112304]  secondary_startup_64+0xa4/0xb0
[  351.112376] ---[ end trace 11b80bf4fe3c0b11 ]---
[  351.112469] r8152 4-2:1.0 eth6: Tx timeout
[  351.114952] r8152 4-2:1.0 eth6: Tx status -2
[  351.115046] r8152 4-2:1.0 eth6: Tx status -2
[  351.115144] r8152 4-2:1.0 eth6: Tx status -2
[  351.115235] r8152 4-2:1.0 eth6: Tx status -2
[  351.249066] usb 4-2: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[  351.270604] r8152 4-2:1.0 eth6: Promiscuous mode enabled
[  357.257719] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  357.284319] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  357.326892] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  357.424363] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  360.596213] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  360.734763] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  364.113950] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  364.256783] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  366.148495] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  366.212100] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  367.953607] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  370.203645] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  370.438021] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  371.822255] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  371.993696] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  373.390915] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  373.557135] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  374.458926] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  374.462695] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  374.464375] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  374.479476] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  374.497584] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  374.546842] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  388.777345] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  388.778349] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  388.855192] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  388.865413] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  388.878133] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  388.948285] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  388.999537] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  389.219711] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  389.312153] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  389.515984] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  389.624055] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  389.838729] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  389.930679] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  390.078278] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  390.238640] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90
[  390.390723] r8152 4-2:1.0 eth6: skb_to_sgvec fail -90

minlang112, 2021-12-11:
看起来你这是虚拟机，RTL8156你是直通USB设备给虚拟机的？

alexliu269, 2021-12-11:
> 看起来你这是虚拟机，RTL8156你是直通USB设备给虚拟机的？

对的 直通的  重新编译后 还出现过一次 然后就不出现报错了。虽然不知道是什么情况 但是目前没有报错了

closed 2021-12-22 by alexliu269