coolsnowwolf/lede #8407: 小米路由器3的mt7620a，刷最新源码默认生成的固件，隔几分钟就断网一次，有线无线都断。

MortalM, 2021-12-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
小米路由器3的mt7620a，刷最新源码默认生成的固件，隔几分钟就断网一次，有线无线都断，网上查了下说可能是mt7620的交换器的驱动有问题？希望大佬帮看下。 万分感谢，祝大佬新年快乐。

### (2) 路由器型号和固件版本
openwrt21.12.1         kernel5.4

### (3) 详细日志
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.919823] Call Trace:
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.922356] [<8000b3d0>] show_stack+0x30/0x100
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.926909] [<80025c18>] __warn+0xc0/0x10c
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.931088] [<80025cf0>] warn_slowpath_fmt+0x8c/0xac
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.936173] [<803d7808>] dev_watchdog+0x26c/0x274
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.941000] [<80069e34>] call_timer_fn.isra.34+0x20/0x90
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.946441] [<8006a048>] run_timer_softirq+0x1a4/0x1ec
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.951706] [<80513fe8>] __do_softirq+0x110/0x298
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.956528] [<80005e90>] except_vec_vi_end+0xb8/0xc4
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.961600] [<8050d280>] __udelay+0x30/0x40
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.965905] [<80325ed0>] _ra_nand_pull_data.constprop.21+0xa4/0xfc
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.972233] [<80326454>] nfc_read_page+0xec/0x1f0
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.977039] [<80326db0>] ramtd_nand_read+0x170/0x3d4
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.982141] [<803163d0>] part_read+0x60/0xb8
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.986502] [<80313558>] mtd_read_oob+0xe8/0x100
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.991212] [<803135ac>] mtd_read+0x3c/0x54
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  236.995498] [<8031da08>] mtdblock_readsect+0x58/0x17c
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.000655] [<8031d2f0>] mtd_queue_rq+0x408/0x5a4
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.005482] [<80257cac>] __blk_mq_try_issue_directly+0x140/0x204
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.011615] [<80258cc4>] blk_mq_request_issue_directly+0xb8/0x13c
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.017851] [<80258da0>] blk_mq_try_issue_list_directly+0x58/0xf4
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.024099] [<8025d84c>] blk_mq_sched_insert_requests+0xc4/0x198
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.030229] [<80258bc8>] blk_mq_flush_plug_list+0x120/0x164
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.035944] [<8024dd84>] blk_flush_plug_list+0xc4/0xe8
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.041189] [<8024ddd0>] blk_finish_plug+0x28/0x3c
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.046100] [<800da474>] read_pages+0x78/0x168
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.050637] [<800da728>] __do_page_cache_readahead+0x1c4/0x20c
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.056628] [<800d0ae0>] filemap_fault+0x62c/0x9a0
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.061543] [<800fb260>] __do_fault+0x3c/0x154
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.066103] [<8010024c>] handle_mm_fault+0x7c0/0xd00
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.071174] [<800f9adc>] __get_user_pages.part.51+0x238/0x450
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.077058] [<800faa40>] populate_vma_page_range+0x70/0x8c
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.082679] [<800fab24>] __mm_populate+0xc8/0x178
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.087498] [<800ec594>] vm_mmap_pgoff+0xcc/0x118
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.092320] [<80104640>] ksys_mmap_pgoff+0xc4/0x134
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.097308] [<800106ec>] syscall_common+0x34/0x58
Thu Dec  9 19:38:04 2021 kern.warn kernel: [  237.102126] ---[ end trace 5c0a9ee7ea219047 ]---
Thu Dec  9 19:38:04 2021 kern.err kernel: [  237.106849] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Dec  9 19:38:04 2021 kern.info kernel: [  237.113444] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Dec  9 19:38:04 2021 kern.info kernel: [  237.119850] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=064d4000, max=1024, ctx=531, dtx=531, fdx=513, next=531
Thu Dec  9 19:38:04 2021 kern.info kernel: [  237.131148] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=05194000, max=1024, calc=840, drx=844
Thu Dec  9 19:38:10 2021 kern.err kernel: [  240.032056] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Dec  9 19:38:10 2021 kern.info kernel: [  240.038653] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Dec  9 19:38:10 2021 kern.info kernel: [  240.045085] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=064d4000, max=1024, ctx=317, dtx=317, fdx=113, next=317
Thu Dec  9 19:38:10 2021 kern.info kernel: [  240.056373] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06434000, max=1024, calc=683, drx=684
Thu Dec  9 19:38:20 2021 kern.err kernel: [  245.297690] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Dec  9 19:38:20 2021 kern.info kernel: [  245.304302] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Dec  9 19:38:20 2021 kern.info kernel: [  245.310739] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=064d4000, max=1024, ctx=124, dtx=124, fdx=80, next=124
Thu Dec  9 19:38:20 2021 kern.info kernel: [  245.321942] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=05194000, max=1024, calc=491, drx=492
Thu Dec  9 19:38:30 2021 kern.err kernel: [  250.242440] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Dec  9 19:38:30 2021 kern.info kernel: [  250.249034] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Dec  9 19:38:30 2021 kern.info kernel: [  250.255472] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=064d4000, max=1024, ctx=80, dtx=80, fdx=16, next=80
Thu Dec  9 19:38:30 2021 kern.info kernel: [  250.266409] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06434000, max=1024, calc=208, drx=209
Thu Dec  9 19:38:54 2021 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 4c:79:6e:a0:ef:e2
[... 30 lines cut ...]
Thu Dec  9 19:39:25 2021 kern.info kernel: [  278.757898] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06434000, max=1024, calc=57, drx=59
Thu Dec  9 19:39:27 2021 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 4c:79:6e:a0:ef:e2
Thu Dec  9 19:39:31 2021 daemon.info hostapd: wlan0: STA 4c:79:6e:a0:ef:e2 IEEE 802.11: authenticated
Thu Dec  9 19:39:31 2021 daemon.info hostapd: wlan0: STA 4c:79:6e:a0:ef:e2 IEEE 802.11: associated (aid 4)
Thu Dec  9 19:39:31 2021 daemon.notice hostapd: wlan0: STA-OPMODE-SMPS-MODE-CHANGED 4c:79:6e:a0:ef:e2 off
Thu Dec  9 19:39:31 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 4c:79:6e:a0:ef:e2
Thu Dec  9 19:39:31 2021 daemon.info hostapd: wlan0: STA 4c:79:6e:a0:ef:e2 RADIUS: starting accounting session 813F38EA8D98ADB0
Thu Dec  9 19:39:31 2021 daemon.info hostapd: wlan0: STA 4c:79:6e:a0:ef:e2 WPA: pairwise key handshake completed (RSN)
Thu Dec  9 19:39:31 2021 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 4c:79:6e:a0:ef:e2
Thu Dec  9 19:39:35 2021 kern.err kernel: [  282.374864] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Dec  9 19:39:35 2021 kern.info kernel: [  282.381468] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Dec  9 19:39:35 2021 kern.info kernel: [  282.387924] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=064d4000, max=1024, ctx=22, dtx=22, fdx=16, next=22
Thu Dec  9 19:39:35 2021 kern.info kernel: [  282.398860] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=05194000, max=1024, calc=35, drx=36
Thu Dec  9 19:45:18 2021 daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS 4c:79:6e:a0:ef:e2 2 ack=1
Thu Dec  9 19:45:18 2021 daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS 3a:6c:27:50:c3:5b 1 ack=1
Thu Dec  9 19:45:18 2021 daemon.notice hostapd: wlan0: BEACON-RESP-RX 4c:79:6e:a0:ef:e2 2 00 802c8c8351200000000058020872ff7811dc5c7ec600ccd0412001d5ccd041200000000064001110000b4e69616e6747756f5f354701088c129824b048606c03012c050400020000070c55532024081e640c1e95051e30140100000fac040100000fac040100000fac020c000b050400010000460572000000003b0280002d1aef0103ffff0000000000000000000001000000000000000000003d162c0504000000000000000000000000000000000000007f080400080200000140bf0cb0018031faff0000faff0000c005012a00fcffc304023c3c3cdd180050f2020101 [... cut ...]
Thu Dec  9 19:48:40 2021 kern.err kernel: [  482.920900] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Dec  9 19:48:40 2021 kern.info kernel: [  482.927505] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Dec  9 19:48:40 2021 kern.info kernel: [  482.933936] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=064d4000, max=1024, ctx=255, dtx=255, fdx=252, next=255
Thu Dec  9 19:48:40 2021 kern.info kernel: [  482.945227] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=05194000, max=1024, calc=528, drx=529

MortalM, 2021-12-09:
出现断网的时候，日志里就会出现kern.err kernel: [ 482.920900] ralink_soc_eth 10100000.ethernet eth0: transmit timed out

MortalM, 2021-12-11:
有时候又有这样的报错。。 但是这样的报错好像不会导致ssh中断，就是QQ会断线在重连。[  105.024798] device wlan0 left promiscuous mode
[  105.029606] br-lan: port 3(wlan0) entered disabled state
[  105.445609] br-lan: port 2(wlan1) entered disabled state
[  105.451224] br-lan: port 1(eth0.1) entered disabled state
[  105.536451] device wlan1 left promiscuous mode
[  105.541236] br-lan: port 2(wlan1) entered disabled state
[  105.551211] device eth0.1 left promiscuous mode
[  105.556031] br-lan: port 1(eth0.1) entered disabled state
[  105.656458] device eth0 left promiscuous mode
[  118.119909] 8021q: adding VLAN 0 to HW filter on device eth0
[  118.137278] device eth0 entered promiscuous mode
[  118.148073] br-lan: port 1(eth0.1) entered blocking state
[  118.153595] br-lan: port 1(eth0.1) entered disabled state
[  118.159639] device eth0.1 entered promiscuous mode
[  118.187062] br-lan: port 1(eth0.1) entered blocking state
[  118.192587] br-lan: port 1(eth0.1) entered forwarding state
[  119.158793] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[  124.729346] ieee80211 phy1: rt2800_rf_self_txdc_cal: Info - RF Tx self calibration start
[  124.738179] ieee80211 phy1: rt2800_rf_self_txdc_cal: Info - RF Tx self calibration end
[  126.847211] ieee80211 phy1: rt2800_loft_iq_calibration: Info - LOFT Calibration Done!
[  126.859229] ieee80211 phy1: rt2800_iq_search: Info - IQCalibration Start!
[  126.880122] ieee80211 phy1: rt2800_iq_search: Info - IQCalibration Done! CH = 0, (gain= 3, phase=3a)
[  126.890370] ieee80211 phy1: rt2800_iq_search: Info - IQCalibration Start!
[  126.911702] ieee80211 phy1: rt2800_iq_search: Info - IQCalibration Done! CH = 1, (gain= 1, phase=3d)
[  126.921074] ieee80211 phy1: rt2800_loft_iq_calibration: Info - TX IQ Calibration Done!
[  126.968235] ieee80211 phy1: rt2800_rxiq_calibration: Info - RXIQ G_imb=0, Ph_rx=3
[  126.983752] ieee80211 phy1: rt2800_rxiq_calibration: Info - RXIQ G_imb=-1, Ph_rx=-1
[  127.027435] br-lan: port 2(wlan1) entered blocking state
[  127.032966] br-lan: port 2(wlan1) entered disabled state
[  127.038848] device wlan1 entered promiscuous mode
[  127.492156] br-lan: port 3(wlan0) entered blocking state
[  127.497591] br-lan: port 3(wlan0) entered disabled state
[  127.503960] device wlan0 entered promiscuous mode
[  127.508928] br-lan: port 3(wlan0) entered blocking state
[  127.514424] br-lan: port 3(wlan0) entered forwarding state
[  128.180687] br-lan: port 3(wlan0) entered disabled state
[  128.853933] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  128.860857] br-lan: port 3(wlan0) entered blocking state
[  128.866284] br-lan: port 3(wlan0) entered forwarding state
[  129.443098] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  129.449840] br-lan: port 2(wlan1) entered blocking state
[  129.455344] br-lan: port 2(wlan1) entered forwarding state
[  236.896547] ralink_soc_eth 10100000.ethernet eth0: port 0 link up (100Mbps/Full duplex)
[  270.147691] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[  271.106538] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[  364.383554] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[  365.330317] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[  486.649252] ralink_soc_eth 10100000.ethernet eth0: port 0 link down
[  489.493163] ralink_soc_eth 10100000.ethernet eth0: port 0 link up (100Mbps/Full duplex)
[  490.181106] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[  491.124559] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[  508.943074] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[  509.808465] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[  639.987096] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[  640.834750] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[  667.266157] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[  668.248389] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[  668.288946] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[  669.191955] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[  669.330269] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[  670.277141] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[  676.730269] ralink_soc_eth 10100000.ethernet eth0: port 0 link down
[  676.851621] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[  676.860917] br-lan: port 1(eth0.1) entered disabled state
[  679.156010] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[  679.164833] br-lan: port 1(eth0.1) entered blocking state
[  679.170388] br-lan: port 1(eth0.1) entered forwarding state
[  682.081774] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[  682.091812] br-lan: port 1(eth0.1) entered disabled state
[  684.226672] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[  684.235375] br-lan: port 1(eth0.1) entered blocking state
[  684.240951] br-lan: port 1(eth0.1) entered forwarding state
[  685.632235] ralink_soc_eth 10100000.ethernet eth0: port 0 link up (100Mbps/Full duplex)

MortalM, 2021-12-11:
@AmadeusGhost 大佬来看看我这个啊，编译了10几次了，都是差不多的情况

chobitsos, 2021-12-13:
我自己改了个补丁，目前在测试https://github.com/coolsnowwolf/lede/issues/8425

LINGJP, 2021-12-15:
7620a单核580mhz终究是老了，回4.14内核吧

chobitsos, 2021-12-20:
> 7620a单核580mhz终究是老了，回4.14内核吧

这跟老不老没什么关系，顶多带不动多台机器，同样的内核隔壁x-wrt稳如狗，换L大的固件就断流，这是一个bug

chobitsos, 2021-12-20:
> 7620a单核580mhz终究是老了，回4.14内核吧

顺便说下，官方固件纯净版5.10内核，价格科学插件仍然能带十几台电脑，暴力测试有线上下行跑满10M，也没断流，L大这边无论有线无线上下行还没跑1M就time out断流了

xiaopang0v0, 2022-01-12:
其实你这个issue我以前就发过的，我觉得是交换机驱动的问题，后面编译了好几个固件，实在是烦了，就没弄了
https://github.com/coolsnowwolf/lede/issues/8046

chobitsos, 2022-05-22:
已换SDK
测试30分钟双向跑满没有断流，适配lean的packges仓库
[image]

[image]

[image]

coolsnowwolf, 2026-09-05:
小米路由器 3 的 `ralink_soc_eth` transmit timeout 来自旧版 MT7620/5.4 内核，缺少当前版本复现和硬件对照；关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf