coolsnowwolf/lede #8410: Error make

usermigallon, 2021-12-09:
Hi, I have this error
make[5]: *** [Makefile:41: /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-mediatek_mt7623/tmp/openwrt-mediatek-mt7623-bananapi_bpi-r2-u-boot.bin] Error 1
make[5]: Leaving directory '/home/user/lede/target/linux/mediatek/image'
make[4]: *** [Makefile:17: install] Error 2
make[4]: Leaving directory '/home/user/lede/target/linux/mediatek'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/user/lede/target/linux'
time: target/linux/install#110.21#11.66#94.37
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/install] Error 1
make[2]: Leaving directory '/home/user/lede'
make[1]: *** [target/Makefile:19: /home/user/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/user/lede'
make: *** [/home/user/lede/include/toplevel.mk:230: world] Error 2
Thanks in advance

MortalM, 2021-12-10:
maybe you can use 
cd /lede
make dirclean
rm -rf ./tmp && rm -rf .config
./scripts/feeds clean

and then restart you compile

MortalM, 2021-12-10:
please close this issue ,if you resolved this problem

usermigallon, 2021-12-10:
Thank you very much, I will do so

usermigallon, 2021-12-10:
Same, do not work, I have this output now 
bash: po2lmo: command not found
make[3]: *** [Makefile:44: /home/user/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/default-settings-2/.pkgdir/default-settings.installed] Error 127
make[3]: Leaving directory '/home/user/lede/package/lean/default-settings'
time: package/lean/default-settings/compile#0.14#0.08#0.20
    ERROR: package/lean/default-settings failed to build.
make[2]: *** [package/Makefile:116: package/lean/default-settings/compile] Error 1
make[2]: Leaving directory '/home/user/lede'
make[1]: *** [package/Makefile:110: /home/user/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/lede'
make: *** [/home/user/lede/include/toplevel.mk:230: world] Error 2

closed 2022-02-12 by Beginner-Go