coolsnowwolf/lede #8417: 最新batman-adv崩溃

ntlf9t, 2021-12-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：easymesh设置保存后路由失联无限重启，ttl连接查明是batman-adv 2021.04问题，建议大家近期先别集成或者回退batman-adv 至2021.03版

### (2) 路由器型号和固件版本

A：cm520 新3 等

### (3) 详细日志

A：
[   36.726825] 8<--- cut here ---
[   36.726862] Unable to handle kernel NULL pointer dereference at virtual address 000002fc
[   36.728801] pgd = e6966e8d
[   36.737041] [000002fc] *pgd=9c8dd835, *pte=00000000, *ppte=00000000
[   36.739567] Internal error: Oops: 17 [#1] SMP ARM
[   36.745708] Modules linked in: pppoe ppp_async iptable_nat batman_adv ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_CT pppox ppp_mppe ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_socket xt_policy xt_multiport xt_mark xt_mac xt_limit xt_esp xt_comment xt_TPROXY xt_TCPMSS xt_LOG ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm tcp_bbr slhc nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 macvlan libcrc32c iptable_raw iptable_mangle iptable_filter ipt_ah ip_tables hwmon crc_ccitt compat
[   36.745894]  asn1_decoder ledtrig_usbport cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 tunnel6 tunnel4 tun xfrm_user xfrm_ipcomp af_key xfrm_algo vfat fat nls_utf8 nls_iso8859_1 nls_cp437 shortcut_fe_ipv6 shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic sha1_arm_neon sha1_arm md5 ghash_generic ghash_arm_ce gf128mul gcm echainiv cmac ccm authenc arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 exfat(C) antfs crc32c_generic
[   36.901884] CPU: 1 PID: 1868 Comm: batctl Tainted: G         C        5.4.162 #0
[   36.924113] Hardware name: Generic DT based system
[   36.931708] PC is at batadv_tt_local_add+0x38c/0x794 [batman_adv]
[   36.936263] LR is at batadv_tt_local_add+0x2e8/0x794 [batman_adv]
[   36.942407] pc : [<bf7f2ea8>]    lr : [<bf7f2e04>]    psr: 60000093
[   36.948483] sp : dd19d9c0  ip : ddee76f8  fp : c0c03d00
[   36.954574] r10: dd281000  r9 : 00000000  r8 : 000002c0
[   36.959768] r7 : 00000000  r6 : df4dc040  r5 : 00000001  r4 : 00000000
[   36.964998] r3 : 00000000  r2 : 00000000  r1 : 1f006000  r0 : 60000013
[   36.971594] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
[   36.978112] Control: 10c5387d  Table: 9d23406a  DAC: 00000051
[   36.985313] Process batctl (pid: 1868, stack limit = 0x6ad728eb)
[   36.991126] Stack: (0xdd19d9c0 to 0xdd19e000)
[   36.997209] d9c0: ddee7000 c09cb270 c0c04f7c c0c04e70 ddee74c0 dd282b00 dd281b00 00000000
[   37.001465] d9e0: 00000000 00000000 ddee7000 c0c2d5ec c0c28380 00000000 00000005 00000000
[   37.009606] da00: ddee74c0 00000000 dd37d11c ddee7568 00000000 ddee7000 dd8dae00 bf7efa84
[   37.017805] da20: 00000000 c07c9ae0 ddee7000 00000005 ddee7000 00000000 dd19dab0 00000005
[   37.025945] da40: 00000000 bf7e61d4 dd19da48 dd19da48 ddee7000 a28d91cf c0850b84 ffffffe9
[   37.034105] da60: bf61d020 00000000 dd19dab0 00000005 00000000 ddee7000 dd8dae00 c033ee18
[   37.042272] da80: ddee7000 00000001 00000000 00000000 dd19dd5c c033ef0c 00000000 ddee7000
[   37.050425] daa0: 00000001 c073ef60 dd19dd5c c0c28380 ddee7000 00000000 bf7fb0c8 c0c28380
[   37.058585] dac0: bf7ee98c c0c28380 bf7fb0c8 c0754e54 dd19dd5c dd19dd5c dd8dae00 00000000
[   37.066770] dae0: c09b9184 dd49a210 c09b923c c0c28380 00000000 c0c04e48 dd8dae00 ddc8ce40
[   37.074912] db00: 00000000 00000000 00000000 00000000 dd49a230 dd49a23c 00000000 00000000
[   37.083070] db20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   37.091222] db40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   37.099377] db60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   37.107540] db80: 00000000 00000000 00000000 00000000 00000001 c0c2e780 00000008 c0c56a08
[   37.115703] dba0: 00000000 c0423560 dd19dca4 00000201 dff7cf38 dff7cf14 00000000 00000000
[   37.123880] dbc0: 00000000 dd49a220 00000000 00000000 00000000 00000000 00000000 00000000
[   37.132029] dbe0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   37.140182] dc00: dd49a22c 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   37.148337] dc20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   37.156526] dc40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   37.164657] dc60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   37.172801] dc80: 00000000 00000000 30746162 00000000 00000000 00000000 61746162 00007664
[   37.180960] dca0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   37.189120] dcc0: 00000000 00000000 00000000 00000000 00000000 a28d91cf dd19dd88 dd8dae00
[   37.197280] dce0: ddc8ce40 dd49a200 dd19dd5c ddc8ce40 dd19dd5c c0c6648c 00000000 c0755024
[   37.205441] dd00: 00000000 df4c1cc0 dd49a200 c0c6648c ddc8ce40 c074f4f0 c0c2efc0 dfb4a0c0
[   37.213600] dd20: df437000 c05715a8 df437000 c077b1e8 c0c28380 ddc8ce40 c0c04e48 c074f2c8
[   37.221758] dd40: dd49a200 00000040 dda6e400 00000000 00000000 c077d598 c0778fe0 00000000
[   37.229917] dd60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 a28d91cf
[... 84 lines cut ...]
[   37.883066] [<c04b42a8>] (proc_sys_write) from [<c0442fa4>] (__vfs_write+0x1c/0x1a4)
[   37.891484] [<c0442fa4>] (__vfs_write) from [<c0444f1c>] (vfs_write+0xa4/0x188)
[   37.899294] [<c0444f1c>] (vfs_write) from [<c0445144>] (ksys_write+0x48/0xb4)
[   37.906326] [<c0445144>] (ksys_write) from [<c0301000>] (ret_fast_syscall+0x0/0x54)
[   37.913608] Exception stack(0xdd17dfa8 to 0xdd17dff0)
[   37.921076] dfa0:                   00000000 00000000 00000012 0002fac6 00000001 00000000
[   37.926290] dfc0: 00000000 00000000 00000012 00000004 b6ec36c0 00000000 b6e31a13 b6e31a05
[   37.934446] dfe0: bef019a8 bef01998 b6f454cc b6f44988
[   37.942603] CPU3: stopping
[   37.947634] CPU: 3 PID: 0 Comm: swapper/3 Tainted: G      D  C        5.4.162 #0
[   37.950242] Hardware name: Generic DT based system
[   37.957800] [<c030e524>] (unwind_backtrace) from [<c030afa8>] (show_stack+0x10/0x14)
[   37.962402] [<c030afa8>] (show_stack) from [<c08773e4>] (dump_stack+0x94/0xa8)
[   37.970301] [<c08773e4>] (dump_stack) from [<c030d7d8>] (handle_IPI+0x164/0x19c)
[   37.977332] [<c030d7d8>] (handle_IPI) from [<c05bb660>] (gic_handle_irq+0x8c/0x90)
[   37.984884] [<c05bb660>] (gic_handle_irq) from [<c0301a8c>] (__irq_svc+0x6c/0x90)
[   37.992255] Exception stack(0xdf475f60 to 0xdf475fa8)
[   37.999810] 5f60: 00000000 0000ea04 dfb652b4 c0313c40 ffffe000 c0c04e68 c0c04ea8 00000008
[   38.004851] 5f80: 00000000 c0b3fe28 c0c0dfc0 00000000 1f024000 df475fb0 c03083c0 c03083c4
[   38.013003] 5fa0: 60000013 ffffffff
[   38.021164] [<c0301a8c>] (__irq_svc) from [<c03083c4>] (arch_cpu_idle+0x38/0x3c)
[   38.024471] [<c03083c4>] (arch_cpu_idle) from [<c0349e30>] (do_idle+0x1e4/0x21c)
[   38.032108] [<c0349e30>] (do_idle) from [<c034a0f8>] (cpu_startup_entry+0x18/0x1c)
[   38.039483] [<c034a0f8>] (cpu_startup_entry) from [<803024ac>] (0x803024ac)
[   38.061083] Rebooting in 3 seconds..

MortalM, 2021-12-11:
那么如何回退至2021.3呢大佬。

ntlf9t, 2021-12-11:
feeds.conf.default
修改
src-git routing https://git.openwrt.org/feed/routing.git^a7b79a1d4fce4f443854115ee4b2ed2aafba5c06

MortalM, 2021-12-13:
经过测试ec33ebf0f638dcf793d13a9d88dc860f06a677e1的2021.4的也可以使用，但是不能用最新的master分支，否则无限重启

zhq1, 2021-12-17:
我说怎么弄了之后一开就死机，无限循环 还以为是easymesh的问题

xmbnl, 2022-02-04:
> feeds.conf.default 修改 src-git routing [https://git.openwrt.org/feed/routing.git^a7b79a1d4fce4f443854115ee4b2ed2aafba5c06](https://git.openwrt.org/feed/routing.git%5Ea7b79a1d4fce4f443854115ee4b2ed2aafba5c06)

无法解析

haohaoget, 2022-02-19:
> 经过测试ec33ebf0f638dcf793d13a9d88dc860f06a677e1的2021.4的也可以使用，但是不能用最新的master分支，否则无限重启

请问使用的哪个分支呢？我现在还是一开启easymesh就无限重启

loydcheung, 2022-03-09:
请问现在最新的源解决了吗？还是只能用之前版本的batman？

zhq1, 2022-03-09:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

zhengmz, 2022-04-27:
@ntlf9t 大侠，目前有什么新变化，还是只能用 2021-3 版本吗？

我使用的是 mt7621，使用 batman-adv 2022.0 版本可以编译通过，但配置完成后，互相看不到对应的 node

ayancn, 2022-06-04:
kmod-batman-adv | 5.10.100+2022.0-1
    这个版本还是有同样问题打开简单mesh保存设置就无限重启,能解决吗

alps2006, 2022-09-21:
目前我使用最新 master 源码编译, 第一次编译的简洁版本开启`简单mesh` 使用并没有问题, 后面陆续增加插件编译, 确实会出现开启就卡死的情况.  

经过编译插件信息对比发现应该是 `adbyby-plus` 插件与 `ssr-plus` 插件所依赖的 `adbyby` 冲突导致的,  把 `adbyby-plus` 插件替换为 `adblock` 即解决.

loydcheung, 2022-09-21:
> 目前我使用最新 master 源码编译, 第一次编译的简洁版本开启`简单mesh` 使用并没有问题, 后面陆续增加插件编译, 确实会出现开启就卡死的情况.  
> 
> 
> 
> 经过编译插件信息对比发现应该是 `adbyby-plus` 插件与 `ssr-plus` 插件所依赖的 `adbyby` 冲突导致的,  把 `adbyby-plus` 插件替换为 `adblock` 即解决.  
> 
> 

这个信息很有用，感谢大佬反馈测试结果。

haohaoget, 2022-09-21:
> 目前我使用最新 master 源码编译, 第一次编译的简洁版本开启`简单mesh` 使用并没有问题, 后面陆续增加插件编译, 确实会出现开启就卡死的情况.
> 
> 经过编译插件信息对比发现应该是 `adbyby-plus` 插件与 `ssr-plus` 插件所依赖的 `adbyby` 冲突导致的, 把 `adbyby-plus` 插件替换为 `adblock` 即解决.

请问ssr-plus可以用passwall之类的替换吗？

closed 2022-09-21 by ntlf9t