coolsnowwolf/lede #8418: g++: error: unrecognized command-line option '-m32'

CMLNT, 2021-12-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
使用的是MBP M1MAX 
ARM ubuntu2004 server
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
安装了大部分依赖包 部分gcc部分的依赖包安装不上
看日志好像百度说是-32 已经不再使用 
但是找不到具体是哪个sh包含-32 还没尝试删除
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：arm-openwrt-linux-muslgnueabi-gcc-ar crsT /home/cmlnt/wrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.18.1/out/Release/obj.target/libnode.a @/home/cmlnt/wrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.18.1/out/Release/obj.target/libnode.a.ar-file-list
  touch /home/cmlnt/wrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.18.1/out/Release/obj.target/node_dtrace_header.stamp
  touch /home/cmlnt/wrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.18.1/out/Release/obj.target/node_dtrace_ustack.stamp
  touch /home/cmlnt/wrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.18.1/out/Release/obj.target/node_dtrace_provider.stamp
  touch /home/cmlnt/wrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.18.1/out/Release/obj.host/tools/v8_gypfiles/v8_version.stamp
  touch /home/cmlnt/wrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.18.1/out/Release/obj.host/tools/v8_gypfiles/v8_headers.stamp
  g++ -m32 -o /home/cmlnt/wrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.18.1/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o ../deps/v8/src/base/bits.cc '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-D__STDC_FORMAT_MACROS' '-DV8_TARGET_ARCH_ARM' '-DCAN_USE_ARMV7_INSTRUCTIONS' '-DCAN_USE_VFP3_INSTRUCTIONS' '-DCAN_USE_VFP32DREGS' '-DCAN_USE_NEON' '-DV8_EMBEDDER_STRING="-node.84"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DOBJECT_PRINT' '-DV8_CONCURRENT_MARKING' '-DV8_ARRAY_BUFFER_EXTENSION' '-DV8_ENABLE_LAZY_SOURCE_POSITIONS' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' '-DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH' '-DV8_SNAPSHOT_COMPRESSION' '-DBUILDING_V8_BASE_SHARED' '-DUSE_EABI_HARDFLOAT=1' -I/home/cmlnt/wrt/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include -I../deps/v8 -I../deps/v8/include  -pthread -Wno-unused-parameter -Wno-return-type -fno-strict-aliasing -m32 -O3 -fno-omit-frame-pointer -fdata-sections -ffunction-sections -O3 -fno-rtti -fno-exceptions -std=gnu++1y -MMD -MF /home/cmlnt/wrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.18.1/out/Release/.deps//home/cmlnt/wrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.18.1/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o.d.raw   -c

g++: error: unrecognized command-line option '-m32'
g++: error: unrecognized command-line option '-m32'

make[5]: *** [tools/v8_gypfiles/v8_libbase.host.mk:159: /home/cmlnt/wrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.18.1/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o] Error 1
make[4]: *** [Makefile:109: node] Error 2
make[4]: Leaving directory '/home/cmlnt/wrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.18.1'
make[3]: *** [Makefile:204: /home/cmlnt/wrt/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/node-v14.18.1/.built] Error 2
make[3]: Leaving directory '/home/cmlnt/wrt/lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#106.02#10.26#113.36
    ERROR: package/feeds/packages/node failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/node/compile] Error 1
make[2]: Leaving directory '/home/cmlnt/wrt/lede'
make[1]: *** [package/Makefile:110: /home/cmlnt/wrt/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/cmlnt/wrt/lede'
make: *** [/home/cmlnt/wrt/lede/include/toplevel.mk:230: world] Error 2

zcc0030, 2021-12-31:
俺也一样，貌似目前arm平台还不能编译，老老实实用x86平台编译吧

Ra2-IFV, 2022-01-01:
ARM...（托腮）
用ARM编译会缺很多东西，墙裂建议用x86平台，如果没有电脑就嫖台云服务器吧

closed 2022-01-24 by AmadeusGhost