coolsnowwolf/lede #8419: xtables-addons 在线编译报错

Dimitri2020007, 2021-12-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：xtables-addons 在线编译报错

### (2) 路由器型号和固件版本

A：x86_64-efi

### (3) 详细日志

A：是不是占用时间太长被取消编译了？

2021-12-11T16:36:58.0544220Z make[3]: Entering directory '/workdir/openwrt/feeds/packages/net/xtables-addons'
2021-12-11T16:36:58.2271428Z mkdir -p /workdir/openwrt/dl
2021-12-11T16:36:58.2311855Z SHELL= flock /workdir/openwrt/tmp/.xtables-addons-3.13.tar.xz.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "xtables-addons-3.13.tar.xz" "893c0c4ea09759cda1ab7e68f1281d125e59270f7b59e446204ce686c6a76d65" "" "https://inai.de/files/xtables-addons/"    '
2021-12-11T16:36:58.2692584Z touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/.prepared_6a45badc62ff54aee3794999e845212c_6664517399ebbbc92a37c5bb081b5c53_check
2021-12-11T16:36:58.2718823Z . /workdir/openwrt/include/shell.sh; xzcat /workdir/openwrt/dl/xtables-addons-3.13.tar.xz | tar -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/.. -xf -
2021-12-11T16:36:58.3023448Z [ ! -d ./src/ ] || cp -fpR ./src/. /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13
2021-12-11T16:36:58.3083027Z 
2021-12-11T16:36:58.3084502Z Applying ./patches/001-fix-kernel-version-detection.patch using plaintext: 
2021-12-11T16:36:58.3097016Z patching file configure.ac
2021-12-11T16:36:58.3101045Z 
2021-12-11T16:36:58.3102393Z Applying ./patches/100-add-rtsp-conntrack.patch using plaintext: 
2021-12-11T16:36:58.3114597Z patching file extensions/rtsp/Kbuild
2021-12-11T16:36:58.3117207Z patching file extensions/rtsp/netfilter_helpers.h
2021-12-11T16:36:58.3121105Z patching file extensions/rtsp/netfilter_mime.h
2021-12-11T16:36:58.3123731Z patching file extensions/rtsp/nf_conntrack_rtsp.c
2021-12-11T16:36:58.3127937Z patching file extensions/rtsp/nf_conntrack_rtsp.h
2021-12-11T16:36:58.3130437Z patching file extensions/rtsp/nf_nat_rtsp.c
2021-12-11T16:36:58.3134324Z patching file extensions/Kbuild
2021-12-11T16:36:58.3137575Z patching file mconfig
2021-12-11T16:36:58.3141316Z 
2021-12-11T16:36:58.3142658Z Applying ./patches/200-add-lua-packetscript.patch using plaintext: 
2021-12-11T16:36:58.3168189Z patching file extensions/LUA/byte_array.c
2021-12-11T16:36:58.3171094Z patching file extensions/LUA/controller.c
2021-12-11T16:36:58.3175148Z patching file extensions/LUA/controller.h
2021-12-11T16:36:58.3178145Z patching file extensions/LUA/Kbuild
2021-12-11T16:36:58.3180428Z patching file extensions/LUA/libxt_LUA.c
2021-12-11T16:36:58.3183347Z patching file extensions/LUA/libxt_LUA.man
2021-12-11T16:36:58.3186736Z patching file extensions/LUA/lua/include/ctype.h
2021-12-11T16:36:58.3189226Z patching file extensions/LUA/lua/include/errno.h
2021-12-11T16:36:58.3191568Z patching file extensions/LUA/lua/include/locale.h
2021-12-11T16:36:58.3193823Z patching file extensions/LUA/lua/include/setjmp.h
2021-12-11T16:36:58.3196093Z patching file extensions/LUA/lua/include/stdio.h
2021-12-11T16:36:58.3198547Z patching file extensions/LUA/lua/include/stdlib.h
2021-12-11T16:36:58.3200970Z patching file extensions/LUA/lua/include/string.h
2021-12-11T16:36:58.3203125Z patching file extensions/LUA/lua/lapi.c
2021-12-11T16:36:58.3207658Z patching file extensions/LUA/lua/lapi.h
2021-12-11T16:36:58.3210139Z patching file extensions/LUA/lua/lauxlib.c
2021-12-11T16:36:58.3213986Z patching file extensions/LUA/lua/lauxlib.h
2021-12-11T16:36:58.3216700Z patching file extensions/LUA/lua/lbaselib.c
2021-12-11T16:36:58.3220356Z patching file extensions/LUA/lua/lcode.c
2021-12-11T16:36:58.3224291Z patching file extensions/LUA/lua/lcode.h
2021-12-11T16:36:58.3226810Z patching file extensions/LUA/lua/ldebug.c
2021-12-11T16:36:58.3230388Z patching file extensions/LUA/lua/ldebug.h
2021-12-11T16:36:58.3232743Z patching file extensions/LUA/lua/ldo.c
2021-12-11T16:36:58.3236044Z patching file extensions/LUA/lua/ldo.h
2021-12-11T16:36:58.3239288Z patching file extensions/LUA/lua/ldump.c
2021-12-11T16:36:58.3241980Z patching file extensions/LUA/lua/lfunc.c
2021-12-11T16:36:58.3244560Z patching file extensions/LUA/lua/lfunc.h
2021-12-11T16:36:58.3246773Z patching file extensions/LUA/lua/lgc.c
[... 41 lines cut ...]
2021-12-11T16:36:58.3373196Z patching file extensions/LUA/prot_buf_raw.c
2021-12-11T16:36:58.3375456Z patching file extensions/LUA/prot_buf_tcp.c
2021-12-11T16:36:58.3378047Z patching file extensions/LUA/prot_buf_tftp.c
2021-12-11T16:36:58.3380362Z patching file extensions/LUA/prot_buf_udp.c
2021-12-11T16:36:58.3382590Z patching file extensions/LUA/xt_LUA.h
2021-12-11T16:36:58.3384749Z patching file extensions/LUA/xt_LUA_target.c
2021-12-11T16:36:58.3387788Z patching file extensions/Kbuild
2021-12-11T16:36:58.3390853Z patching file extensions/Mbuild
2021-12-11T16:36:58.3393763Z patching file mconfig
2021-12-11T16:36:58.3398722Z 
2021-12-11T16:36:58.3400025Z Applying ./patches/201-fix-lua-packetscript.patch using plaintext: 
2021-12-11T16:36:58.3424255Z patching file extensions/LUA/xt_LUA_target.c
2021-12-11T16:36:58.3425571Z patching file extensions/LUA/lua/llimits.h
2021-12-11T16:36:58.3426326Z patching file extensions/LUA/lua/lapi.c
2021-12-11T16:36:58.3427062Z patching file extensions/LUA/lua/ltable.c
2021-12-11T16:36:58.3427863Z patching file extensions/LUA/lua/luaconf.h
2021-12-11T16:36:58.3428726Z patching file extensions/LUA/lua/llex.h
2021-12-11T16:36:58.3429970Z 
2021-12-11T16:36:58.3430963Z Applying ./patches/210-freebsd-build-fix.patch using plaintext: 
2021-12-11T16:36:58.3447695Z patching file extensions/LUA/Makefile
2021-12-11T16:36:58.3476113Z touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/.prepared_6a45badc62ff54aee3794999e845212c_6664517399ebbbc92a37c5bb081b5c53
2021-12-11T16:36:58.3501968Z rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13/.configured_*
2021-12-11T16:36:58.3533148Z rm -f /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.xtables-addons_installed
2021-12-11T16:36:58.3575368Z (cd /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.13; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /workdir/openwrt/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/workdir/openwrt/staging_dir/host/bin/autom4te AUTOCONF=/workdir/openwrt/staging_dir/host/bin/autoconf AUTOMAKE=/workdir/openwrt/staging_dir/host/bin/automake ACLOCAL=/workdir/openwrt/staging_dir/host/bin/aclocal AUTOHEADER=/workdir/openwrt/staging_dir/host/bin/autoheader LIBTOOLIZE=/workdir/openwrt/staging_dir/host/bin/libtoolize LIBTOOL=/workdir/openwrt/staging_dir/host/bin/libtool M4=/workdir/openwrt/staging_dir/host/bin/m4 AUTOPOINT=true LIBTOOLIZE='/workdir/openwrt/staging_dir/host/bin/libtoolize --install' /workdir/openwrt/staging_dir/host/bin/autoreconf -v -f -i -s -B /workdir/openwrt/staging_dir/host/share/aclocal -I /workdir/openwrt/staging_dir/target-x86_64_musl/host/share/aclocal -I /workdir/openwrt/staging_dir/hostpkg/share/aclocal -I /workdir/openwrt/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . . || true; fi; );
2021-12-11T16:36:58.5215770Z ##[error]The operation was canceled.

i4210577, 2021-12-12:
make[3]: Entering directory '/workdir/openwrt/package/lean/default-settings'
rm -rf /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/default-settings-2/.pkgdir/default-settings.installed /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/default-settings-2/.pkgdir/default-settings
mkdir -p /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/default-settings-2/.pkgdir/default-settings
install -d -m0755 /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/default-settings-2/.pkgdir/default-settings/etc/uci-defaults
install -m0755 ./files/zzz-default-settings /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/default-settings-2/.pkgdir/default-settings/etc/uci-defaults/99-default-settings
install -d -m0755 /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/default-settings-2/.pkgdir/default-settings/usr/lib/lua/luci/i18n
po2lmo ./po/zh-cn/default.po /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/default-settings-2/.pkgdir/default-settings/usr/lib/lua/luci/i18n/default.zh-cn.lmo
bash: po2lmo: command not found
make[3]: *** [Makefile:44: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/default-settings-2/.pkgdir/default-settings.installed] Error 127
make[3]: Leaving directory '/workdir/openwrt/package/lean/default-settings'
time: package/lean/default-settings/compile#0.14#0.04#0.17
    ERROR: package/lean/default-settings failed to build.
make[2]: *** [package/Makefile:116: package/lean/default-settings/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

suyuan168, 2021-12-29:
同样  也出错了

coolsnowwolf, 2026-09-06:
xtables-addons 日志在下载/编译中截断，没有实际失败点，无法判断是超时还是编译错误，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf