coolsnowwolf/lede #8420: x86版对i350支持有问题

freesilent, 2021-12-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：7月份开始编译的x86版本无法正常使用，web更新之后无论是否保存设置，都是webui无法进入，ssh也无法进入。

### (2) 路由器型号和固件版本

A：跑在esxi6.7里的x86版，

### (3) 详细日志

A：刚才用12月5日新编译的版本尝试了一下，web刷新，无论是否保存配置，在虚拟的op里加入两个网卡都会造成webui无法进入，ssh也无法进入，esxi的webui也会卡住，ssh无法进入，只能通过ipmi界面重启esxi。
后来尝试直接上传vmdk文件，只加入一个vm network，成功进入op，可以操作。
测试1，使用vm network作为lan，vm network2（i350-t4网卡）作为wan，可以正常使用。
测试2，使用vm network作为lan，直通的i350网口作为wan，浮现上述问题，webui无法进入，ssh无法进入，esxi同样，无法看到日志。只能通过ipmi重启esxi。
原先配置，esxi6.7，Intel(R) Xeon(R) CPU E5-2630L v3，超微x10dri 
直通自带的i350网口1个作为wan，另一个网口给esxi作为管理端口和lan。分1个核，1g内存。
版本：OpenWrt  R21.4.18 / LuCI Master (git-21.114.56541-974fb04)
使用两个vm network（不同网口）没有问题
使用一个vm network，一个直通的i350网口 有问题
使用一个vm network 没问题

ricksuzade-maker, 2021-12-12:
esxi 6.7有没有升级最新的升级补丁？

freesilent, 2021-12-13:
> esxi 6.7有没有升级最新的升级补丁？

刚才升级了最新版esxi6.7，6.7.0 Update 3 (Build 18828794)。问题依然出现。
5月份编译的没有问题的op
`root@OpenWrt:~# dmesg | grep eth1
[    6.574701] igb 0000:0b:00.0: added PHC on eth1
[    6.578065] igb 0000:0b:00.0: eth1: (PCIe:5.0Gb/s:unknown) 0c:c4:7a:ab:83:d6
[    6.579887] igb 0000:0b:00.0: eth1: PBA No: 070B00-000
[    8.887274] 8021q: adding VLAN 0 to HW filter on device eth1
[   12.358757] igb 0000:0b:00.0 eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[   12.363000] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
root@OpenWrt:~# dmesg | grep eth0
[    0.326400] vmxnet3 0000:15:00.0 eth0: NIC Link is Up 10000 Mbps
[    1.432130] vmxnet3 0000:15:00.0 eth0: intr type 3, mode 0, 2 vectors allocated
[    1.433514] vmxnet3 0000:15:00.0 eth0: NIC Link is Up 10000 Mbps
[    1.434546] 8021q: adding VLAN 0 to HW filter on device eth0
[    8.820640] vmxnet3 0000:15:00.0 eth0: intr type 3, mode 0, 2 vectors allocated
[    8.821583] vmxnet3 0000:15:00.0 eth0: NIC Link is Up 10000 Mbps
[    8.822337] 8021q: adding VLAN 0 to HW filter on device eth0
[    8.823236] br-lan: port 1(eth0) entered blocking state
[    8.823770] br-lan: port 1(eth0) entered disabled state
[    8.824396] device eth0 entered promiscuous mode
[    8.831577] br-lan: port 1(eth0) entered blocking state
[    8.832111] br-lan: port 1(eth0) entered forwarding state`

eth0是虚拟的vmxnet3，eth1是直通的i350
`root@OpenWrt:~# dmesg | grep net
[    0.324057] VMware vmxnet3 virtual NIC driver - version 1.4.17.0-k-NAPI
[    0.324336] vmxnet3 0000:15:00.0: # of Tx queues : 1, # of Rx queues : 1
[    0.324672] vmxnet3 0000:15:00.0: enabling device (0000 -> 0003)
[    0.326400] vmxnet3 0000:15:00.0 eth0: NIC Link is Up 10000 Mbps
[    0.326858] hv_vmbus: registering driver hv_netvsc
[    1.432130] vmxnet3 0000:15:00.0 eth0: intr type 3, mode 0, 2 vectors allocated
[    1.433514] vmxnet3 0000:15:00.0 eth0: NIC Link is Up 10000 Mbps
[    6.496208] Initializing XFRM netlink socket
[    6.507407] igb: Intel(R) Gigabit Ethernet Network Driver - version 5.6.0-k
[    6.576341] igb 0000:0b:00.0: Intel(R) Gigabit Ethernet Network Connection
[    6.589244] Ethernet Channel Bonding Driver: v3.7.1 (April 27, 2011)
[    6.609525] 8139too: 8139too Fast Ethernet driver 0.9.28
[    6.618064] i40e: Intel(R) Ethernet Connection XL710 Network Driver - version 2.8.20-k
[    6.619731] Intel(R) 2.5G Ethernet Linux Driver - version 0.0.1-k
[    6.637022] pcnet32: pcnet32.c:v1.35 21.Apr.2008 tsbogend@alpha.franken.de
[    8.820640] vmxnet3 0000:15:00.0 eth0: intr type 3, mode 0, 2 vectors allocated
[    8.821583] vmxnet3 0000:15:00.0 eth0: NIC Link is Up 10000 Mbps`

下面是有问题的新编译的op，不知道怎么复制日志，只能截图。启动到eth0 ready后，页面卡住，然后10s后出现如下代码，继续卡住，然后断开，web无法访问，esxi webui无法访问。
`c2 e8 f7 f7 11 00 <0f> ob eb co 55 48 89 e5 41 57 49 89 d7 41 56 4d 89 c6 41 55 49 89
[ 17.972713] RSP: 0018: ffffc90000003e90 EFLAGS: 00010286
[ 17.974931] RAX: 0000000000000037 RBX: 0000000000000000 RCX: 00000000000(30006
[ 17.977625] RDX: 0000000000000007 RSI: 0000000000000002 RDI: ffff88803ec1c4d0
[ 17.978795] RBP: ffffc90000003ec0 R08: 0000000000000000 R09: 00000000000004e9
[ 17.980016] R10: 0000000000000001 R11: 0000000000000001 R12: ffff888800d7bc000
[ 17.981238] R13: ffff88800d7hc480 R14: 0000000000000008 R15: 0000000000000000
[ 17.982493] FS: 0000000000000000 (0000) GS:ffff88803ec00000(0000) knlGS:0000000000000000
[ 17.983659] CS: 0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 17.987288] CR2: 000000000041a4bb CR3: 000000000e2e4002 CR4: 00000000001606f0
[ 17.988516] Call Trace:
[ 17.989607] <IRQ>
[ 17.990668] ? dev_reset_queue. constprop.48+0x80/0x80 
[ 17.991865] call_timer_fn. isra.40+0x16/0x80
[ 17.993007] run_timer_softirq+0x164/0x180
[ 17.995093] ? lapic_next_deadline+0x27/0x30
[ 17.997727] ? clockevents_program_event+0x8b/0xf0
[ 17.998781] _do_softirq+0xc9/0x20e
[ 17.999805] irq_exit+0x6f/0x80>
[ 18.000887] smp_apic_timer_interrupt+0x5e/0x9o 
[ 18.001932] apic_timer_interrupt+0xf/0x20
[ 18.002898] </IRQ>
[ 18.003969] RIP: 0010:native_safe_halt+0x17/0x20 
[ 18.007261] Code: 00 a8 08 75 c3 e9 77 ff ff ff 90 90 90 90 90 90 90 90 90 55 8b 05 d9 e7 af 00 45 09 e5 85 c0 7e 07 0f 00 2d 
bb ec 52 00 fb f4 <5d> c3 0f If 80 00 00 00 00 55 8b 05 b9 e7 af 00 48 89 e5 85 c0 7e
[ 18.009484] RSP: 0018: ffffffff82203e40 EFLAGS: 00000246 ORIG_RAX: ffffffffffffff13
[ 18.010446] RAX: 0000000000000000  RBX: 0000000000000000  RCX: 0000000000000001
[ 18.011391] RDX: 0000000000007dde RSI: 0000000000000000 RDX: 0000000000000000
[ 18.012355] RBP: ffffffff82203e40 R08: ffff88803e8af340 R09: 0000000000000000
[ 18.013250] R10: 00000000000002d4 R11: 0000000000000000 R12: ffffffff82294900
[ 18.014147] R13: 0000000000000000 R14: 0000000000000000 R15: 000000000b96d54c
[ 18.017320] default_idle+0x9/0x10
[ 18.010365] arch_cpu_idle+0x10/0x20
[ 18.019158] default_idle_call+0x27/0x30
[ 18.019932] do_idle+0x19f/0x1c0
[ 18.020675] cpu_startup_entry+0x18/0x20
[ 18.021497] rest_init+0xa4/0xa6
[ 18.022201] arch_call_rest_init+0x9/0xc
[ 18.022095] start_kernel+0x559/0x567
[ 18.023622] x86_64_start_reservations+0x29/0x2b
[ 18.024375] x86_64_start_kernel+0x71/0x74 
[ 18.027333] secondary_startup_64+0xa4/0xb0
[ 18.028102] --[ end trace 1afc121b2a3867c3 ]--
[ 18.020002] igb 0000:13:00.0 eth1: Reset adapter
[ 18.219316] 8021q: adding VLAN 0 to HW filter on device eth1
[ 21.099104] igb 0000:13:00.0 eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX 
[ 21.906343] IPV6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready`

wi11iamzhao, 2021-12-15:
ESXI7.0U2，I350T2直通，近期版本正常（5.10内核）

ettwz, 2021-12-26:
esxi 6.7.0(Build 8019320)用主分支最新代码一样的问题 r3210-3403deeaf正常

shadowcovering, 2023-02-14:
问题解决了吗？我也出现一样的问题，有人说升级到esxi7.0就可以解决，我没条件测试，我板载一块r8168，esxi7.0无法驱动

freesilent, 2023-02-18:
目前没有问题了，使用tag 20221001代码编译可以两个网口，一个直通，一个vm network。使用最新代码编译也可以，但op不稳定，偶尔cpu占用率100%，此时网络中断，webui卡死，一段时间自行恢复。

coolsnowwolf, 2026-09-05:
报告来自 R21.4.18 的 ESXi 6.7 直通场景，宿主机/直通设备导致整台 ESXi 卡死，未提供当前 LEDE 内核日志或可复现配置；无法归因于当前仓库代码，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf