coolsnowwolf/lede #8423: Dokerd 校验码更新导致编译错误

xinb, 2021-12-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
Dokerd 校验码更新导致编译错误

`( DEP_VER=$( grep --only-matching --perl-regexp '(?<=PKG_SOURCE_VERSION:=)(.*)' "../runc/Makefile" ); VEN_VER=$( grep --only-matching --perl-regexp '(?<=_COMMIT:=)(.*)(?=})' "/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/build_dir/target-x86_64_musl/dockerd-20.10.11/hack/dockerfile/install/runc.installer" ); if [ "${VEN_VER}" != "${DEP_VER}" ]; then echo "ERROR: Expected 'PKG_SOURCE_VERSION:=${VEN_VER}' in '../runc/Makefile', found 'PKG_SOURCE_VERSION:=${DEP_VER}'"; exit 1; fi )
ERROR: Expected 'PKG_SOURCE_VERSION:=52b36a2dd837e8462de8e01458bf02cf9eea47dd' in '../runc/Makefile', found 'PKG_SOURCE_VERSION:=f46b6ba2c9314cfc8caae24a32ec5fe9ef1059fe'`

WYC-2020, 2021-12-13:
这docker 真坑，动不动就更新然后不匹配，不能想另外的库那样，直接配套

WYC-2020, 2021-12-13:
他这个是监测runc 版本是1.0.3了然后就报这个问题，但是这个dockerd 20.10.11 代码里用的是1.0.2才对，不知道为啥他会去 取到1.0.3的commit 这个有点见鬼，难道这个runc 每次都是要求的最新，但是看哪个run.install 别人明显就是checkout 1.0.2的，这个感觉可以用一个补丁修正这个问题
[image]

WYC-2020, 2021-12-13:
实测去掉哪个依赖版本检测 
[image]
可以编译

WYC-2020, 2021-12-13:
@xinb  是这个提交导致的
[image]
这个提交吧runc升级到了1.0.3 但是dockerd 是依赖1.0.2的 ，吧哪个revert 就好了 哪个提交

xyskwjy, 2021-12-13:
弱弱弱的问一下：云编绎的，怎么改？

kokeri, 2021-12-13:
> 弱弱弱的问一下：云编绎的，怎么改？

对openwrt/feeds/packages/utils/dockerd/Makefile 文件执行 sed -i 's/^\s*$[(]call\sEnsureVendoredVersion/#&/'

xyskwjy, 2021-12-13:
> > 弱弱弱的问一下：云编绎的，怎么改？
> 
> 对openwrt/feeds/packages/utils/dockerd/Makefile 文件执行 sed -i 's/^\s*$[(]call\sEnsureVendoredVersion/#&/'

谢谢大哥

xyskwjy, 2021-12-13:
> openwrt/feeds/packages/utils/dockerd/Makefile

# 去除dockerd校验
sed -i 's/^\s*$[(]call\sEnsureVendoredVersion/#&/g' feeds/packages/utils/dockerd/Makefile

先谢谢哥。不过还是不行。提示：
Run [ -e $FEEDS_CONF ] && mv $FEEDS_CONF openwrt/feeds.conf.default
Cloning into 'package/lean/luci-app-dnsfilter'...
Cloning into 'package/lean/luci-app-poweroff'...
sed: can't read feeds/packages/utils/dockerd/Makefile: No such file or directory
Error: Process completed with exit code 2.

xiangfeidexiaohuo, 2021-12-13:
提交的时候，没成想大雕直接就合了。

可以直接拉openwrt官方packages下的runc，还是1.0.2，还没合上去。

两行命令搞定：

rm -rf ./feeds/packages/utils/runc/Makefile

svn export https://github.com/openwrt/packages/trunk/utils/runc/Makefile ./feeds/packages/utils/runc/Makefile

chiyu521, 2021-12-16:
我尝试直接修改12行 别的什么都没改
PKG_SOURCE_VERSION:=f46b6ba2c9314cfc8caae24a32ec5fe9ef1059fe
PKG_SOURCE_VERSION:=52b36a2dd837e8462de8e01458bf02cf9eea47dd
貌似可以编译过去 你们也测试下

closed 2021-12-28 by AmadeusGhost