coolsnowwolf/lede #8424: Releae 21.11.11 海凌科HLK-MT7621开发板，按reboot后，无法重新启动，挂住

lovelypp777, 2021-12-13:
按reboot后，串口显示：
...
...
[ 1385.496866] reboot: Restarting system

然后串口终端没有任何输出，也不接受任何输入，系统应该是失去响应了，或者挂住了

谢谢！

lovelypp777, 2021-12-14:
回退到内核4.14，可以重新启动了，应该是内核的问题

PokersKun, 2021-12-14:
打个PATCH就行，这是32M的闪存芯片（如W25Q256）导致的：
```
--- a/drivers/mtd/spi-nor/spi-nor.c
+++ b/drivers/mtd/spi-nor/spi-nor.c
@@ -4840,8 +4840,7 @@
 void spi_nor_restore(struct spi_nor *nor)
 {
 	/* restore the addressing mode */
-	if (nor->addr_width == 4 && !(nor->flags & SNOR_F_4B_OPCODES) &&
-	    nor->flags & SNOR_F_BROKEN_RESET)
+	if (nor->addr_width == 4 && !(nor->flags & SNOR_F_4B_OPCODES))
 		nor->params.set_4byte(nor, false);
 }
 EXPORT_SYMBOL_GPL(spi_nor_restore);

lovelypp777, 2021-12-14:
按照您的提示解决了，谢谢！

AmadeusGhost, 2021-12-14:
5.4 内核上你需要在 dts 写上 `broken-flash-reset;`

closed 2021-12-15 by AmadeusGhost