coolsnowwolf/lede #8425: 优酷L1C交换机驱动ETDEV WATCHDOG: eth0 (ralink_soc_eth): transmit queue 0 timed out

chobitsos, 2021-12-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
好几个类似的ISSUE
https://github.com/coolsnowwolf/lede/issues/7960
https://github.com/coolsnowwolf/lede/issues/7665

### (1) 具体问题
A：流量大就transmit queue 0 timed out

### (2) 路由器型号和固件版本

A：YOUKUL1C

### (3) 详细日志

[ 6366.668795] ------------[ cut here ]------------
[ 6366.678047] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:473 dev_watchdog+0x26c/0x274
[ 6366.694535] NETDEV WATCHDOG: eth0 (ralink_soc_eth): transmit queue 0 timed out
[ 6366.708926] Modules linked in: xt_FULLCONENAT rt2800soc rt2800mmio rt2800lib pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT rt2x00soc rt2x00mmio rt2x00lib pppox ppp_mppe ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_socket xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TPROXY xt_TCPMSS xt_LOG ts_kmp ts_fsm ts_bm tcp_bbr slhc nf_tproxy_ipv4 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_log_common nf_defrag_ipv4 macvlan iptable_raw iptable_mangle iptable_filter ip_tables crc_ccitt compat asn1_decoder ledtrig_usbport cryptodev xt_set x_tables ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport
[ 6366.709193]  ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg hmac ghash_generic gf128mul gcm ecb ctr cmac ccm authenc arc4 mmc_block mtk_sd mmc_core leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug usbcore nls_base usb_common aead cryptomgr crypto_null crypto_hash
[ 6366.994696] CPU: 0 PID: 0 Comm: swapper Not tainted 5.4.162 #0
[ 6367.006320] Stack : 805f0000 80592d50 00000000 00000000 80591e48 87c0be1c 805cab00 805ca6e3
[ 6367.022982]         80534354 00000000 807233cc 805e0000 00200000 00000001 87c0bdd0 a975b3b8
[ 6367.039642]         00000000 00000000 80750000 00000000 00000030 000000c9 312e342e 23203236
[ 6367.056305]         00000000 0000013c 00000000 000f2d88 00000000 00000009 00000000 803d4a88
[ 6367.072962]         00000009 805e0000 00200000 00000122 00000001 802ef2f0 00000000 80720000
[ 6367.089621]         ...
[ 6367.094481] Call Trace:
[ 6367.099375] [<8000b3d0>] show_stack+0x30/0x100
[ 6367.108237] [<80025c18>] __warn+0xc0/0x10c
[ 6367.116411] [<80025cf0>] warn_slowpath_fmt+0x8c/0xac
[ 6367.126326] [<803d4a88>] dev_watchdog+0x26c/0x274
[ 6367.135735] [<80069e34>] call_timer_fn.isra.34+0x20/0x90
[ 6367.146330] [<8006a048>] run_timer_softirq+0x1a4/0x1ec
[ 6367.156583] [<804ae288>] __do_softirq+0x110/0x298
[ 6367.165963] [<80005e90>] except_vec_vi_end+0xb8/0xc4
[ 6367.175862] [<804ae054>] r4k_wait_irqoff+0x18/0x24
[ 6367.185415] ---[ end trace 2755d59ec8792744 ]---
[ 6367.194629] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
[ 6367.207473] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[ 6367.219979] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=068cc000, max=1024, ctx=14, dtx=14, fdx=1004, next=14
[ 6367.241656] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=068f4000, max=1024, calc=217, drx=699
[ 6369.723084] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
[ 6369.735957] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[ 6369.748479] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=068cc000, max=1024, ctx=556, dtx=556, fdx=240, next=556
[ 6369.770505] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=068fc000, max=1024, calc=324, drx=330
[ 6373.830837] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
[ 6373.843710] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065

chobitsos, 2021-12-13:
已解决，补丁在这里
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=f8b849103dfe1b21a6462837efc9460ca3cca02d
但是我不会push，我觉得可以commits一下

MortalM, 2021-12-13:
@chobitsos 老大 你这是7621的， 我7620的也有问题，能麻烦教我弄下吗 补丁怎么搜，然后在哪里修改完，编译才有用啊。。谢谢

chobitsos, 2021-12-13:
> @chobitsos 老大 你这是7621的， 我7620的也有问题，能麻烦教我弄下吗 补丁怎么搜，然后在哪里修改完，编译才有用啊。。谢谢

我在测试一下看看，没问题的话我传个补丁

chobitsos, 2021-12-13:
https://forum.openwrt.org/t/mtk-soc-eth-watchdog-timeout-after-r11573/50000/83

MortalM, 2021-12-13:
我也改了7620的配置，在测试了，没问题的话， 我留个言然后吧issue关了， 不过这种关闭某个服务的方式感觉还是没换交换机驱动靠谱啊， 不知道有什么办法可以换个稳定的驱动啊。。

MortalM, 2021-12-13:
我在7620上按照这个https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=f8b849103dfe1b21a6462837efc9460ca3cca02d，修改配置文件关闭了 TSO功能
现在不在出现time out超时，但是我的网速从百兆测速95 变成77左右了。。 几乎每次都跳到90多，然后掉到77左右。
大佬，你这里的补丁方式测试的怎么样https://forum.openwrt.org/t/mtk-soc-eth-watchdog-timeout-after-r11573/50000/83

chobitsos, 2021-12-13:
> 我在7620上按照这个https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=f8b849103dfe1b21a6462837efc9460ca3cca02d，修改配置文件关闭了 TSO功能
> 
> 依然还是会出现timeout的超时
> 
> 

只关TSO不能完全解决，我在对比大堆2021-06-24前的源码，因为我之前编译出来5.4.126的内核没有这个问题，还需要改动驱动文件，参考mt7621

MortalM, 2021-12-13:
那就等大佬作业了

chobitsos, 2021-12-13:
> 那就等大佬作业了

尝试了一晚，应该就是lean的坑了，因为我用隔壁的https://github.com/x-wrt/x-wrt编译同样的内核5.4.162是无论我怎么折腾都不会transmit timed out，我想换SDK了

MortalM, 2021-12-14:
上次我小米路由器3没有nand驱动的问题，也是x-wrt的作者ptpt52大神的代码解决的。 我去他哪儿看看。。

MortalM, 2021-12-14:
我在那边找到相关的问题。https://github.com/x-wrt/x-wrt/issues/2
x-wrt确实没有这个问题，但是我看不懂大佬究竟修复了什么？哎。。

chobitsos, 2021-12-14:
> 我在那边找到相关的问题。[x-wrt/x-wrt#2](https://github.com/x-wrt/x-wrt/issues/2) x-wrt确实没有这个问题，但是我看不懂大佬究竟修复了什么？哎。。

这个ISSUEES已关了不要回，昨晚把ptpt52的编译出来了，跑了一晚稳得一批，我就懒得折腾了，就用ptpt52的固件吧，我就只用一个科学插件而已

Ho-Lay, 2021-12-16:
@coolsnowwolf 大佬，可以帮忙修复一下吗

chobitsos, 2021-12-19:
估计是上游的问题，L大也无能为力，