coolsnowwolf/lede #8432: 最近mt7620和mt7621的多个机型都有大流量断流出现time out超时问题，希望大佬处理下

MortalM, 2021-12-14:
反馈bug/问题模板，提建议请删除

Q：是否搜索了issue (使用 "x" 选择)
* [ √] 没有类似的issue
#7960   #7665  #8407  #8425  
## 2. 详细叙述

### (1) 具体问题

使用联发科mt7620和mt7621的多个机器比如小米mini，小米3，优酷y/L1c，在大流量使用时，会出现kern.err kernel: [ 240.032056] ralink_soc_eth 10100000.ethernet eth0: transmit timed out的问题。 经过我查验发现旧版本的内核没有问题，然后x-wrt的5.4的内核也没有这个问题，希望大佬能处理下，非常感谢。

### (2) 路由器型号和固件版本

小米mini，小米3，优酷y/L1c

### (3) 详细日志

Thu Dec 9 19:39:15 2021 kern.err kernel: [ 275.058897] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Dec 9 19:39:15 2021 kern.info kernel: [ 275.065491] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Dec 9 19:39:15 2021 kern.info kernel: [ 275.071923] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=064d4000, max=1024, ctx=97, dtx=97, fdx=48, next=97
Thu Dec 9 19:39:15 2021 kern.info kernel: [ 275.082856] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=05194000, max=1024, calc=104, drx=105
Thu Dec 9 19:39:25 2021 kern.err kernel: [ 278.733921] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Dec 9 19:39:25 2021 kern.info kernel: [ 278.740516] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Dec 9 19:39:25 2021 kern.info kernel: [ 278.746952] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=064d4000, max=1024, ctx=33, dtx=33, fdx=16, next=33
Thu Dec 9 19:39:25 2021 kern.info kernel: [ 278.757898] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06434000, max=1024, calc=57, drx=59
Thu Dec 9 19:39:27 2021 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 4c:79:6e:a0:ef:e2
Thu Dec 9 19:39:31 2021 daemon.info hostapd: wlan0: STA 4c:79:6e:a0:ef:e2 IEEE 802.11: authenticated
Thu Dec 9 19:39:31 2021 daemon.info hostapd: wlan0: STA 4c:79:6e:a0:ef:e2 IEEE 802.11: associated (aid 4)
Thu Dec 9 19:39:31 2021 daemon.notice hostapd: wlan0: STA-OPMODE-SMPS-MODE-CHANGED 4c:79:6e:a0:ef:e2 off
Thu Dec 9 19:39:31 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 4c:79:6e:a0:ef:e2
Thu Dec 9 19:39:31 2021 daemon.info hostapd: wlan0: STA 4c:79:6e:a0:ef:e2 RADIUS: starting accounting session 813F38EA8D98ADB0
Thu Dec 9 19:39:31 2021 daemon.info hostapd: wlan0: STA 4c:79:6e:a0:ef:e2 WPA: pairwise key handshake completed (RSN)
Thu Dec 9 19:39:31 2021 daemon.notice hostapd: wlan0: EAPOL-4WAY-HS-COMPLETED 4c:79:6e:a0:ef:e2
Thu Dec 9 19:39:35 2021 kern.err kernel: [ 282.374864] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Dec 9 19:39:35 2021 kern.info kernel: [ 282.381468] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Dec 9 19:39:35 2021 kern.info kernel: [ 282.387924] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=064d4000, max=1024, ctx=22, dtx=22, fdx=16, next=22
Thu Dec 9 19:39:35 2021 kern.info kernel: [ 282.398860] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=05194000, max=1024, calc=35, drx=36
Thu Dec 9 19:45:18 2021 daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS 4c:79:6e:a0:ef:e2 2 ack=1
Thu Dec 9 19:45:18 2021 daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS 3a:6c:27:50:c3:5b 1 ack=1
Thu Dec 9 19:45:18 2021 daemon.notice hostapd: wlan0: BEACON-RESP-RX 4c:79:6e:a0:ef:e2 2 00

有时候又是重复的断连断连。。确定不是网线物理连接导致的断连。。

[ 509.808465] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[ 639.987096] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[ 640.834750] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[ 667.266157] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[ 668.248389] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[ 668.288946] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[ 669.191955] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[ 669.330269] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[ 670.277141] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[ 676.730269] ralink_soc_eth 10100000.ethernet eth0: port 0 link down
[ 676.851621] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[ 676.860917] br-lan: port 1(eth0.1) entered disabled state
[ 679.156010] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[ 679.164833] br-lan: port 1(eth0.1) entered blocking state
[ 679.170388] br-lan: port 1(eth0.1) entered forwarding state
[ 682.081774] ralink_soc_eth 10100000.ethernet eth0: port 1 link down
[ 682.091812] br-lan: port 1(eth0.1) entered disabled state
[ 684.226672] ralink_soc_eth 10100000.ethernet eth0: port 1 link up (100Mbps/Full duplex)
[ 684.235375] br-lan: port 1(eth0.1) entered blocking state
[ 684.240951] br-lan: port 1(eth0.1) entered forwarding state
[ 685.632235] ralink_soc_eth 10100000.ethernet eth0: port 0 link up (100Mbps/Full duplex)

Wilson-coder, 2021-12-15:
我的X86，升级到5.4.162内核也有这个现象，之前用5.4.155内核的时候从来没有遇到过。本来还以为是最近电信网络不稳定，看起来是路由器更新之后带来的问题？

chobitsos, 2021-12-15:
> 我的X86，升级到5.4.162内核也有这个现象，之前用5.4.155内核的时候从来没有遇到过。本来还以为是最近电信网络不稳定，看起来是路由器更新之后带来的问题？

换了x-wrt，跑了两天没有出现transmit timed out
我对比了x-wrt源码官方源码immortalwrt源码，在驱动文件里面有一处都有被改过的
[		.name = "ralink_soc_eth",](https://github.com/coolsnowwolf/lede/blob/1c6367234929e06f588625cca73bf7a9c110626f/target/linux/ramips/files/drivers/net/ethernet/ralink/mtk_eth_soc.c#L1741)
这个文件只有lean大神是这个值ralink_soc_eth，其他都一样，但是我不懂代码哈，没试过不知道是不是这个问题

MortalM, 2021-12-16:
> 换了x-wrt，跑了两天没有出现transmit timed out 我对比了x-wrt源码官方源码immortalwrt源码，在驱动文件里面有一处都有被改过的 [ .name = [...]

把这个指删了试试呢？

AmadeusGhost, 2021-12-17:
> 把这个指删了试试呢？

这个是驱动名字啊，别瞎删。。

slipdaly, 2021-12-20:
感觉是dns解析问题，我遇到这个问题，重启广告过滤（会重启dnsmasq）就恢复了。

Harvey365, 2022-02-04:
k2最新编译的固件也遇到了这个问题
9965.599103] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06a08000, max=1024, ctx=230, dtx=230, fdx=161, next=230
[ 9965.621120] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06a54000, max=1024, calc=874, drx=875
[ 9967.163335] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
[ 9967.176202] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[ 9967.188726] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06a08000, max=1024, ctx=316, dtx=316, fdx=176, next=316
[ 9967.210749] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06a58000, max=1024, calc=17, drx=18
[ 9968.462115] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
[ 9968.474980] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[ 9968.487503] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06a08000, max=1024, ctx=463, dtx=463, fdx=192, next=463
[ 9968.509528] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06a54000, max=1024, calc=276, drx=278
[ 9970.005436] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
[ 9970.018320] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[ 9970.030837] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06a08000, max=1024, ctx=123, dtx=123, fdx=48, next=123
[ 9970.052681] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06a58000, max=1024, calc=433, drx=435
[ 9971.550848] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
[ 9971.563719] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[ 9971.576242] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=06a08000, max=1024, ctx=27, dtx=27, fdx=16, next=27
[ 9971.597574] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06a54000, max=1024, calc=123, drx=124
[ 9973.207222] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
[ 9973.220089] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
[ 9973.232610] ralink_soc_eth 10100000

evfun, 2022-03-18:
我的hc5962也断流，十几分钟一次，一次20秒左右，系统日志里面没有任何提示，网卡状态也没有发生变化
系统版本OpenWrt R22.3.13 / LuCI Master (git-22.068.45502-a50e601)
内核版本5.4.175

chobitsos, 2022-05-22:
已换SDK
测试30分钟双向跑满没有断流，适配lean的packges仓库
[image]

[image]

[image]

coolsnowwolf, 2026-09-05:
这是多个旧 MT7620/MT7621 设备的汇总报告，日志均为旧 `ralink_soc_eth` watchdog，缺少统一复现条件；无法据此制作安全的通用修复，关闭为“不计划处理”。

closed 2026-09-05 by coolsnowwolf