coolsnowwolf/lede #8437: 3965U掉网口内核报错

lixiangyu1314, 2021-12-14:


Guxing0321, 2021-12-17:
这个问题，我之前也遇到过，当时表现是使用3-5天的样子，机器会自动关机

属于编译环境缺少一个底层依赖

我的解决办法如下：
sudo mkdir /lib/firmware/i915
sudo curl -L https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/i915/kbl_dmc_ver1_04.bin -o /lib/firmware/i915/kbl_dmc_ver1_04.bin

Guxing0321, 2021-12-17:
当时5.4的内核，现在应该是5.10

```

# 添加 'i915/kbl_dmc_ver1_04.bin' 到5.10的内核
echo 'CONFIG_FIRMWARE_IN_KERNEL=y' >> target/linux/x86/config-5.10
echo 'CONFIG_EXTRA_FIRMWARE="i915/kbl_dmc_ver1_04.bin"' >> target/linux/x86/config-5.10
echo 'CONFIG_EXTRA_FIRMWARE_DIR="/lib/firmware"' >> target/linux/x86/config-5.10
```

lixiangyu1314, 2021-12-19:
关键是小白不会啊

Guxing0321, 2021-12-20:
这个是用lean大佬的源码编译出来的固件，x86系列的，支持3965U这个系列的
https://github.com/Guxing0321/AutoBuild_Lede/releases/tag/2021.12.20-0806

bendfeng, 2021-12-20:
我的内核日记里面也有这个提示，单线多拨情况下，开起turbo acc隔个几天多拨就会掉线，并不会重启。但是把turbo acc关闭后没出现多拨掉线的问题

XinSSS, 2022-01-28:
3865U也有这个报错， 
CONFIG_PACKAGE_i915-firmware=y
已经勾选了，还是报错

XinSSS, 2022-01-28:
> 当时5.4的内核，现在应该是5.10
>
> ```
>
> # 添加 'i915/kbl_dmc_ver1_04.bin' 到5.10的内核
> echo 'CONFIG_FIRMWARE_IN_KERNEL=y' >> target/linux/x86/config-5.10
> echo 'CONFIG_EXTRA_FIRMWARE="i915/kbl_dmc_ver1_04.bin"' >> [...]

感谢， 你提供的这个很有用，问题解决了。。。
#8654 里说commit已经修复， 但是我尝试勾选重新编译升级后还是能看到那个错误提示，不知道问题在哪里

coolsnowwolf, 2026-09-06:
3965U 网口掉线只有图片，没有 NIC 型号、驱动和完整内核日志，无法定位，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf