coolsnowwolf/lede #8438: 请问 ttyd 如何在https下访问？

Azreulo, 2021-12-14:
请教一下，在#4635看到有兄弟是可以的。最新的patch里似乎有一个关于ca证书的修正，请问如何才能正常在https下使用ttyd？

WYC-2020, 2021-12-15:
开启相关选项不就完事的，这玩意我都不开https，开了很多其他下载工具https无法兼容，ttyd也问了作者不会同时支持http或者https运行，也就是说 你开了https你op也必须开https

VergilGao, 2021-12-16:
> 开启相关选项不就完事的，这玩意我都不开https，开了很多其他下载工具https无法兼容，ttyd也问了作者不会同时支持http或者https运行，也就是说 你开了https你op也必须开https

我之前一直听说ttyd用https会有问题所以一直没有用过，看你这么说如果我op本来就是https（用letsencrypt证书），就可以无影响的使用ttyd了呗

ZqinKing, 2021-12-17:
> 请教一下，在#4635看到有兄弟是可以的。最新的patch里似乎有一个关于ca证书的修正，请问如何才能正常在https下使用ttyd？

同问，不知道怎么开启ttyd的https

WYC-2020, 2021-12-17:
配置文件加入三个选项，一个ssl还有两个是ssl-什么忘记了，看ttyd服务有名字

Azreulo, 2021-12-20:
> 配置文件加入三个选项，一个ssl还有两个是ssl-什么忘记了，看ttyd服务有名字

能说得具体一些吗？

hello--world, 2021-12-20:
停掉之前的，自己起，官方的可以配置
```shell
#!/bin/sh /etc/rc.common

START=99

CONFIG=xxx-ttyd
LOG_PATH=/var/log/$CONFIG
LOG_APP_FILE=$LOG_PATH/app.log

echolog() {
	echo -e "$(date "+%Y-%m-%d %H:%M:%S"): $1" >> $LOG_APP_FILE
}

start() {
	mkdir -p  $LOG_PATH
    /usr/bin/ttyd --ssl --ssl-cert xxx.crt --ssl-key xxx.key /bin/login >> $LOG_APP_FILE &
}

stop() {
	killall ttyd
}

restart() {
	stop
	sleep 1
	start
}
```

Yulin0310, 2023-07-27:
同问

closed 2026-09-05 by coolsnowwolf