coolsnowwolf/lede #8446: 二次编译，报错

jy3344511, 2021-12-15:
Creating directory /usr/lib/python3.9/config-3.9
Creating directory /usr/lib/pkgconfig
/home/jy3344511/lede/staging_dir/host/bin/install -c -m 644 Modules/config.c /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/config-3.9/config.c
/home/jy3344511/lede/staging_dir/host/bin/install -c -m 644 Programs/python.o /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/config-3.9/python.o
/home/jy3344511/lede/staging_dir/host/bin/install -c -m 644 ./Modules/config.c.in /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/config-3.9/config.c.in
/home/jy3344511/lede/staging_dir/host/bin/install -c -m 644 Makefile /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/config-3.9/Makefile
/home/jy3344511/lede/staging_dir/host/bin/install -c -m 644 ./Modules/Setup /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/config-3.9/Setup
/home/jy3344511/lede/staging_dir/host/bin/install -c -m 644 Modules/Setup.local /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/config-3.9/Setup.local
/home/jy3344511/lede/staging_dir/host/bin/install -c -m 644 Misc/python.pc /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/pkgconfig/python-3.9.pc
/home/jy3344511/lede/staging_dir/host/bin/install -c -m 644 Misc/python-embed.pc /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/pkgconfig/python-3.9-embed.pc
/home/jy3344511/lede/staging_dir/host/bin/install -c ./Modules/makesetup /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/config-3.9/makesetup
/home/jy3344511/lede/staging_dir/host/bin/install -c ./install-sh /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/config-3.9/install-sh
/home/jy3344511/lede/staging_dir/host/bin/install -c python-config.py /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/config-3.9/python-config.py
/home/jy3344511/lede/staging_dir/host/bin/install -c python-config /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/bin/python3.9-config
_PYTHON_PROJECT_BASE=/home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5 _PYTHON_HOST_PLATFORM=linux-x86_64 PYTHONPATH=/home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/build/lib.linux-x86_64-3.9:./Lib _PYTHON_SYSCONFIGDATA_NAME=_sysconfigdata python3.9 ./setup.py install \
   	--prefix=/usr \
	--install-scripts=/usr/bin \
	--install-platlib=/usr/lib/python3.9/lib-dynload \
	--root=/home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/
running install
running build
running build_ext
INFO: Can't locate Tcl/Tk libs and/or headers

Python build finished successfully!
The necessary bits to build these optional modules were not found:
_tkinter              nis                                      
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

The following modules found by detect_modules() in setup.py, have been
built by the Makefile instead, as configured by the Setup files:
_abc                  atexit                pwd                
time                  zlib                                     

running build_scripts
copying /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/Tools/scripts/pydoc3 -> build/scripts-3.9
copying /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/Tools/scripts/idle3 -> build/scripts-3.9
copying /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/Tools/scripts/2to3 -> build/scripts-3.9
renaming build/scripts-3.9/pydoc3 to build/scripts-3.9/pydoc3.9
renaming build/scripts-3.9/idle3 to build/scripts-3.9/idle3.9
renaming build/scripts-3.9/2to3 to build/scripts-3.9/2to3-3.9
running install_lib
creating /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/cmath.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/spwd.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_curses_panel.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_blake2.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_testbuffer.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_ctypes_test.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_csv.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/xxlimited.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_lzma.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_queue.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_uuid.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_heapq.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_crypt.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_decimal.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_socket.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/array.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_bz2.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_testcapi.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/math.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/ossaudiodev.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_sha1.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/mmap.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_contextvars.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/pyexpat.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/grp.cpython-39.so -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
copying build/lib.linux-x86_64-3.9/_sysconfigdata.py -> /home/jy3344511/lede/build_dir/target-x86_64_musl/Python-3.9.5/ipkg-install/usr/lib/python3.9/lib-dynload
[... cut ...]
touch /home/jy3344511/lede/staging_dir/target-x86_64_musl/stamp/.python3_installed
make[3]: Leaving directory '/home/jy3344511/lede/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/compile#717.95#32.26#642.70
make[2]: Leaving directory '/home/jy3344511/lede'
make[1]: *** [package/Makefile:110: /home/jy3344511/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jy3344511/lede'
make: *** [/home/jy3344511/lede/include/toplevel.mk:230：world] 错误 2

WYC-2020, 2021-12-15:
没遇到过，我经常多次编译没遇到过

closed 2021-12-17 by AmadeusGhost

sjkhsl, 2022-03-04:
我和你的错误一样，解决了吗？好像是之前的.config文件的问题？用之前的.config文件就错误，新编译就成功。关键之前的.config文件花太多时间搜索设置了，都忘了设置哪些内容了，这特么的！