coolsnowwolf/lede #8449: 12月的代码编译后root分区占用增加100多M，那位大神知道具体增加了什么内容？

WhiteHDD, 2021-12-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
12月的代码编译后root分区占用增加100多M。

## 2. 详细叙述
### (1) 具体问题

A：12月的代码编译后root分区占用增加100多M，那位大神知道具体增加了什么内容？编译配置与之前完全一样，也重新下载的所有代码，这个增量有点大，也超过了缺省160M的rom配置，11月底前编译的版本都没有这个问题。
 
### (2) 路由器型号和固件版本

A：X86 2955U

### (3) 详细日志

[image]

EnnawYang, 2021-12-16:
docker？

wi11iamzhao, 2021-12-16:
可能是因为某个commit中默认编译了docker
去掉luci-app-docker
还有luci > collection和utilities里面的

WYC-2020, 2021-12-16:
对比一下 代码 看下makefile是不是新增了什么ipk不就知道了

WhiteHDD, 2021-12-16:
> 可能是因为某个commit中默认编译了docker 去掉luci-app-docker 还有luci > collection和utilities里面的

已经去掉docker相关的了，docker也占不了那么多

zorrose82, 2021-12-16:
同问，没改任何配置，11月编译固件gz大小80mb，12月编译128mb

XinSSS, 2021-12-16:
[image]

Cathgao, 2021-12-17:
软件包搜索docker有惊喜

WhiteHDD, 2021-12-17:
> 即使你的config没有改动， 但如果之前你的固件里没docker的话，这次是会加上去的，不只是luci>app里的docker还有luci>collection和utilities里的相关docker全部去掉再编译一遍试试看

确实不只是docker，这只占一点点。 重新对照了编译后的组件列表，>50增加，应该是版本升级后配置中缺省选上的（类似于docker），慢慢清理了，很多确实用不到的。谢谢各位

WhiteHDD, 2021-12-17:
确实不只是docker，这只占一点点。 重新对照了编译后的组件列表，>50增加，应该是版本升级后配置中缺省选上的（类似于docker），慢慢清理了，很多确实用不到的。谢谢各位

WYC-2020, 2021-12-17:
不可能是docker了，docker后面不是被去掉了再说一个docker没那么大

WhiteHDD, 2021-12-17:
> 不可能是docker了，docker后面不是被去掉了再说一个docker没那么大

是的，新增加的kmod类占多数，谢谢

closed 2021-12-17 by WhiteHDD