coolsnowwolf/lede #8451: 最新编译的固件Apple AirPlay 2 无损音频接收服务器无法使用

a8877778, 2021-12-16:
反馈bug/问题模板，提建议请删除

1.关于你要提交的问题
最新编译的固件Apple AirPlay 2 无损音频接收服务器无法使用

2. 详细叙述
(1) 具体问题
Apple AirPlay 2 无损音频接收服务器所有参数都是默认设置，点击启用按钮，显示Apple AirPlay 2 Receiver 运行中，当iphone播放音频点击连接时，手机会提示无法连接，这时openwrt会显示Apple AirPlay 2 Receiver 未运行，换过不同型号系统版本的iphone，也换路由器刷固件测试，均无法使用

(2) 路由器型号和固件版本
A：YouHua WR1200JS  固件版本：OpenWrt R21.12.1 / LuCI Master (git-21.335.48743-5f363d9)
B：Newifi2 D1  固件版本：OpenWrt R21.12.1 / LuCI Master (git-21.335.48743-5f363d9)

(3) 详细日志
Sun Dec 12 01:26:05 2021 kern.notice kernel: [    0.000000] Linux version 5.4.162 (a123456@DESKTOP-RRLFRPB) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3972-ab8bdce1c)) #0 SMP Sat Dec 11 17:25:54 2021
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] MIPS: machine is YouHua WR1200JS
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Sun Dec 12 01:26:05 2021 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sun Dec 12 01:26:05 2021 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun Dec 12 01:26:05 2021 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] Zone ranges:
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000]   HighMem  empty
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] Movable zone start for each node
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] Early memory node ranges
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Sun Dec 12 01:26:05 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Sun Dec 12 01:26:05 2021 kern.debug kernel: [    0.000000]   Normal zone: 288 pages used for memmap
Sun Dec 12 01:26:05 2021 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sun Dec 12 01:26:05 2021 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s26992 r8192 d22160 u57344
Sun Dec 12 01:26:05 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s26992 r8192 d22160 u57344 alloc=14*4096
Sun Dec 12 01:26:05 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Sun Dec 12 01:26:05 2021 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] Writing ErrCtl register=00010040
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] Readback ErrCtl register=00010040
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] Memory: 120344K/131072K available (6134K kernel code, 255K rwdata, 1288K rodata, 1268K init, 238K bss, 10728K reserved, 0K cma-reserved, 0K highmem)
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] NR_IRQS: 256
Sun Dec 12 01:26:05 2021 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] CPU Clock: 880MHz
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.007814] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.073976] pid_max: default: 32768 minimum: 301
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.078775] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.085974] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.096593] rcu: Hierarchical SRCU implementation.
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.101956] smp: Bringing up secondary CPUs ...
Sun Dec 12 01:26:05 2021 kern.warn kernel: [    3.681782] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sun Dec 12 01:26:05 2021 kern.warn kernel: [    3.681792] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun Dec 12 01:26:05 2021 kern.warn kernel: [    3.681805] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sun Dec 12 01:26:05 2021 kern.info kernel: [    3.681919] CPU1 revision is: 0001992f (MIPS 1004Kc)
Sun Dec 12 01:26:05 2021 kern.info kernel: [    0.166741] Synchronize counters for CPU 1: done.
Sun Dec 12 01:26:05 2021 kern.warn kernel: [    3.403328] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Sun Dec 12 01:26:05 2021 kern.warn kernel: [    3.403338] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun Dec 12 01:26:05 2021 kern.warn kernel: [    3.403347] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Sun Dec 12 01:26:05 2021 kern.info kernel: [    3.403403] CPU2 revision is: 0001992f (MIPS 1004Kc)
[... 423 lines cut ...]
Thu Dec 16 15:16:47 2021 kern.info kernel: [  162.742583] usb 1-2: new full-speed USB device number 2 using xhci-mtk
Thu Dec 16 15:17:25 2021 authpriv.info dropbear[4997]: Child connection from 192.168.1.159:8348
Thu Dec 16 15:17:33 2021 authpriv.notice dropbear[4997]: Password auth succeeded for 'root' from 192.168.1.159:8348
Thu Dec 16 15:18:59 2021 daemon.info hostapd: wlan0: STA 72:52:3b:ca:78:0a IEEE 802.11: authenticated
Thu Dec 16 15:18:59 2021 daemon.info hostapd: wlan0: STA 72:52:3b:ca:78:0a IEEE 802.11: associated (aid 1)
Thu Dec 16 15:18:59 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 72:52:3b:ca:78:0a
Thu Dec 16 15:18:59 2021 daemon.info hostapd: wlan0: STA 72:52:3b:ca:78:0a RADIUS: starting accounting session D77B8853A1DEE010
Thu Dec 16 15:19:03 2021 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 72:52:3b:ca:78:0a
Thu Dec 16 15:19:03 2021 daemon.info hostapd: wlan0: STA 72:52:3b:ca:78:0a IEEE 802.11: disassociated
Thu Dec 16 15:19:04 2021 daemon.info hostapd: wlan0: STA 72:52:3b:ca:78:0a IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu Dec 16 15:19:59 2021 daemon.info hostapd: wlan1: STA 9a:3c:db:47:24:41 IEEE 802.11: authenticated
Thu Dec 16 15:19:59 2021 daemon.info hostapd: wlan1: STA 9a:3c:db:47:24:41 IEEE 802.11: associated (aid 1)
Thu Dec 16 15:19:59 2021 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 9a:3c:db:47:24:41
Thu Dec 16 15:19:59 2021 daemon.info hostapd: wlan1: STA 9a:3c:db:47:24:41 RADIUS: starting accounting session F913AE76370B66B5
Thu Dec 16 15:20:16 2021 kern.info kernel: [  370.755504] do_page_fault(): sending SIGSEGV to shairport-sync for invalid write access to 00000000
Thu Dec 16 15:20:16 2021 kern.info kernel: [  370.764805] epc = 00415d17 in shairport-sync[400000+27000]
[... cut ...]
Thu Dec 16 15:32:12 2021 daemon.notice hostapd: wlan1: AP-STA-CONNECTED f2:9e:47:e1:b4:b4
Thu Dec 16 15:32:12 2021 daemon.info hostapd: wlan1: STA f2:9e:47:e1:b4:b4 RADIUS: starting accounting session 8AFAD8E65A84B585
Thu Dec 16 15:37:31 2021 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED f2:9e:47:e1:b4:b4
Thu Dec 16 15:37:31 2021 daemon.info hostapd: wlan1: STA f2:9e:47:e1:b4:b4 IEEE 802.11: disassociated due to inactivity
Thu Dec 16 15:37:32 2021 daemon.info hostapd: wlan1: STA f2:9e:47:e1:b4:b4 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

loydcheung, 2022-01-01:
我的刚刷完新固件也经常不能用，主要表现为服务显示运行中，但苹果设备无法在局域网中找到连接点，我一般重启一下就正常了。

iKenzo, 2022-05-13:
貌似问题和MT7621有关，连续在两款同时使用MT7621的不同厂商的路由器上应用失败（极路由4+GL-MT1300）,其他路由器正常

soundghost, 2023-08-06:
我K3，编译成功后luci里改好配置，点击运行成功，但是iphone搜不到airplay服务。
K2P usb版的，编译成功，改好配置运行成功，但偶尔会出现找不到airplay服务的情况，无论改端口还是怎样都是找不到airplay服务，只能恢复出厂设置重新配置一次，但故障还是偶尔会复现。

closed 2026-09-05 by coolsnowwolf