coolsnowwolf/lede #8453: 最新源码编译的luci-app-easymesh的v1.9版本，开启后会发生内核报错。

MortalM, 2021-12-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：最新源码编译的easymesh，mt7621的cr660x开启后内核会报错，但是mt7620的没事，是不是跟dsa驱动有关呢。。

### (2) 路由器型号和固件版本

A：mt7621的小米cx660x开启会报错， mt7620的小米3不会报错。
batman-adv的版本试过2021.4和2021.3的多个版本，都会报错。。

### (3) 详细日志

A：Thu Dec 16 16:11:56 2021 kern.warn kernel: [  259.889012] ------------[ cut here ]------------
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  259.893845] WARNING: CPU: 1 PID: 9854 at backports-5.10.42-1/net/mac80211/mesh_hwmp.c:348 airtime_link_metric_get+0xb8/0x688 [mac80211]
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  259.905999] Modules linked in: pppoe ppp_async batman_adv pppox ppp_mppe ppp_generic mt7915e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ts_fsm ts_bm tcp_bbr slhc sch_cake nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan libcrc32c iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables hwmon crypto_hw_eip93 crc_ccitt compat asn1_decoder sch_tbf sch_ingress sch_htb sch_hfsc
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  259.906199]  em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb sit tunnel4 ip_tunnel crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 hmac ghash_generic gf128mul gcm ecb des_generic libdes ctr cmac ccm authenc arc4 leds_gpio gpio_button_hotplug crc32c_generic
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.072144] CPU: 1 PID: 9854 Comm: luci Tainted: G        W         5.4.162 #0
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.079336] Stack : 80770000 8008013c 80720000 80720fc4 80790000 80720f8c 8071fe28 8e0cb8dc
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.087667]         808d0000 8cb45b68 8076ad43 806add88 00000001 00000001 8e0cb880 b4e57281
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.096009]         00000000 00000000 80910000 00000000 00000030 000001b9 312e342e 23203236
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.104336]         00000000 0000000c 00000000 000119d0 80000000 80790000 00000000 8d669f98
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.112662]         00000009 00000010 8d438c20 807682d0 00000001 803919d4 00000004 808d0004
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.120987]         ...
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.123424] Call Trace:
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.125885] [<8000ce38>] show_stack+0x30/0x100
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.130343] [<805eb6c8>] dump_stack+0xa4/0xdc
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.134709] [<8002e3b8>] __warn+0xc0/0x10c
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.138787] [<8002e460>] warn_slowpath_fmt+0x5c/0xac
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.143857] [<8d669f98>] airtime_link_metric_get+0xb8/0x688 [mac80211]
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.150428] [<8d609e34>] sta_set_sinfo+0xd10/0x1034 [mac80211]
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.156285] [<8d6247dc>] ieee80211_nan_func_match+0x1ddc/0x465c [mac80211]
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.163245] [<8d52c578>] nl80211_put_sta_rate+0x1430/0x15cc [cfg80211]
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.170003] ---[ end trace 8eb3ab2cd86840ee ]---
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.176151] ------------[ cut here ]------------
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.181069] WARNING: CPU: 1 PID: 9854 at backports-5.10.42-1/net/mac80211/mesh_hwmp.c:348 airtime_link_metric_get+0xb8/0x688 [mac80211]
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.193241] Modules linked in: pppoe ppp_async batman_adv pppox ppp_mppe ppp_generic mt7915e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ts_fsm ts_bm tcp_bbr slhc sch_cake nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan libcrc32c iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables hwmon crypto_hw_eip93 crc_ccitt compat asn1_decoder sch_tbf sch_ingress sch_htb sch_hfsc
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.193421]  em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb sit tunnel4 ip_tunnel crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 hmac ghash_generic gf128mul gcm ecb des_generic libdes ctr cmac ccm authenc arc4 leds_gpio gpio_button_hotplug crc32c_generic
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.359437] CPU: 1 PID: 9854 Comm: luci Tainted: G        W         5.4.162 #0
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.366630] Stack : 80770000 8008013c 80720000 80720fc4 80790000 80720f8c 8071fe28 8e0cb8dc
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.374956]         808d0000 8cb45b68 8076ad43 806add88 00000001 00000001 8e0cb880 b4e57281
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.383285]         00000000 00000000 80910000 00000000 00000030 000001ce 312e342e 23203236
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.391609]         00000000 0000000c 00000000 00057c0d 80000000 80790000 00000000 8d669f98
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.399934]         00000009 00000010 8d438c20 807682d0 00000001 803919d4 00000004 808d0004
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.408257]         ...
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.410693] Call Trace:
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.413157] [<8000ce38>] show_stack+0x30/0x100
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.417610] [<805eb6c8>] dump_stack+0xa4/0xdc
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.421972] [<8002e3b8>] __warn+0xc0/0x10c
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.426053] [<8002e460>] warn_slowpath_fmt+0x5c/0xac
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.431152] [<8d669f98>] airtime_link_metric_get+0xb8/0x688 [mac80211]
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.437720] [<8d609e34>] sta_set_sinfo+0xd10/0x1034 [mac80211]
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.443586] [<8d6247dc>] ieee80211_nan_func_match+0x1ddc/0x465c [mac80211]
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.450550] [<8d52c578>] nl80211_put_sta_rate+0x1430/0x15cc [cfg80211]
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.457480] ---[ end trace 8eb3ab2cd86840ef ]---
Thu Dec 16 16:11:56 2021 kern.warn kernel: [  260.464943] ------------[ cut here ]------------

zhq1, 2021-12-16:
我的直接死机了

zhq1, 2021-12-16:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

L-vivi, 2021-12-17:
我的AX6直接无限重启了，TTL救砖救活了

snowyu, 2021-12-23:
同上 WNDR4300v1，以及 NanoPiR4S master#f07434ac6 只要一执行`batctl if add wlan0` 就死机

sephilex, 2022-01-29:
请问7621有没有能用的版本，大概怎么操作？

coolsnowwolf, 2026-09-06:
旧 luci-app-easymesh/batman-adv 在 MT7621 上的 mac80211 airtime 警告来自旧内核驱动组合，当前代码已演进且无现行复现，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf