coolsnowwolf/lede #8458: 编译报错，显示r8152和toplevel.mk:230: world 错误

bikioxo, 2021-12-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译报错

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：Applying ./patches/010-compat-skb_mark_not_on_list-was-backported-since-ker.patch using plaintext: 
patching file compatibility.h
Hunk #1 FAILED at 548.
1 out of 1 hunk FAILED -- saving rejects to file compatibility.h.rej
Patch failed!  Please fix ./patches/010-compat-skb_mark_not_on_list-was-backported-since-ker.patch!
make[3]: *** [Makefile:62: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/realtek-r8152-linux-2.15.20211119/.prepared_ca31d3e4fff1c2eb102775f8f82a9ad6_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/lean/r8152'
time: package/lean/r8152/compile#0.15#0.06#0.20
    ERROR: package/lean/r8152 failed to build.
make[2]: *** [package/Makefile:116: package/lean/r8152/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

Neurotoxin0, 2021-12-16:
一样的错误，我的workflow从前天开始就崩了，然后本地compile报错也是mk:230

gswangmd-gssi, 2021-12-17:
一样错误，也是r8152

xinb, 2021-12-17:
一样的错误，12.17凌晨出现的

mitsukileung, 2021-12-17:
[r8152: drop patch](https://github.com/coolsnowwolf/lede/pull/8455)
合并了这个PR可以编译完成

lexin8, 2021-12-17:
追求最新源码的话就云编译，过几天就修复的
 ERROR: package/lean/r8152 failed to build.

closed 2021-12-17 by coolsnowwolf