coolsnowwolf/lede #8461: dockerd-20.10.11 报vendor version与depent version不匹配

GreatMichaelLee, 2021-12-17:
SHELL= flock /home/builder/lede_x86/tmp/.dockerd-20.10.11.tar.gz.flock -c '     /home/builder/lede_x86/scripts/download.pl "/home/builder/lede_x86/dl" "dockerd-20.10.11.tar.gz" "6fa7835bf7c17c293621967bd5096642fa1e3e1b597fbc7d7bd35f455d886495" "" "https://codeload.github.com/moby/moby/tar.gz/v20.10.11?"    '
touch /home/builder/lede_x86/build_dir/target-x86_64_musl/dockerd-20.10.11/.prepared_34486257dcbf1e1211fc48ebdf486224_6664517399ebbbc92a37c5bb081b5c53_check
. /home/builder/lede_x86/include/shell.sh; gzip -dc /home/builder/lede_x86/dl/dockerd-20.10.11.tar.gz | tar -C /home/builder/lede_x86/build_dir/target-x86_64_musl/dockerd-20.10.11 --strip-components 1 -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/builder/lede_x86/build_dir/target-x86_64_musl/dockerd-20.10.11

Applying ./patches/001-libdevmapper_name_fix.patch using plaintext:
patching file hack/make.sh
# Verify dependencies are the vendored version
( DEP_VER=$( grep --only-matching --perl-regexp '(?<=PKG_SOURCE_VERSION:=)(.*)' "../containerd/Makefile" ); VEN_VER=$( grep --only-matching --perl-regexp '(?<=_COMMIT:=)(.*)(?=})' "/home/builder/lede_x86/build_dir/target-x86_64_musl/dockerd-20.10.11/hack/dockerfile/install/containerd.installer" ); if [ "${VEN_VER}" != "${DEP_VER}" ]; then echo "ERROR: Expected 'PKG_SOURCE_VERSION:=${VEN_VER}' in '../containerd/Makefile', found 'PKG_SOURCE_VERSION:=${DEP_VER}'"; exit 1; fi )
( DEP_VER=$( grep --only-matching --perl-regexp '(?<=PKG_SOURCE_VERSION:=)(.*)' "../libnetwork/Makefile" ); VEN_VER=$( grep --only-matching --perl-regexp '(?<=_COMMIT:=)(.*)(?=})' "/home/builder/lede_x86/build_dir/target-x86_64_musl/dockerd-20.10.11/hack/dockerfile/install/proxy.installer" ); if [ "${VEN_VER}" != "${DEP_VER}" ]; then echo "ERROR: Expected 'PKG_SOURCE_VERSION:=${VEN_VER}' in '../libnetwork/Makefile', found 'PKG_SOURCE_VERSION:=${DEP_VER}'"; exit 1; fi )
( DEP_VER=$( grep --only-matching --perl-regexp '(?<=PKG_SOURCE_VERSION:=)(.*)' "../runc/Makefile" ); VEN_VER=$( grep --only-matching --perl-regexp '(?<=_COMMIT:=)(.*)(?=})' "/home/builder/lede_x86/build_dir/target-x86_64_musl/dockerd-20.10.11/hack/dockerfile/install/runc.installer" ); if [ "${VEN_VER}" != "${DEP_VER}" ]; then echo "ERROR: Expected 'PKG_SOURCE_VERSION:=${VEN_VER}' in '../runc/Makefile', found 'PKG_SOURCE_VERSION:=${DEP_VER}'"; exit 1; fi )
ERROR: Expected 'PKG_SOURCE_VERSION:=52b36a2dd837e8462de8e01458bf02cf9eea47dd' in '../runc/Makefile', found 'PKG_SOURCE_VERSION:=f46b6ba2c9314cfc8caae24a32ec5fe9ef1059fe'
make[3]: *** [Makefile:169: /home/builder/lede_x86/build_dir/target-x86_64_musl/dockerd-20.10.11/.prepared_34486257dcbf1e1211fc48ebdf486224_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/utils/dockerd'
time: package/feeds/packages/dockerd/compile#0.37#0.82#1.19
    ERROR: package/feeds/packages/dockerd failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/dockerd/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:230：world] 错误 2

GreatMichaelLee, 2021-12-17:
make clean大法和make dirclean大法以及删除builder_dir大法都不管用，只有去除docker编译才行，看看是哪个makefile跟新一下md5把。

cyrinkwork, 2021-12-17:
同问题

GreatMichaelLee, 2021-12-17:
哪位大佬帮忙救下急

WYC-2020, 2021-12-17:
@GreatMichaelLee 看另外几个docker的issue我有说解决方法，最根本是有人升级了runc

cyrinkwork, 2021-12-17:
> 

应该不是runc的问题，runc报错我之前也有，刚才看了一下，runc升级到1.0.3那个已经被移除了

cyrinkwork, 2021-12-17:
@WYC-2020 应该不是runc报错，runc报错的问题我之前也有，但是runc升级到1.0.3那个已经被移除了

WYC-2020, 2021-12-17:
[image]
 看你这个错误明显就是嘛 @cyrinkwork   是云编译还是本地，云编译去掉哪个缓存，哪个作怪，如果是本地编译，万能大法试下

chiyu521, 2021-12-17:
编辑 ../runc/Makefile 
PKG_SOURCE_VERSION：= f46b6ba2c9314cfc8caae24a32ec5fe9ef1059fe
改成
PKG_SOURCE_VERSION：= 52b36a2dd837e8462de8e01458bf02cf9eea47dd
测试编译通过

closed 2021-12-17 by coolsnowwolf