coolsnowwolf/lede #8463: 求助libevdev/compile] Error 1&target-x86_64_musl/stamp/.package_compile] Error 2 mk:230: world] Error 2

freestyleSunday, 2021-12-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：echo "libkmod" >> /home/parallels/lede/staging_dir/target-x86_64_musl/pkginfo/kmod.default.install
make[3]: Leaving directory '/home/parallels/lede/feeds/packages/utils/kmod'
time: package/feeds/packages/kmod/compile#0.20#0.09#0.33
make[3]: Entering directory '/home/parallels/lede/feeds/packages/libs/libevdev'
Makefile:23: ../../devel/meson/meson.mk: No such file or directory
make[3]: *** No rule to make target '../../devel/meson/meson.mk'.  Stop.
make[3]: Leaving directory '/home/parallels/lede/feeds/packages/libs/libevdev'
time: package/feeds/packages/libevdev/compile#0.13#0.05#0.17
    ERROR: package/feeds/packages/libevdev failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/libevdev/compile] Error 1
make[2]: Leaving directory '/home/parallels/lede'
make[1]: *** [package/Makefile:110: /home/parallels/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/parallels/lede'
make: *** [/home/parallels/lede/include/toplevel.mk:230: world] Error 2

### (2) 路由器型号和固件版本

A：最新编译

### (3) 详细日志

A：A：echo "libkmod" >> /home/parallels/lede/staging_dir/target-x86_64_musl/pkginfo/kmod.default.install
make[3]: Leaving directory '/home/parallels/lede/feeds/packages/utils/kmod'
time: package/feeds/packages/kmod/compile#0.20#0.09#0.33
make[3]: Entering directory '/home/parallels/lede/feeds/packages/libs/libevdev'
Makefile:23: ../../devel/meson/meson.mk: No such file or directory
make[3]: *** No rule to make target '../../devel/meson/meson.mk'.  Stop.
make[3]: Leaving directory '/home/parallels/lede/feeds/packages/libs/libevdev'
time: package/feeds/packages/libevdev/compile#0.13#0.05#0.17
    ERROR: package/feeds/packages/libevdev failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/libevdev/compile] Error 1
make[2]: Leaving directory '/home/parallels/lede'
make[1]: *** [package/Makefile:110: /home/parallels/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/parallels/lede'
make: *** [/home/parallels/lede/include/toplevel.mk:230: world] Error 2

WYC-2020, 2021-12-17:
全新编译第一次吗？如果不是 试下万能大法

lsyyy233, 2021-12-17:
> 全新编译第一次吗？如果不是 试下万能大法

试过删除tmp目录 make clean make dirclean，都是一样的问题，而且还会有这样的提示：

closed 2021-12-17 by coolsnowwolf

StoneForests, 2022-11-10:
你的问题解决了吗？我用oracle cloud的amd的虚拟机，ubuntu 18.04编译也是一样的问题。。

`
Makefile:26: ../../devel/meson/meson.mk: No such file or directory
Build-Depends: meson/host

Package: libesmtp
Version: 1.1.0-0
Depends: +libc +USE_GLIBC:librt +USE_GLIBC:libpthread +libpthread +libopenssl
Conflicts: 
Menu-Depends: 
Provides: 
Section: libs
Category: Libraries
Repository: packages
Title: A Library for Posting Electronic Mail
Maintainer: Othmar Truniger <github@truniger.ch>
Source: libesmtp-1.1.0.tar.gz
License: LGPL-2.0-or-later
LicenseFiles: COPYING
Type: ipkg
Description: A Library for Posting Electronic Mail
https://libesmtp.github.io/
Othmar Truniger <github@truniger.ch>
@@

make[2]: *** No rule to make target '../../devel/meson/meson.mk'.  Stop.
`