coolsnowwolf/lede #8465: 缺少meson/meson.py

wi11iamzhao, 2021-12-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
当前版本([a849091](https://github.com/coolsnowwolf/lede/commit/a8490915d3caf2c89938a333e266b7b21e6f3d5d))编译失败
### (2) 路由器型号和固件版本
X86-64软路由,ESXI7.0u2d虚拟化
编译环境：MS Windows11 Pro WS
WSL2 Ubuntu 20.04LTS
### (3) 详细日志
`
git pull
./scripts/feeds update -a && ./scripts/feeds install -a
make defconfig
make -j8 download
PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin make -j1 V=s 
......
/home/lede/staging_dir/host/bin/python: can't open file '/home/lede/staging_dir/host/lib/meson/meson.py': [Errno 2] No such file or directory
make[3]: *** [Makefile:134: /home/lede/build_dir/target-x86_64_musl/glib-2.70.0/.configured_efb2d3cf01c707184fb77fd0fe149f94] Error 2
make[3]: Leaving directory '/home/lede/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/compile#0.19#0.04#0.19
    ERROR: package/feeds/packages/glib2 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/glib2/compile] Error 1
make[2]: Leaving directory '/home/lede'
make[1]: *** [package/Makefile:110: /home/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lede'
make: *** [/home/lede/include/toplevel.mk:230: world] Error 2
`

coolsnowwolf, 2021-12-17:
fixed

jlzsss, 2021-12-17:
还是报错：
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/network/utils/dpdk/Makefile' has a build dependency on 'ninja/host', which does not exist
WARNING: Makefile 'package/feeds/packages/gstreamer1/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/iputils/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/jose/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/json-glib/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/jsoncpp/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/knot-resolver/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/libdrm/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/libinput/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/liblz4/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/libmpdclient/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/libpsl/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/libsoup/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/lighttpd/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/kenzok8/luci-app-k3screenctrl/Makefile' has a dependency on 'phicomm-k3screenctrl', which does not exist
WARNING: Makefile 'package/feeds/packages/mpc/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/nlohmannjson/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/p11-kit/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/php7/Makefile' has a dependency on 'libgd-full', which does not exist
WARNING: Makefile 'package/feeds/packages/php8/Makefile' has a dependency on 'libgd-full', which does not exist
WARNING: Makefile 'package/feeds/packages/pixman/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/pulseaudio/Makefile' has a build dependency on 'meson/host', which does not exist
WARNING: Makefile 'package/feeds/packages/spice-protocol/Makefile' has a build dependency on 'meson/host', which does not exist

wi11iamzhao, 2021-12-17:
[当前版本](https://github.com/coolsnowwolf/lede/commit/ea5c0f00791dba034d52f65e370631081c8b5710)编译成功
closed

closed 2021-12-17 by wi11iamzhao