coolsnowwolf/lede #8468: WARNING: Makefile 'package/feeds/helloworld/naiveproxy/Makefile' has a build dependency on 'ninja/host', which does not exist

GreatMichaelLee, 2021-12-17:
这次又不知道改了什么导致ninja/host说没有，/scripts/feeds install -a 的时候，麻烦查一下

Create index file './feeds/packages.index'
Collecting package info: done
Create index file './feeds/luci.index'
Create index file './feeds/routing.index'
Create index file './feeds/telephony.index'
Create index file './feeds/helloworld.index'
builder@Build-Server:~/lede_x86$ ./scripts/feeds install -a
**WARNING: Makefile 'package/feeds/helloworld/naiveproxy/Makefile' has a build dependency on 'ninja/host', which does not exist**
Installing all packages from feed packages.
WARNING: Not overriding core package 'dnsproxy'; use -f to force
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed helloworld.
builder@Build-Server:~/lede_x86$
builder@Build-Server:~/lede_x86$ vi feeds/packages/utils/runc/Makefile
builder@Build-Server:~/lede_x86$

WYC-2020, 2021-12-17:
l大不是修复了？更新

GreatMichaelLee, 2021-12-17:
哪里修复了？没有。

WYC-2020, 2021-12-17:
@GreatMichaelLee 看大佬package库的更新记录，参考修复，你这个不属于l大的。去作者哪里提，或者自己修复吧，会给你直接关闭的

minlang112, 2021-12-18:
make distclean
可破，亲测

closed 2021-12-19 by AmadeusGhost