coolsnowwolf/lede #8469: Makefile 'package/feeds/packages/v2raya/Makefile' has a dependency on 'xray-core', which does not exist

cirillu69, 2021-12-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

./scripts/feeds install -a
Collecting package info: package/feeds/packages/udp-broadcast-relay-redux-openwr
Collecting package info: done
WARNING: Makefile 'package/feeds/packages/v2raya/Makefile' has a dependency on 'xray-core', which does not exist

### (1) 具体问题
當我嘗試運行命令時，我不斷收到此信息

A：在虛擬盒Debian 64位下為小米AX 3600創建固件

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：cat  feeds.conf.default

src-git packages https://github.com/coolsnowwolf/packages

src-git luci https://github.com/coolsnowwolf/luci

src-git routing https://git.openwrt.org/feed/routing.git

src-git telephony https://git.openwrt.org/feed/telephony.git

#src-git video https://github.com/openwrt/video.git

#src-git targets https://github.com/openwrt/targets.git

#src-git oldpackages http://git.openwrt.org/packages.git

#src-link custom /usr/src/openwrt/custom-feed

WYC-2020, 2021-12-17:
这……

990052053, 2021-12-18:
我也是，一直是依赖不存在

WYC-2020, 2021-12-18:
这个好解决的，l大好像是吧这个移动到helloword

990052053, 2021-12-18:
不是只有这一个，很多的插件在安装阶段都显示没有依赖

990052053, 2021-12-18:
[image]

WYC-2020, 2021-12-18:
我这边自己全套没任何问题，只是是用的我自己的仓库

hcym, 2021-12-18:
这个用master编译，就是定制性不好，而且编译时间极长，这两天的master分支很好，再加上大雕的和passwall插件完美

minlang112, 2021-12-18:
make distclean
可破，亲测

cirillu69, 2021-12-18:
> make distclean

> 这个好解决的，l大好像是吧这个移动到helloword

您的评论是否已修复，不请分享

binhuoca, 2021-12-18:
完全一样的问题，win10 ubuntu，不知如何解决

cirillu69, 2021-12-18:
> 完全一样的问题，win10 ubuntu，不知如何解决

是的，你是对的。用于创建固件的命令 make 不会继续。我是 openwrt 的新手，将不胜感激。

binhuoca, 2021-12-19:
> 亲测两遍，先删除整个lede文件夹，再con给clone开始，最后还是一样问题。
高手能不能指点一下，是否可以手动拉取缺失的文件到本地，然后再install？

binhuoca, 2021-12-19:
> 
麻烦指点一下，你的意思是这个问题可以忽略，系统里另有一套好的v2ray内核；
还是说要手动copy一下,如果这样，该怎么操作?

binhuoca, 2021-12-19:
> 这个好解决的，l大好像是吧这个移动到helloword

麻烦指点一下，你的意思是这个问题可以忽略，系统里另有一套好的v2ray内核；
还是说要手动copy一下,如果这样，该怎么操作?

binhuoca, 2021-12-19:
> make distclean 可破，亲测

亲测两遍，先删除整个lede文件夹，再con给clone开始，最后还是一样问题。
高手能不能指点一下，是否可以手动拉取缺失的文件到本地，然后再install？

gili-gili, 2021-12-20:
> 亲测两遍，先删除整个lede文件夹，再从clone开始，最后还是一样问题。 高手能不能指点一下，是否可以手动拉取缺失的文件到本地，然后再install？

我也这样，但是加了别人的库就好了。
一般可以忽视，你实在要的话，我给个地址，自己clone以后复制一份吧
https://github.com/fw876/helloworld

cirillu69, 2021-12-21:
> > 亲测两遍，先删除整个lede文件夹，再从clone开始，最后还是一样问题。 高手能不能指点一下，是否可以手动拉取缺失的文件到本地，然后再install？
> 
> 我也这样，但是加了别人的库就好了。 一般可以忽视，你实在要的话，我给个地址，自己clone以后复制一份吧 https://github.com/fw876/helloworld

是的，请真的很感激。

Ra2-IFV, 2021-12-21:
> 是的，请真的很感激。

如果解決了的話就close issue吧:P

closed 2021-12-29 by AmadeusGhost