coolsnowwolf/lede #8470: 我的理解是**Turbo ACC 网络加速**是一个集成的网络加速**控制界面**ipk应用，只是为了**方便开关**包含以下几种加速：

Axe-l, 2021-12-17:
我的理解是**Turbo ACC 网络加速**是一个集成的网络加速**控制界面**ipk应用，只是为了**方便开关**包含以下几种加速：
1.Flow Offloading 转发加速 ,来自高通开源的 Shortcut FE 转发加速引擎 (Fast Path) ，一种纯软件算法上的优化。可以针对openwrt所有型号的路由器。
2.HWNAT加速。一般只针对MT762x芯片型号的路由器可以开启。比如常见的NEWIFI D2,K2P等。但目前针对19.07系列不是很完善，会造成路由器崩溃，不是MT762x芯片的路由器你开启也没有作用.
3. BBR加速。是谷歌弄出来的一个内核级别的tcp协议的优化，可以提高网络连接的能力(争抢带宽能力)，这是linux内核4.9以后都可以用的一个东西，内核大于4.9的所有型号路由器都能用
4.DNS 加速。主要是利用 pdnsd 做本地缓存，提高DNS解析速度，所有型号路由器都能用

在这个控制界面里还可以看FULLCONE NAT是否启用的，它的启用和关闭不在Turbo ACC 网络加速 里，而在“网络”->“防火墙”里可以来控制启用。它主要是针对iptables的一个改进吧，不过原作者的github里也说了，只针对几个linux内核版本做过测试，有些linux内核版本上使用是有问题的

_Originally posted by @bluehj777 in https://github.com/coolsnowwolf/lede/issues/4533#issuecomment-623364963_

WYC-2020, 2021-12-18:
这都能提一个issue哈哈

minlang112, 2021-12-18:
什么鬼东西……

hxlls, 2021-12-18:
什么鬼玩意，新开个ISSUE？这是嫌开发者不够累么？

gili-gili, 2021-12-18:
建议Close

bluehj777, 2021-12-18:
啥意思？诈尸了。为啥用我的回复开一个ISSUE

CuteTony, 2021-12-20:
建议尽快close issue

closed 2021-12-28 by AmadeusGhost