coolsnowwolf/lede #8471: Chineese language in mt7615 drivers

ZILLKA, 2021-12-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
I've using english language in LuCI. Almost everything is translated, but wifi driver settings are in chinese.

### (1) 具体问题

A：Is it possible to translate wifi settings to english or russian language?

### (2) 路由器型号和固件版本

A：Beeline SmartBox Turbo+
OpenWrt R21.12.1 / LuCI Master (git-21.335.48743-5f363d9)

### (3) 详细日志
[image]

A：

paldier, 2021-12-22:
The wireless page only support Chinese when using mtk dirver.

ZILLKA, 2021-12-24:
You mean mtk proprietary driver for mt7615? I've tried different options for 7615 when compile firmware, also language only chinese.

skbeh, 2021-12-29:
You can try https://github.com/immortalwrt/immortalwrt. Its mt7615 driver setting is in English.

coolsnowwolf, 2026-09-06:
当前 mt76/无线驱动配置中的中文字符串属于驱动本身的描述/日志内容，不是 LEDE 当前可独立维护的 LuCI 翻译项。该 issue 仅为翻译建议，暂无可验证的仓库缺陷，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf